Sloeber/arduino-eclipse-plugin #868: After upgrading to V4.2 the product is still V4.0

jantje, 2017-11-20:
[image]
Which means #866 is not included.

wimjongman, 2017-11-21:
I don't understand this issue. What exactly is the problem?

wimjongman, 2017-11-21:
I see that 4.2 is installed in the 4.0 product.

jantje, 2017-11-21:
I upgraded my working eclipse and I wanted to check whether everything was upgraded and I saw that the product is still on 4.0 and that is not what I expected.

wimjongman, 2017-11-21:
It is expected.

See this fresh install:

[image]

all is fine:

[image]

wimjongman, 2017-11-21:
There is a way to upgrade the ini file of an existing product but it is not easy. Also a fresh install gets the latest Arduino stuff.

So reinstall the product for a shiny result  
[image]

closed 2017-11-21 by wimjongman

jantje, 2017-11-21:
ok. Thanks