Sloeber/arduino-eclipse-plugin #869: Program "\bin\g++" not found in PATH, Program "\bin\gcc" not found in PATH

Danial-Zafar, 2017-11-21:
Issue while using v 4.1 on win64: When I switch from release to local(which is a debug project), it started to disappear some things from Include folder and give above mentioned 2 error

[image]
[image]
[image]
[image]
[image]
[image]

Danial-Zafar, 2017-11-21:
when I build this in this state it showed an error of 
subdir.mk.24: recipe for target 'ino.cpp.o' failed
make: *** [.ino.cpp.o] Error1

I tried on of your solution that is mentioned one of issues which to add cmd.exe/c in build command

jantje, 2017-11-21:
duplicate of #867

closed 2017-11-21 by jantje

wimjongman, 2017-11-21:
#870

jml3on, 2018-08-10:
HI,

running into this issue as well.
If I run through the clean cycle from the debug video (install mingw in a folder with no space; run sloeber.cmd; create blink project; install hardware from git; create debug config; build), I invevitably get:

08:48:57 **** Incremental Build of configuration Debug for project blink ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"/bin/g++" -c -g -O0 -w -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -DF_CPU=16000000L -DARDUINO=10802 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR_LOCAL   -DLOCAL_RUN   -I"C:\github\hardware\jantje\avr_local\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\avr\1.6.21\variants\standard" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"   -Dsrandom=srand -Drandom=rand
The system cannot find the path specified.
subdir.mk:18: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1

08:48:58 Build Failed. 2 errors, 0 warnings. (took 248ms)

I've tracked it down to the environment variable A.RUNTIME.TOOLS.MINGW.PATH not being set.

If I set it manually in my config, the build works. But after a while (not sure what triggers it), that variable disappears and the problem comes back.

So far, I haven't been able to figure out how those environment variables are generated nor from where.

Pointer would be greatly apreciated.

jml3on, 2018-08-11:
to clarify for others who may get stuck like I did: manually installing jantje/hardware is no longer required since sloeber 4.2 (and will lead to that broken experience).

debug platforms are now bundled and only need to be selected under Arduino > Preferences > Platforms and Boards.

jantje, 2018-08-11:
>So far, I haven't been able to figure out how those environment variables are generated nor from where.

A very old post (by now incomplete) but still very valid
http://blog.baeyens.it/#post24