Sloeber/arduino-eclipse-plugin #870: sloeber(buildall error)

toobayounas, 2017-11-21:
--> windows8.1, sloeber **buildall error**

[image]
 I am trying to debug arduino on sloeber version 4.1, and i'm facing error shown in png,

wimjongman, 2017-11-21:
Are you posting the same issue under two different github users? #869

Danial-Zafar, 2017-11-21:
no

wimjongman, 2017-11-22:
Or should I call you @ichigo92?

wimjongman, 2017-11-22:
@Danial-Zafar @ichigo92 and @toobayounas, why are you posting the same screenshots under three different users?

ichigo92, 2017-11-22:
@wimjongman i don't know what makes you think the other two are me, just to clarify i am not them, i am facing a similar issue as @toobayounas, i posted the details to the error i am getting on  issue #853.

Danial-Zafar, 2017-11-22:
@wimjongman I have no I intention to irritate you. It is just that I and @toobayounas has an assignment related to ardunio unit testing. We are students and we are trying to come up with a solution.
@toobayounas she is my batch mate so she has the similar task to told her about the sloeber. I, @toobayounas  and @ichigo92 are three different persons. 
It will be very kind of you if you please guide us to the solution.

Danial-Zafar, 2017-11-22:
If you are guessing that we are the same persons by the same file names of our projects, it is due to we were following the @jantje youtube debugging and unit testing tutorials.

wimjongman, 2017-11-22:
Alright. It is just that you are posting the same information in the same way about exactly the same problem. 

If you want our help, please do not open different issues about the same problem or crosspost to issues that are already closed. 

From this point on, @Danial-Zafar @ichigo92 and @toobayounas, we will be using this issue to see if we can help you. OK?

ichigo92, 2017-11-22:
@wimjongman alright, i am using v4.2

Danial-Zafar, 2017-11-22:
I'm using 4.2 now before that I was using 4.0 and 4.1

Danial-Zafar, 2017-11-22:
@wimjongman fine.

wimjongman, 2017-11-22:
Okay so what are you trying to do? It looks like you follow an advanced video. Is it your intention to follow an advanced video or do you just want to get started?

Danial-Zafar, 2017-11-22:
Yes, my intention is to follow that, as I want to unit test Arduino code...

jantje, 2017-11-22:
The video is outdated. In 4.2 it is lots easier.
But as @wimjongman  (and I already told you) Do not use spaces in the path. Read the faq and try simple things first (F.I. install zero package and compile)
In a new 4.2 install you should have jantje hardware you can install and use to debug/unit test.
And issues on the jantje hardware part should be posted on the hardware repository https://github.com/jantje/hardware and not here and for sure not on unrelated patron posts.

closed 2017-11-22 by jantje

wimjongman, 2017-11-22:
[image]

Danial-Zafar, 2017-11-22:
@jantje there is no space in path now...

wimjongman, 2017-11-22:
Also not in the workspace path?

uzi18, 2017-11-22:
Anywhere

Danial-Zafar, 2017-11-22:
yes, there is no space in workspace path as well...

jantje, 2017-11-22:
Please provide from one and the system workspace and eclipse installation following information in the order as given (Even better is to record a video doing these actions),
Delete the release and local debug folder in the project
Screenshot of the platform and boards (like wim's screenshot but where we can see what is installed)
Screenshot of the project properties->arduino->localdebug configuration
Screenshot of the whole workspace (visible project manager,console and main ino file)
Do build of release configuration
Screenshot of the whole workspace (visible project manager,console and main ino file)
Do build of localdebug configuration
Screenshot of the whole workspace (visible project manager,console and main ino file)
Content of the console (so no screenshot but actual text)