Sloeber/arduino-eclipse-plugin #871: OTA to ESP8266 not working anymore after 4.2 upgrade

Ullie, 2017-11-24:
After the 4.2 upgrade the OTA to an ESP8266 does not work anymore. When I click "upload sketch" the sketch is compiled, but the ssh uploader is started, instead of the python espota uploader.

The error I receive is: 

Starting upload
using ssh loader
ERROR: Unable to connect to _<x.x.x.x>_
java.net.ConnectException: Connection refused: connect
failed to upload 

In the 4.1 version this functionality worked flawless.

jantje, 2017-11-24:
I seem to have the same behaviour. I'll check

jantje, 2017-11-24:
ok here is how it works
If you have a space in you upload comport ssh uploader is used
if no space in upload comport the esptool is used

Ullie, 2017-11-24:
Aha, nice to know it is reproducable. Is there a workaround? Or do you have to fix it first?

jantje, 2017-11-24:
It is repoducible in the sense that
you have a space in the port-> ssh uploader is used
you have no space->esptool is used
in other words a space in the comport tells sloeber to go for web upload. This is because local comports do not have spaces.
As to me all works as designed.

Ullie, 2017-11-24:
Ah..but the 4.1 behaviour was different: it detected the ota Port by bonjour and used the python uploader. And that is the functionality I need. According to my knowledge the software should check if the ssh_upload tag is set in the mdns and if it is set to no it should use the generic uploader. However I can't find how this is implemented (not a java programmer)

jantje, 2017-11-25:
Lots of things have changed in the upload process so there sure can be differences but it has always been like
-> No space in port name do default upload
-> Space in port  name do something special
The bonjour service always returns a space in the name and as such selecting a bonjour found name always triggers "do something special". 
I have been using ota from before the the yun board came out (using hacked openwrt router) and I'm using it frequently now with yun boards and shields (F.I. with marvin). However I have no active devices based on the esp8266; so for sure none with esp8266 and OTA. And the OTA for esp8266 is far more complicated than yun OTA.
As such it is very hard for me to understand your problem (not a esp8266 programmer).
Therefore it is extremely important to be clear and exact in the communication.

As you are saying there is a regression I propose you do as follows:
Install 4.1 and 4.2 on your system and make a video clearly showing the different behaviour between the 2 versions.

Ullie, 2017-11-25:
Is it possible to install both versions on one system? Or how can I do a roll back the easiest way? Maybe it's easier to do a remote control session or have a phone call? Dutch is lots easier ;)

Ullie, 2017-11-25:
I think the most obvious difference is the way the esp is detected (now: ip & and name, before: bonjour name at ip:portnumber) When I'm home I'll take some pictures

jantje, 2017-11-25:
>Is it possible to install both versions on one system? 

yes

>Or how can I do a roll back the easiest way? 

Backup (= copy) your workspace and eclipse folder before changing

>Maybe it's easier to do a remote control session or have a phone call? 

Are you a seriously paying patron? If not, why would I?

Ullie, 2017-11-25:
Thanks for the tips. No, I am not. But I was offering my time to help you making your software function better/fix a regression issue. Do not appreciate the tone of voice in our conversation. I'll just revert and wait if/until one of your paying customers will find the time to help you fix this major issue in regards to the esp8266 platform.

TrippyLighting, 2017-11-25:
Patreon is a service that was created to support people that provide an enormous amount of voluntary services to society or communities and in turn offer a chance for people to support these efforts with donations as small a 1$ a month, ore even one time donations.

If every user that uses Sloeber to compile their code and is able to dispense with $1 a month would actually spent only $1 a month to support this effort, maybe such requests as yours to require Jantje to volunteer his time and fix an issue that creates a problem for you, would receive a more positive response and Jantje would be in a better mood.

The initial meaning behind "free" in free and open source software was not free software for everyone to take. It was to have the freedom of choice. Open source does not live from the vast amuont of users that gladly take, it lives from contribution of spirits such as @jantje and @wimjongman.

As such I invite you to to choose to contribute just one dollar a month to this project.

Ullie, 2017-11-26:
We all spend our time differently and the force of open source is people helping eachother, creating a solution that suits all needs best. I was offering my time to help Jantje(and myself offcourse), but I like to be as efficient as possible and do not like grumpyness. If I had paid for this software and received an update that broke usability for a major part we would have an entire different conversation, don't you think? As a said: my issue is solved by reverting, everyone happy!

judge2005, 2017-11-30:
I am confused. Jante say:

> you have a space in the port-> **ssh uploader** is used
> you have no space->**esptool** is used

And:

> The bonjour service always returns a space in the name and as such selecting a bonjour found name always triggers "do something special".

But ESP OTA requires the use of the **espota** tool, not the esptool nor the ssh uploader. So I assume there used to be additional logic that checked if the board was an ESP (or was using mdns or some such).

Full disclosure: I am using 4.0.0

jantje, 2017-11-30:
Assuming you use the default uploader
If there is no space in the port name a usb upload is done 
If there is a space in the port nme special code is triggered (My current assumption is it tries to upload over the network but this is all based on the boards.txt stuff.)

miky2k, 2017-12-23:
Starting upload
using ssh loader
ERROR: Unable to connect toambient0
java.net.ConnectException: Connessione rifiutata (Connection refused)
failed to upload

for me too (esp8266 with arduinoOTA enabled and slober 4.2).

PS:I'm on patreon.

jantje, 2017-12-27:
It seems that this change has been undone when I moved code around.
https://github.com/Sloeber/arduino-eclipse-plugin/commit/97aefec6af97a36d05553baf7d822f8cd1c80ae8#diff-c94fbbaf18333d3bb1391f6d9f3245c4R1078

jantje, 2017-12-28:
A workaround can be to add the following code to the programmers.txt of esp8266.   
After saving the programmers.txt go to project properties->arduino and select the programmer and provide the ipadress of the esp8266 as uploadport (no spaces!!) 
Note that this workaround hardcodes the uploadport you can change it to what you need. If you need multiple make multiple programmers.

esp8266OTAProgrammer.name=esp8266 network upload
esp8266OTAProgrammer.program.tool=esp8266OTA
esp8266OTAProgrammer.tools.esp8266OTA.program.pattern={TOOLS.ESPTOOL.UPLOAD.NETWORK_PATTERN}
esp8266OTAProgrammer.network.port=8266

cesarjhony, 2018-01-06:
If your connection has authentication, add line:
esp8266OTAProgrammer.network.password=yourPasswordHere

The location of programmers.txt is "SloeberPath\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.0-rc2"

Thanks jantje, please add some TAG in this issue to be easy for found.

jantje, 2018-01-07:
@cesarjhony 
> please add some TAG in this issue to be easy for found.

Can you be more precise on what would help?

cesarjhony, 2018-01-07:
To help everybody, to find this solution, put keyword, tag, this kind of thing.
I know, that issue is temporary, but for now it is very useful.

jandrassy, 2018-02-25:
esp8266 OTA upload doesn't work in 4.2, 4.1, 4.0. it worked in 3.1

jantje, 2018-02-25:
>in 4.x versions it is not a bug one could find and fix. the implementation is simply not there. it should be in UploadSketchWrapper.internalUpload() in if (host != null) but the network upload tool is not even evaluated into BoardDescriptor or in getBuildEnvironmentVariable(ACTION_UPLOAD)

That is what I meant in https://github.com/Sloeber/arduino-eclipse-plugin/issues/871#issuecomment-354179213
Still the workaround works

jandrassy, 2018-02-25:
btw, in 3.1 for OTA upload to Atmega over esp8266 with WiFi Link firmware or esp-link firmware this works too in pre_processing_platform.txt:
```
#for wifilink network upload
tools.wifilinkOTA={tools.avrdude.network_cmd}
tools.wifilinkOTA.upload.pattern={tools.avrdude.upload.network_pattern}
avr.network.upload.tool=wifilinkOTA
```
In standard Arduino IDE it works from the box, no additional configuration needed.

zerorist, 2018-03-15:
Hello
I work with esp8266, sloeber and OTA for 1 year now.

It worked flawlessly with 4.1
The update to 4.2 was kind of messy, and I had to reconfigure a few things to make everything work.
OTA on 4.2 was not working
I followed jantje patch on programmers.txt file, it fixed my issue.

Thank you a lot for this very usefull piece of software.

jandrassy, 2018-03-15:
@Ullie and @zerorist, you had 3.1 not 4.1 until the upgrade to 4.2?

jantje, 2018-03-15:
>@Ullie and @zerorist, you had 3.1 not 4.1 until the upgrade to 4.2?

I think so because the upgrade from V3 to V4 is messy (like all major upgrades) and OTA for esp8266 does not work in 4.X out of the box.
The workaround should work in any tool that supports the arduino framework and programmers.
IMHO the OTA of esp8266 forces a dirty implementation by not simply being a uploader.
When I fix this it will probably be the workaround implementation.

jandrassy, 2018-03-15:
@jantje, yes espota and ardinoOTA are not uploaders to mcu over Serial (direct or transparent over network), they do network upload to the system that uploads bin/hex to mcu.  

I have  this 'workaround' for AVR arduinoOTA tool
```
otaProgrammer.name=network upload
otaProgrammer.program.tool=arduinoOTA
otaProgrammer.tools.arduinoOTA.program.pattern={TOOLS.AVRDUDE.UPLOAD.NETWORK_PATTERN}
```

jantje, 2018-03-15:
Basically I see no difference between programmers and uploaders.
I think the difference is historically in the sense that Arduino uses usb upload (using a bootloader) and then started to support hardware programmers (that for instance use the jtag). Therefore they put these in programmers.
But from a end user perspective there is only "get my compiled code on the mcu". The user doesn't care how that is done. Current Arduino ide uploads are:
1) over a serial port and with a bootloader (standard avr)
2) over network with commands run on remote linux (yun network upload)
3) Teensy loader (I'm not sure how this one works)
4) esp OTA (well this is actually network upload as only the last part is wifi for sure)

So IMHO there is no difference between upload and program and things would be lots easier/cleaner if we would see "upload" as "default way to program". and "programmers" as "other ways to program".

Continuing this line of thought: As you need to program the esp8266 with a special program (over usb);
 "upload" has to be usb programming and 
  OTA becomes a "programmer". 

I discussed this with the arduino IDE OTA implementer a long time ago here on github (did not start nice, did not end nice). The typical clash between someone trying to fix 1 thing and someone looking at the overall picture.

jandrassy, 2018-03-15:
Isn't the only difference the reset of the board? For the "USB upload" the IDE disables the Serial Monitor, resets the board and the starts the external tool like avrdude. All other ways are only execution of the right external tool with the right parameters?

jantje, 2018-03-15:
Don't read if you can't stand a rant :-)

>Isn't the only difference the reset of the board?

No. Resetting is a option. More and more arduino boards are reset with the programmer tool (like avrdude). This is reflected by following options (default they are false)
```
yun.upload.use_1200bps_touch=true
yun.upload.wait_for_upload_port=true
```

>All other ways are only execution of the right external tool with the right parameters?

No Arduino avr programmers.txt v1.6.21 contains following line:
`avrisp.communication=serial`
This allows the IDE to know the serial port needs to be closed and reopened (and this is the only use I can  think off).
And even if it were ... IMHO not a good enough reason to create 2 concepts all the way to implementation.

I don't want to think about it to much as it makes me sad but as we are on the topic (and you seem to be interested) The confusion between the difference between upload and program is bad but the "different types of upload" and the implementation is worse.
Look at:
`yun.upload.via_ssh=true`
This key,value pair states that upload to yun is possible via ssh but not that it is enforced (which I think is acceptable). 
It also does not enforce that the command to run is the same as if you were local. Which I think is unacceptable.
This because in the "upload tool" the code decides to go to another tool which is exactly what programmers are meant for. (resulting in the same thing programmed in 2d different ways in the same program which is bad bad bad ...)
That is why platform.txt contains multiple upload patterns

```
 tools.avrdude_remote.upload.pattern=/usr/bin/run-avrdude ...
 tools.avrdude.upload.pattern="{cmd.path}" 
```
And why avrdude_remote and not avrdude.remote????

Apart from that it would have been much easier to select a uploader/programmer with
`yun.upload.via_ssh=yunSSHProgrammer`
With the yunSSHProgrammer defined in it's own space.

Again note that IMHO programmer and upload only are different because of history and this results in having 2 keys (and an extra file) for the same thing, adding to the confusion.

jandrassy, 2018-03-16:
now I got it. and it is `pattern` versus `network_pattern` in platform.txt for esp8266 and  in platform.txt for AVR other then Yun.
```
tools.avrdude.network_cmd={runtime.tools.arduinoOTA.path}/bin/arduinoOTA
tools.avrdude.upload.network_pattern="{network_cmd}" -address {serial.port} -port {upload.network.port} -sketch "{build.path}/{build.project_name}.hex" -upload {upload.network.endpoint_upload} -sync {upload.network.endpoint_sync} -reset {upload.network.endpoint_reset} -sync_exp {upload.network.sync_return}
```
so if network port is selected and no other special cases apply (like Yun or Tensy), the Arduino IDE evaluates the tool with 'network_pattern' instead of 'pattern'.

but there are other network upload tools for AVR (mcuota or avrdude has network upload too) and then I need to change the platform.txt. It is simpler to have programmers configured.

jantje, 2018-03-16:
> It is simpler to have programmers configured.

That is my point. Thanks for understanding.

jantje, 2018-05-13:
I'm having a hard time to reactivate this code :-s
I tried in Arduino IDE and I don't understand how it is supposed to work there. As mnds is not part of the OTA example... how do you tell Arduino IDE the ip adress to go to?

In my version here I get the following now
```
Launching: python.exe C:\eclipse\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.4.1/tools/espota.py -i 192.168.1.144 -p 8266 --auth="admin" -f C:\workspaces\runtime-EclipseApplication\atotest/Release/atotest.bin 
Output:
Authenticating...FAIL
00:48:16 [ERROR]: No Answer to our Authentication
python.exe  finished
upload done
```

with this sketch running on a lolin board
```
#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

const char* ssid = "slavinnenhof";
const char* password = "Jan Was Here Aga1n to Kick you in the balls with his f0od";

void setup() {
  Serial.begin(115200);
  Serial.println("Waiting for serial monitor");
  delay(10000);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  ArduinoOTA.setHostname("myesp8266");
  Serial.println("OTA Hostname set");

  // No authentication by default
  ArduinoOTA.setPassword("admin");
  Serial.println("OTA password set");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH)
      type = "sketch";
    else // U_SPIFFS
      type = "filesystem";

    // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
    else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
    else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
    else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
    else if (error == OTA_END_ERROR) Serial.println("End Failed");
  });
  ArduinoOTA.begin();
  Serial.println("Ready");

}

void loop() {
  ArduinoOTA.handle();
}
```

jantje, 2018-05-14:
@jandrassy 
Thanks
```
void ArduinoOTAClass::begin() {
...
  MDNS.begin(_hostname.c_str());
...
}
```
Apparently esp8266 mDNS is not working good(?) on my  network. Yun mDNS works fine, making me think it mDNS was not enabled.
I fear I have to fix a network issue first :-(

jandrassy, 2018-05-14:
but in Sloeber project preferences the port field is editable. I enter the IP address there for 'the workaround' if it is not read from mDNS. the mDNS propagation is sometimes slow.

jantje, 2018-05-14:
@jandrassy 
that is what I did but then I got the error as posted above. So I wanted to check the upload command using Arduino IDE.
Then I bumped into the mDNS issue.
If you can confirm the OTA upload command is ok I can push the changes to the nightly... but....
It has been quite a invasive change. So I want to check other uploads before I release this change. 
Also note that currently the port and password are grabbed from the code and as such changing the port and password in a OTA upload is not supported.

jandrassy, 2018-05-14:
I never tested ota upload with authentication. Did you test it without authentication?

jantje, 2018-05-14:
It used to work with authentication. But now it doesn't work at all :-(

jandrassy, 2018-05-14:
the 2.4.1 release of esp8266 arduino core uses newer Espressif SDK then previous versions and there is some compatibility broken with WiFi settings remembered on the flash of the esp8266. I recommend to delete the esp8266 flash with esptool erase_flash command to clear those settings

jantje, 2018-05-14:
Thanks for the tip. I tried erase_flash didn't help.
It takes a very long time for the esp to get a ip adress. but it is gettingone.
mDNS does not seem to get through. 
Should I be able to ping to the esp?

jantje, 2018-05-15:
It is still not working for me but I think that is due to local network issue.
I tested upload over serial with following boards
```
			MCUBoard[] boards = {  
					Teensy.teensypp2("COM10"),
					Teensy.Teensy3_1("COM24"), 
					ESP8266.wemosD1("COM23"),
					Arduino.fried("COM22"), 
					Arduino.yun("COM5"), 
					Arduino.uno("COM6"),
					Arduino.getMega2560Board("COM11"), 
					Arduino.zero("COM14"), 
					Arduino.dueprogramming("COM8"),
					Arduino.leonardo("COM30"), 
					Arduino.arduino_101("COM15") };
```
These work fine in windows

jantje, 2018-05-16:
Can someone give this a swing?

jantje, 2018-05-16:
As to using ip versus network name. The network name is the way to go there the ip adress is grabbed by sloeber at "configuration time". If the device uses a ip address provided by the network the ip address will change over time. but the name.local should not.

jandrassy, 2018-05-18:
it is working. thank you

closed 2018-05-18 by jantje