Sloeber/arduino-eclipse-plugin #880: Auto select serial port

ianmacl, 2017-12-08:
One thing I've always found annoying is that if I unplug my device and plug it back in, it sometimes gets assigned a different path, especially if I didn't remember to plug it into the same USB port.  It would be nice if the serial monitor and uploader could automatically determine which port to use.

jantje, 2017-12-08:
I agree it would be nice but I have no clue how to implement a functionality like this.

von13, 2018-01-14:
On Linux you can use 'by-path' enumeration, like this /dev/serial/by-path/pci-0000:00:1a.0-usb-0:1.2:1.0

Just check what you have in /dev/serial/by-path

In OS X it doesn't change in my case (high sierra), it depends on which port I use so it works like 'by-path' in Linux.

Maybe there is a way on Windows, but I don't use it

jantje, 2018-03-04:
@ianmacl Now we had our discussions on Slack and we had some time to think about this.
Do you see it feasible to implement what you want?
Or shall we close the issue

ianmacl, 2018-03-06:
I think it is feasible to implement; however at this point in time I don't have the time to dig in and implement it with the suggestions that were made, so I'll probably just run a local fork with the approach that I used and deal with it.

You can close and if I have time at a later date I'll resubmit.

Thanks for all you do with this project!

closed 2018-03-06 by ianmacl