Sloeber/arduino-eclipse-plugin #884: Eclipse Sloeber upload error after update

adimtheworst, 2017-12-12:
!ENTRY io.sloeber.arduino.core 4 0 2017-12-12 13:57:49.538
!MESSAGE tools.Program tool not properly configured.program.pattern : not found in the platform.txt file

!ENTRY io.sloeber.arduino.core 4 0 2017-12-12 13:57:49.901
!MESSAGE Failed to run the Upload recipe 
!STACK 0
java.io.IOException: Cannot run program "" (in directory "/home/adim/eclipse-workspace"): error=2, No such file or directory
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1048)
	at io.sloeber.core.tools.ExternalCommandLauncher.launch(ExternalCommandLauncher.java:294)
	at io.sloeber.core.tools.uploaders.GenericLocalUploader.RunConsoledCommand(GenericLocalUploader.java:42)
	at io.sloeber.core.tools.uploaders.arduinoUploader.uploadUsingPreferences(arduinoUploader.java:87)
	at io.sloeber.core.tools.uploaders.UploadSketchWrapper$UploadJobWrapper.run(UploadSketchWrapper.java:178)
	at org.eclipse.core.internal.jobs.Worker.run(Worker.java:56)
Caused by: java.io.IOException: error=2, No such file or directory
	at java.lang.UNIXProcess.forkAndExec(Native Method)
	at java.lang.UNIXProcess.<init>(UNIXProcess.java:247)
	at java.lang.ProcessImpl.start(ProcessImpl.java:134)
	at java.lang.ProcessBuilder.start(ProcessBuilder.java:1029)
	... 5 more

adimtheworst, 2017-12-12:
I am sorry i thought i wrote a text, 
Hi, my computer updated the eclipse and after the update took place when i try to upload an Arduino sketch an error occurs saying 'Failed to run the Upload recipe' 'tools.Program tool not properly configured.program pattern: not found in the platform.txt file'. Also after the update while the eclipse was loading another error was telling me to go to preferences c/C++ and disable something in the index, i did it and all this happened. Can anyone please help me, eclipse my main IDE

Thank you

ALUBahn, 2018-01-05:
Hello, please try 
[image]
i had the same Problem and this solved it.

by

jantje, 2018-01-07:
Did you do the workspace update actions as described here?
http://eclipse.baeyens.it/how_to.shtml#/n

@ALUBahn 
Please do not mess with the environment variables. And do not advice others to do so.

ALUBahn, 2018-01-07:
hi @jantje, ok i normaly agree with you and i don't do this again, but after hours on searching a solution for me i found this way. changing the value from /dev/ttyACM0 to ttyACM0 works. BUT IT IS only a WORKAROUND. After every changing the Port from unplug the Arduino i must do this. 
Please check this for a SOLUTION. 
Thank you.

jantje, 2018-01-08:
This envionment variable should contain the exact same text you enter in project properties->arduino->uploadport.
This field is a combo textbox which means you can enter any value there  and have a list of selectable options. So entering ttyACM0 instead of  /dev/ttyACM0 there should be the way to go.

ALUBahn, 2018-01-08:
Hi jantje, 
Ok, i am sorry but 

i used Sloeber 4.1 and had no Problems with the Upload. I updated to Sloeber 4.2 and the  Issue is there. I use a Arduino Due on Programming Port. 
If i enter in properties->arduino->uploadport --> /dev/ttyACM0 the Reset works but the Upload fails with the Message 'no device found'. If i enter in properties->arduino->uploadport --> ttyACM0 the Reset won't work with the Message 'no device found'. 
If i enter in properties->arduino->uploadport --> /dev/ttyACM0 the Reset works and i set the unwanted value of the Env Var to ttyACM0 the Upload works to. 
The two settings must differ and all things good. Thats the issue. 

i hope it helps to find the solution.

regards ALUBahn

jantje, 2018-01-08:
@ALUBahn 
First off. I just noticed you are setting a build variable and not a environment variable (which I assumed).
I have never used build variables (with sloeber) so I have no clue about what is the difference. Seems I have something new to learn here, thanks. :-)

What you describe sounds like a real issue that can be worked around by modifying the build variables. I have to give you that :-) . 

What we know:.
Board: due on programming port
os: linux
domain upload
problem description reset ok upload fails with device not found
workaround : after setting the uploadport to /dev/ttyACM0  change the build (not the environment) variable JANTJE.COM_PORT to ttyACM0 

To further investigate please provide : 
1) the platform version number: I assume Arduino SAM 1.6.11.
2)the upload console output (in text) from the "as should work"  (upload should fail)
3)the upload console output (in text) with your work around. (upload is ok)

PS: Note I don't think it is related to the original poster issue but we'll use this issue anyway.

ALUBahn, 2018-01-08:
Hello Jantje, that's right. 
[image]
[image]

Ok. Thank you. 

regards ALUBahn

jantje, 2018-01-08:
It looks like resetting the board is not needed and if the reset works the upload command is triggered before the port is back available.
Can you try  to run the upload commands from the terminal?
I guess both will work

ALUBahn, 2018-01-08:
Hello, the reset isn't perhaps needed, but only ttyACM0 works sometimes in this constellation:

` ~ $ /home/yyyyz/apps/Arduino-Sloeber/sloeber//arduinoPlugin/packages/arduino/tools/bossac/1.6.1-arduino/bossac --port=ttyACM0 -U false -e -w -b /home/yyyy/Dokumente/arduino-sloeber-workspace/Due_Farbverlauf_2/Release/Due_Farbverlauf_2.bin -R 
No device found on ttyACM0
~ $ /home/yyyy/apps/Arduino-Sloeber/sloeber//arduinoPlugin/packages/arduino/tools/bossac/1.6.1-arduino/bossac --port=/dev/ttyACM0 -U false -e -w -b /home/yyyy/Dokumente/arduino-sloeber-workspace/Due_Farbverlauf_2/Release/Due_Farbverlauf_2.bin -R 
No device found on /dev/ttyACM0
~ $ /home/yyyy/apps/Arduino-Sloeber/sloeber//arduinoPlugin/packages/arduino/tools/bossac/1.6.1-arduino/bossac --port=/dev/ttyACM1 -U false -e -w -b /home/yyyy/Dokumente/arduino-sloeber-workspace/Due_Farbverlauf_2/Release/Due_Farbverlauf_2.bin -R 
No device found on /dev/ttyACM1
~ $ /home/yyyy/apps/Arduino-Sloeber/sloeber//arduinoPlugin/packages/arduino/tools/bossac/1.6.1-arduino/bossac --port=ttyACM1 -U false -e -w -b /home/yyyy/Dokumente/arduino-sloeber-workspace/Due_Farbverlauf_2/Release/Due_Farbverlauf_2.bin -R 
No device found on ttyACM1
~ $ /home/lutz/apps/Arduino-Sloeber/sloeber//arduinoPlugin/packages/arduino/tools/bossac/1.6.1-arduino/bossac --port=ttyACM0 -U false -e -w -b /home/yyyy/Dokumente/arduino-sloeber-workspace/Due_Farbverlauf_2/Release/Due_Farbverlauf_2.bin -R 
Atmel SMART device 0x285e0a60 found
Erase flash
done in 0.041 seconds

Write 35720 bytes to flash (140 pages)
[==============================] 100% (140/140 pages)
done in 7.173 seconds
Set boot flash true
CPU reset.
~ $ /home/yyyy/apps/Arduino-Sloeber/sloeber//arduinoPlugin/packages/arduino/tools/bossac/1.6.1-arduino/bossac --port=/dev/ttyACM0 -U false -e -w -b /home/yyyy/Dokumente/arduino-sloeber-workspace/Due_Farbverlauf_2/Release/Due_Farbverlauf_2.bin -R 
No device found on /dev/ttyACM0
`
I now think there is a issue with bossac using the port.

ALUBahn, 2018-01-08:
I tryed, to change the build Var for bossacpath to set on 1.7.0 and it work fine:

Starting upload
using arduino loader
Forcing reset using 1200bps open/close on port/dev/ttyACM0
Continuing to use/dev/ttyACM0
Ending reset

Launching/home/yyyy/apps/Arduino-Sloeber/sloeber//arduinoPlugin/packages/arduino/tools/bossac/1.7.0/bossac --port=/dev/ttyACM0 -U false -e -w -b /home/yyyy/Dokumente/arduino-sloeber-workspace/Due_Farbverlauf_2/Release/Due_Farbverlauf_2.bin -R 
Output:
Atmel SMART device 0x285e0a60 found
Erase flash
done in 0.037 seconds

Write 35720 bytes to flash (140 pages)

[                              ] 0% (0/140 pages)
[===                           ] 10% (14/140 pages)
[======                        ] 20% (28/140 pages)
[=========                     ] 30% (42/140 pages)
[============                  ] 40% (56/140 pages)
[===============               ] 50% (70/140 pages)
[==================            ] 60% (84/140 pages)
[=====================         ] 70% (98/140 pages)
[========================      ] 80% (112/140 pages)
[===========================   ] 90% (126/140 pages)
[==============================] 100% (140/140 pages)
done in 7.177 seconds
Set boot flash true
CPU reset.
bossac finished
upload done

i had set 
[image]

perhabs it is the wanted solution??

Gretings ALUBahn

jantje, 2018-01-09:
This workaround assumes a recent arduino samd platform has been installed. But I prefer it to the JANTJE.COM_PORT workaround.

jantje, 2018-01-10:
Seems like the exact same issue now. Even with the arduino IDE.
Other people reported similar issues with due uploads.
I'm closing this issue as it is not a Sloeber issue.
If you disagree please reopen stating why you disagree

closed 2018-01-10 by jantje

y2d2, 2019-12-12:
Dear Sloeber Community, 
I disagree with the last response of jantje because I tried it out on both the arduino IDE 1.8.10 and the stand alone sloeber V4.3.2 (freshly downloaded yesterday). On Ubuntu 18.04. For both I used the Arduino SAM board v1.6.12. Where as I could push the sketch with out issues or tricks to the Arduino Due with the Arduino IDE. When using Sloeber I had the same error as described above. Changing the build variable (JANTJE.COM_PORT) as suggested by ALUBahn  did the trick though. 
So not a serious issue, certainly one with a workarround, but not an arduino issue as such. 

PS: I really  like the work done with Sloeber! Great Job!