Sloeber/arduino-eclipse-plugin #885: Space in Folder name kills Make

aberglas, 2017-12-16:
Tried to set up project under 
c:\...\Google Drive\...
(Note space)

Would not make with obscure errors.

Suspect some missing ""s around file names.

Spaces in folder names are common in Windows, so this is important for beginners.  Took me a little while to figure out the cause.

Thanks for providing Sloeber.

wimjongman, 2017-12-16:
Yeah that is a bummer. It is described in the FAQ though, and I'm pretty sure we post a message about spaces in paths, don't we?

http://eclipse.baeyens.it/faq.shtml#/troubleshooting

https://stackoverflow.com/questions/9838384/can-gnu-make-handle-filenames-with-spaces

closed 2017-12-16 by wimjongman

aberglas, 2017-12-17:
Fair enough that make is broken.  However, given that this is often used by beginners, and often on windows, I would suggest an explicit test in the build so it spits out an error saying no spaces in files.

wimjongman, 2017-12-18:
Hi @aberglas. Yes that might not be a bad solution. Are you able to contribute this patch? Let me know if you need help getting started.

wimjongman, 2017-12-18:
@aberglas I have created a contribution guide here:

https://github.com/Sloeber/arduino-eclipse-plugin/wiki/Contribution-Guide

aberglas, 2017-12-18:
Thanks for that.

I am flat out with other commitments at the moment but might look at it in
the new year.

Anthony