Sloeber/arduino-eclipse-plugin #889: Eclipse environment variables

oscil, 2017-12-19:
HI. When creating project in eclipse- environment variables is project settings set wrong. For example- avr-gcc is in /usr/bin folder, but in project settings he try to find compiler in just /bin folder in var A.COMPILER.PATH. It creates a error while compilng. But, if i change the path- it is OK. And there is about 7-10 places, those think that smth is in /bin folder.
So- how can i set this environment variables for all projects (new and already created)? 
I tryed make smth in window-prefences-c/c++-build-environment, but it didnt help.
I tryed copy settings from one project to another- and it didnt help again)
I tryed reinstall Sloeber after installing avr-gcc, same result.

wimjongman, 2017-12-19:
Did you try to toy with the C++/Build/Settings preferences?

oscil, 2017-12-19:
Nope. And i dont see any setting, that can help me. May be you know, what setting i should change there?

jantje, 2017-12-19:
All the A. environment variables are Sloeber controlled variables and should not be messed with.
You should not install avr-gcc next to sloeber.
Have you seen some of these video's?
https://www.youtube.com/playlist?list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a financial contribution, this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://eclipse.baeyens.it/donate.shtml

oscil, 2017-12-19:
no, i didnt see any of this videos, i just wanted to make some simple code in eclipse) and trouble is not that sloeber installed with avr-gcc, trouble is that project by default try to find files in place where is no them. 
As you can see in images, in default project he try to find compiler in **/bin** folder, but there is no compiler.
But if write there **/usr/bin** - it works well, make sketch and upload it to arduino. 
And the question- where should i make changes, those will affect all future projects?
[image]
[image]

Thank you

jantje, 2017-12-19:
>no, i didnt see any of this videos

You should
> trouble is not that sloeber installed with avr-gcc, trouble is that project by default try to find files in place where is no them.

no trouble is that you have not watched the video's. As such I assume you are not using sloeber the default way  and claim you do.
We have spend years to develop the code and write documentation and make video's and you don't bother watching them. (do you see the difference of effort?)
Still you are convinced you have found a issue in the default way of using Sloeber which is used for more than 10.000 builds a day by plenty of people worldwide.
Think again.

oscil, 2017-12-20:
so, i watched some videos and read some manual, and found what was the trouble) same files in eclipse plugin folder didnt download completely and could not be extracted correctly. And plugin couldnt install correct platform build variant. So, i delete and reinstall plugin and everything is ok now, thank you)

closed 2017-12-20 by wimjongman

jantje, 2017-12-20:
Thanks for putting in the effort.

jantje, 2023-11-07:
As this is an old thread where Sloeber still uses A.XX (uppercased and A. prefix for arduino variables) environment variables please open a new issue and explain in detail what you did and what didn't work.