Sloeber/arduino-eclipse-plugin #891: envirenment variables should use some built-in vars (such as ${ProjDirPath})  instead of hardcode absolute path

skcks, 2017-12-21:
such as 

`environment/project/io.sloeber.core.toolChain.release.2138783379/A.RECIPE.AR.PATTERN.1/value="/Applications/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru "/Volumes/Doc/eclipse-project/test_pwm/Release/arduino.ar" 
`

change to 

`environment/project/io.sloeber.core.toolChain.release.2138783379/A.RECIPE.AR.PATTERN.1/value="/Applications/Eclipse.app/Contents/Eclipse//arduinoPlugin/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-ar" cru "${ProjDirPath}/Release/arduino.ar" 
`
there are few environment variables need to change .
it will be more flexible for  someone write the code in the different directory of different computer,
it is a little tedious to config all paths again once change the diretory of project  .

wimjongman, 2017-12-21:
Thanks for filing. Are you able to provide a patch @skcks? We have a contribution guide here [1]

[1] https://github.com/Sloeber/arduino-eclipse-plugin/wiki/Contribution-Guide

jantje, 2018-01-07:
Honestly I think this is a misunderstanding of the internal workings of sloeber.
All the A.xxx environment variables are Automatic variables and are calculated as needed.
Moving a workspace, opening the workspace from a different eclipse install or changing from OS will trigger a recalculation of the A.xxx variables. This is why there are redundant environment variables with the name
JANTJE.WORKSPACE_LOCATION
JANTJE.PROJECT_NAME
JANTJE.OS_NAME
But basically: "All the A. and JANTJE. variables are managed via the gui. Do not modify them directly. Don't even look at them." If you need to play around with these values use platform.txt, boards.txt or programmers.txt files as described in the arduino platform documentation.

If you have a problem please provide a usecase that shows this issue. Do not assume there is a issue because....
If you find one in this area: note that the workaround is : open the project properties->arduino->apply->ok. As this triggers a recalculation of all the calculated environment variables. And yes there is a very good reason these environment variables are extended.

closed 2018-01-07 by jantje