Sloeber/arduino-eclipse-plugin #893: No carriage return in the serial monitor

netbomo, 2017-12-28:
Hello,
I've just install your plugin on Eclipse Neon on a debian stretch laptop.
I'm using a simple Serial.println("foo"); and new line have an effect but not the carriage return.
I mean, on GTKterm each line is correctly set on the left but not on your serial monitor (shell console).

There is a option somewhere to select the right end line sequence?
I've taken a look in your FAQ but I've found no things.

Thanks you,
gilou

wimjongman, 2017-12-28:
[image]

netbomo, 2017-12-28:
Hello,
Thanks you for your reply.
My console don't look as your, I've not the control bar with lists and buttons...
[image]

wimjongman, 2017-12-28:
That is not our software. Please download the Sloeber IDE here [1].

[1] http://eclipse.baeyens.it/stable.php?OS=Linux

closed 2017-12-28 by wimjongman

netbomo, 2017-12-28:
Hum,
Now I understand. I've install the eclipse plugin, not download the complete IDE.

Thanks.
gilou

wimjongman, 2017-12-28:
You did not install the Sloeber plugin but something else.