Sloeber/arduino-eclipse-plugin #894: Looks Like Teensy Upload Broken by Upgrade to v4.2

gfvalvo, 2018-01-06:
Sloeber v4.2
Arduino IDE v1.8.5
Teensyduino v1.40
"Private hardware path" in Sloeber pointing to: D:\arduino-1.8.5\hardware\teensy\avr
Win 7 Pro, SP1

It looks like Sloeber 4.2 upgrade broke ability to work with Teensy boards.

After attempting to upload code to Teensy 3.2 from Sloeber, the USB COM port on the PC "disappears" from Windows Device Manager.

Unplugging and Replugging the board from USB port doesn't bring it back. Changing USB ports does not bring it back.

Only way I've found to get the port with Teensy back is to upload a sketch to the Teensy using Arduino IDE.

wimjongman, 2018-01-07:
Can you throw away the arduinoPlugin directory and restart? This allows Sloeber to freshly fetch the arduinoPlugin code.

gfvalvo, 2018-01-07:
Just to be perfectly clear, I have Sloeber installed here: D:\sloeber. 
You I suggesting that I delete folder: D:\sloeber\arduinoPlugin
And then restart the Sloeber / Eclipse IDE?

Is there anything I should do to the Arduino v1.8.5 / Teensyduino v1.40 combination that's installed at:
D:\arduino-1.8.5  ?

I'm asking because the Teensy support seems different than other boards. You need to have:
"Private hardware path" in Sloeber pointing to: D:\arduino-1.8.5\hardware\teensy\avr

wimjongman, 2018-01-07:
Yes, delete (or rename) D:\sloeber\arduinoPlugin and restart.

I am not aware of any special Teensy hacks for 4.2. Try disabling that private hardware path.

gfvalvo, 2018-01-07:
No Joy. After compiling Sloeber opens up the Teensy Loader which appears to successfully program and reboot the Teensy 3.2. The console shows:

LaunchingD:/arduino-1.8.5/hardware/teensy/../tools/teensy_post_compile -test -file=NonBlockingTempTeensy -path=D:\SloeberWorkspace\NonBlockingTempTeensy/Release -tools=D:/arduino-1.8.5/hardware/teensy/../tools -board=TEENSY31 -reboot 
Output:
Opening Teensy Loader...
D:/arduino-1.8.5/hardware/teensy/../tools/teensy_post_compile finished
upload done

However, the Teensy COM Port then disappears from Win 7 Device Manager. After this Sloeber can't find the Port for either uploads or Serial Monitor. The only way I've found to get the COM Port back is to upload a sketch using Arduino 1.8.5 instead of Sloeber.

wimjongman, 2018-01-07:
So you did not disable that private hardware path?

gfvalvo, 2018-01-07:
No, should I try that?

gfvalvo, 2018-01-07:
After some more testing, it looks like the problem is sketch-related. Compile and upload from Sloeber works OK with the simple Blink sketch, but not a more complex one I was working on. I'll do some more tests and report back.

gfvalvo, 2018-01-07:
My bad. I had the wrong USB Type selected in the project's Properties --> Arduino. I had accidently selected "Flight Sim Controls + Joystick" when I created the project. That's right under the correct choice of "Serial" choice in the drop down menu.

All is well now. Sorry for the bother. Thanks for taking an interest.

wimjongman, 2018-01-07:
Consider supporting the project: https://www.patreon.com/bePatron?c=177453

closed 2018-01-07 by wimjongman

gfvalvo, 2018-01-08:
Thanks, I will.