Sloeber/arduino-eclipse-plugin #896: Sloeber package damaged on MacOS

kenkopelson, 2018-01-11:
Hello,
I downloaded the 4.2 version of the Sloeber bundle today, and found that the package is damaged, and unable to be opened.  We downloaded using several computers, all with the same result.  It seems that your MacOS package has been corrupted somehow, so I was wondering if you could please test this, and put up a new package that is working properly?  Many thanks!  Please advise when this is complete, if at all possible :)
Cheers,
Ken

jantje, 2018-01-11:
Do you have the same problem with the nightly?

kenkopelson, 2018-01-13:
Since I cannot launch any version of Sloeber product, I cannot use Nightly Builds, as they are only updates to an existing running Sloeber.
When anyone downloads the Sloeber bundled product, and I try to open it on MacOS (High Sierra), a box immediately opens telling me that my Sloeber (the one I'm trying to open) is damaged and to send it to the trash.

jantje, 2018-01-13:
A) you can download the nightly product http://eclipse.baeyens.it/nightly.php
B) you can download 4.1 and upgrade to 4.2 using the nightly update site
C) have you read this page http://eclipse.baeyens.it/installAdvice.shtml

kenkopelson, 2018-01-15:
Hello :)  Yes, I read the "installAdvice" page, and specifically, that is why I am asking for help.  In that advice it says:

* use java 8 (included in the windows product downloads)
* Take the product and not the plugin.
* Take the latest stable version.
* Read your OS specific info linked on this page at the right.

I tried to use the Sloeber plugin with Eclipse, and it was unable to upload to the Arduino because in the place where it specifies the Serial port, it says "/dev/su." as a prefix, whereas the latest version of OS/X requires it to say just "su." (stripping the /dev/ from the Serial port name).  Others are also having this particular issue as well.  So, this is why I am wanting to use the Sloeber product, on the chance that you have fixed this issue.  But I see your point in using the Nightly, as it has the best chance of having the fix.  On this issue however, I guess I was really wanting to use the "Stable" version, instead of a possibly shakier "Nightly" build version.  We are very pressed for time on a deadline, so I'm really hoping to minimise any downtime due to tool issues.

By the way, I very much appreciate the excellent work you've done on Sloeber, and as a developer I do appreciate all the hard work that has gone into it.  That is why we want to use it...it's really a great piece of software, and there really aren't any viable alternatives out there for Arduino boards.

I will try to do the "upgrade to version 4.2" route, and see how it goes.  I'll post something more if I run into problems.  Many thanks once again!  Your impressive work truly is much appreciated and admired!

jantje, 2018-01-15:
> I tried to use the Sloeber plugin with Eclipse, and it was unable to upload to the Arduino because in the place where it specifies the Serial port, it says "/dev/su." as a prefix, whereas the latest [...]

The upload port is a text field you can enter whatever you want.

> We are very pressed for time on a deadline,

We are very pressed for budget. 
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2018-01-18 by jantje