Sloeber/arduino-eclipse-plugin #897: Environment variables settings for ESP32

steblerjun, 2018-01-14:
Hello Jantje,
can you tell me how the environment variables (project properties -> C/C++ Build -> Environment) are set?
My Problem:
I have a ESP32-project (Board: ESP32 Dev Module) and I want to use the NeoPixelBus_by_Makuna-library (version 2.2.9) but compile fails:
then I compared some build commands with the Arduino IDE:
if I change the value of the environment variable A.BUILD.ARCH
from ARDUINO-ESP32 to ESP32
and a command at those variables A.RECIPE.C.O.PATTERN.1 + A.RECIPE.CPP.O.PATTERN.1 + A.RECIPE.S.O.PATTERN.1
from -DARDUINO_ARCH_ARDUINO-ESP32 to -DARDUINO_ARCH_ESP32
then it compiles!
because of this "wrong" platform setting I'm getting this error:
error: #error "Platform Currently Not Supported, please add an Issue at Github/Makuna/NeoPixelBus"

I also saw another difference between sloeber and Arduino IDE:
build command in sloeber:
"-DARDUINO_BOARD=\"ESP32_DEV\""
in Arduino IDE:
-DARDUINO_BOARD="ESP32_DEV" -> the are less quotation marks and back slashes...
at those environment variables A.RECIPE.C.O.PATTERN.1 + A.RECIPE.CPP.O.PATTERN.1 + A.RECIPE.S.O.PATTERN.1
but I don't know if this makes a difference to the build job.

I'm using Win10 Pro + sloeber 4.2 (Windows 4.2 Latest Stable 64bit) and installed ESP32 as you told here: http://eclipse.baeyens.it -> youtu.be/szRrvJPvQz4

I'm asking you how the environment variables are set because every time I change, for example the com port of the ESP32 (in project properties -> Arduino -> Arduino board selection), then click "Apply and Close", these variables are overwritten to the "default" values...

king regards
Franz

jantje, 2018-01-14:
A very long time ago I have written an blog post about environment variables. It is still very valid:
http://blog.baeyens.it/#post24

As to fixing problems changing environment variables is very seldom the way to go. I assume that in your case there is something like below in your code
```
#ifdef XXXXX
....
error: #error "Platform Currently Not Supported, please add an Issue at Github/Makuna/NeoPixelBus"
.....
#endif
```
the quick fix is to add -DXXX to the extra compile options in project properties->arduino
>then I compared some build commands with the Arduino IDE:

The problem here is: what are we comparing. To investigate we need a "fixed ESP32" version we can install in Arduino IDE and Sloeber in a "documented way" before we can start comparing. To me that means having a package json file. Until then I do not feel like investigating differences.

As to "/'/´....
Sloeber uses the windows commands directly and Arduino IDE uses something else (I think I remember a cygwin thingy) resulting in different interpretations of command line amongst others in regards quotes.
This is one of the hard to get working cross-platform stuff of sloeber. IMHO the example you are providing is one of the easier cases and is fine.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

steblerjun, 2018-01-14:
Thank you for your quick responce jantje!

> the quick fix is to add -DXXX to the extra compile options in project properties->arduino

is working so far...

the next problem is now when I add this *-DARDUINO_ARCH_ESP32* to the extra compile option it automatically **!?** adds the FreeRTOS-Library (Eclipse-menu / Arduino / Add a library to the selected project / Import Arduino libraries) to my project and I'm getting this error:

*C:\sloeber-Eclipse\arduinoPlugin\libraries\FreeRTOS\10.0.0-3\src\FreeRTOSVariant.h:31:20: fatal error: avr/io.h: No such file or directory*

I have to remove the FreeRTOS library and I'm finally able to compile my project again...

and by the way, I'm already a supporter at patreon (5$/month)!
but only since 7 days ;-)

Franz

jantje, 2018-01-14:
>and by the way, I'm already a supporter at patreon (5$/month)!

I thought so. Thanks.

The auto import library sometimes imports to many libraries. You can safely delete these (also the needed ones, they should reappear though). You can also turn off the auto import functionality in windows->preferences and use arduino->import libraries to manually decide which libraries you want.

The problem with the autoimport importing to many libraries is quite fundamental. Basically sloeber is using the cdt indexer functionality "search for unresolved includes".
However the indexer is originally written to "find as much code as possible" where for import libraries you only want to find "part of my sketch code".
Next to that lots of code needs to be excluded in arduino world. Excluding code is not so common in C/C++ world which results in indexer confusion.
What I'm trying to say is that hickups like wrongly importing libraries can only be avoided by turning the auto import functionality off.

jantje, 2018-01-18:
can this issue be closed?

steblerjun, 2018-01-18:
yes, this issue can be closed - I'm able to compile so far...

please try to fix the environment variable A.BUILD.ARCH from *ARDUINO-ESP32* to *ESP32* for the next version of sloeber!

thank you

closed 2018-01-18 by steblerjun

jantje, 2018-01-19:
Thanks.
I verified where A.BUILD.ARCH is comming from it is the foldername in capitals.
Telling git to extract in the folder esp32 instead of arduino-esp32 is the real solution :-)

steblerjun, 2018-01-21:
I changed my folder name and look at that! thank you jantje!! 

that's what I wanted to find out when I was asking:

> can you tell me how the environment variables (project properties -> C/C++ Build -> Environment) are set?

when I opened this issue...

jantje, 2018-01-21:
> can you tell me how the environment variables (project properties -> C/C++ Build -> Environment) are set?

There is no simple answer to that question. There are many things that end up in the environment variables and I had forgotten about the folder name being one.
What I can think about is:
There is the boards.txt platform.txt programmers.txt from both the referencing and referenced platform.
There is the json file again from the referenced and referencing platform
There is the .txt files in the arduinoPlugin folder
There are folder names
And the settings in the gui
and probably others.