Sloeber/arduino-eclipse-plugin #898: Cannot IDE/plugin build on Mac OS X

von13, 2018-01-14:
I'm trying to build sloeber IDE from source (4d3fe0a45fda5ccb408a89bab9de2636c2eb3fe1), but something is wrong with dependencies. Could you please help me with resolving these dependencies?

What do I need to install before building Sloeber IDE?

./build_then_launch.sh 
Trying to build and then launch the Arduino Eclipse IDE
First we build with Maven. This may download a lot and take some time (it is downloading all Eclipse CDT for you)
Apache Maven 3.5.2 (138edd61fd100ec658bfa2d307c43b76940a5d7d; 2017-10-18T10:58:13+03:00)
Maven home: /usr/local/Cellar/maven/3.5.2/libexec
Java version: 1.8.0_31, vendor: Oracle Corporation
Java home: /Library/Java/JavaVirtualMachines/jdk1.8.0_31.jdk/Contents/Home/jre
Default locale: en_US, platform encoding: UTF-8
OS name: "mac os x", version: "10.13.2", arch: "x86_64", family: "mac"
[INFO] Scanning for projects...
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[INFO] Computing target platform for MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] Fetching p2.index from http://download.eclipse.org/nebula/snapshot/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/nebula/snapshot
[INFO] Adding repository http://download.eclipse.org/tools/cdt/releases/9.3
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.3/cdt9.3.3/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.3/cdt9.3.1/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.3/cdt9.3.0/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/launchbar/oxygen.0.1/ (0B at 0B/s)
[INFO] Fetching p2.index from http://gnuarmeclipse.sourceforge.net/updates/ (0B of 173B at 0B/s)
[INFO] Adding repository http://gnuarmeclipse.sourceforge.net/updates
[INFO] Resolving dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] {osgi.os=macosx, osgi.ws=cocoa, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: io.sloeber.feature.feature.group 4.2.0.qualifier
[ERROR]   Missing requirement: ilg.gnuarmeclipse.core 3.4.1.201704251808 requires 'bundle org.eclipse.ui.workbench 3.105.0' but it could not be found
[ERROR]   Cannot satisfy dependency: ilg.gnuarmeclipse.debug.gdbjtag.openocd 4.1.4.201704251808 depends on: bundle ilg.gnuarmeclipse.core 0.0.0
[ERROR]   Cannot satisfy dependency: ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.group 4.1.4.201704251808 depends on: ilg.gnuarmeclipse.debug.gdbjtag.openocd [4.1.4.201704251808]
[ERROR]   Cannot satisfy dependency: io.sloeber.feature.feature.group 4.2.0.qualifier depends on: ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.group 0.0.0
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException
Problem in build

jantje, 2018-01-14:
There seems to be a problem with gnuarmeclipse as I see the following on my build server

```
[INFO] Resolving dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/io.sloeber.feature/pom.xml
[WARNING] The following locally built units have been used to resolve project dependencies:
[WARNING]   ilg.gnuarmeclipse.debug.gdbjtag.openocd/4.1.4.201707110257
[WARNING]   ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.group/4.1.4.201707110257
[WARNING]   ilg.gnuarmeclipse.debug.core/1.2.1.201707110257
[WARNING]   ilg.gnuarmeclipse.debug.gdbjtag.restart/1.3.1.201707110257
[WARNING]   ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.jar/4.1.4.201707110257
[WARNING]   ilg.gnuarmeclipse.debug.gdbjtag/3.2.1.201707110257
[WARNING]   ilg.gnuarmeclipse.packs.core/1.1.3.201707110257
[WARNING]   ilg.gnuarmeclipse.core/3.4.1.201707110257
```

After a quick look it seems gnuarmeclipse has been renamed to gnu mcu eclipse.
The update site moved to http://gnu-mcu-eclipse.netlify.com/v4-neon-updates/
Changing the update site should do the trick.
Can you do this?

von13, 2018-01-14:
I've changed http://gnuarmeclipse.sourceforge.net/updates to http://gnu-mcu-eclipse.netlify.com/v4-neon-updates, but it didn't solve the issue.

BTW, I downloaded Sloeber r4.2 source and didn't observe any issue with build. So look like some some changes between 4.2 and current broke the build.

>[ERROR] Missing requirement: ilg.gnuarmeclipse.core 3.4.1.201704251808 requires 'bundle org.eclipse.ui.workbench 3.105.0' but it could not be found
As I understand, the issue is related to org.eclipse.ui.workbench, not to ilg.gnuarmeclipse.core. Looks like mvn can't download org.eclipse.ui.workbench 

$ git diff
diff --git a/io.sloeber.parent/pom.xml b/io.sloeber.parent/pom.xml
index 9c92b328..ccad92dc 100644
--- a/io.sloeber.parent/pom.xml
+++ b/io.sloeber.parent/pom.xml
@@ -483,7 +483,7 @@
                <repository>
                        <id>gnuarm</id>
                        <layout>p2</layout>
-                       <url>http://gnuarmeclipse.sourceforge.net/updates</url>
+                       <url>http://gnu-mcu-eclipse.netlify.com/v4-neon-updates</url>
                </repository>
        </repositories>
 
diff --git a/io.sloeber.product.sdk/arduino.product b/io.sloeber.product.sdk/arduino.product
index 9f0c4b23..0b4e0629 100644
--- a/io.sloeber.product.sdk/arduino.product
+++ b/io.sloeber.product.sdk/arduino.product
@@ -375,7 +375,7 @@ rights to a jury trial in any resulting litigation.&lt;/p&gt;
       <repository location="http://eclipse.baeyens.it/update/V4/nightly" enabled="false" />
       <repository location="http://eclipse.baeyens.it/update/V4/stable" enabled="true" />
       <repository location="http://download.eclipse.org/nebula/snapshot" enabled="true" />
-      <repository location="http://gnuarmeclipse.sourceforge.net/updates" enabled="true" />
+      <repository location="http://gnu-mcu-eclipse.netlify.com/v4-neon-updates" enabled="true" />
    </repositories>
 
    <preferencesInfo>
diff --git a/io.sloeber.product/arduino.product b/io.sloeber.product/arduino.product
index a355a688..17733b18 100644
--- a/io.sloeber.product/arduino.product
+++ b/io.sloeber.product/arduino.product
@@ -369,7 +369,7 @@ rights to a jury trial in any resulting litigation.&lt;/p&gt;
       <repository location="http://eclipse.baeyens.it/update/V4/nightly" enabled="false" />
       <repository location="http://eclipse.baeyens.it/update/V4/stable" enabled="true" />
       <repository location="http://download.eclipse.org/nebula/snapshot" enabled="true" />
-      <repository location="http://gnuarmeclipse.sourceforge.net/updates" enabled="true" />
+      <repository location="http://gnu-mcu-eclipse.netlify.com/v4-neon-updates" enabled="true" />^M
    </repositories>
 
    <preferencesInfo>
diff --git a/io.sloeber.updatesite/category.xml b/io.sloeber.updatesite/category.xml
index 6178445d..8d6976ff 100644
--- a/io.sloeber.updatesite/category.xml
+++ b/io.sloeber.updatesite/category.xml
@@ -54,5 +54,5 @@
    <repository-reference location="http://download.eclipse.org/nebula/snapshot" enabled="true" />
    <repository-reference location="http://download.eclipse.org/tools/cdt/releases/9.2" enabled="true" />
    <repository-reference location="http://download.eclipse.org/tools/cdt/releases/8.8.1" enabled="true" />
-   <repository-reference location="http://gnuarmeclipse.sourceforge.net/updates" enabled="true" />
+   <repository-reference location="http://gnu-mcu-eclipse.netlify.com/v4-neon-updates" enabled="true" />
 </site>

$ ./build_then_launch.sh 
Trying to build and then launch the Arduino Eclipse IDE
First we build with Maven. This may download a lot and take some time (it is downloading all Eclipse CDT for you)
Apache Maven 3.5.2 (138edd61fd100ec658bfa2d307c43b76940a5d7d; 2017-10-18T10:58:13+03:00)
Maven home: /usr/local/Cellar/maven/3.5.2/libexec
Java version: 1.8.0_31, vendor: Oracle Corporation
Java home: /Library/Java/JavaVirtualMachines/jdk1.8.0_31.jdk/Contents/Home/jre
Default locale: en_US, platform encoding: UTF-8
OS name: "mac os x", version: "10.13.2", arch: "x86_64", family: "mac"
[INFO] Scanning for projects...
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[... 6 lines cut ...]
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[INFO] Computing target platform for MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] Fetching p2.index from http://download.eclipse.org/nebula/snapshot/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/nebula/snapshot
[INFO] Adding repository http://download.eclipse.org/tools/cdt/releases/9.3
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.3/cdt9.3.3/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.3/cdt9.3.1/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.3/cdt9.3.0/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/launchbar/oxygen.0.1/ (0B at 0B/s)
[INFO] Fetching p2.index from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates/ (0B of 172B at 0B/s)
[INFO] Adding repository http://gnu-mcu-eclipse.netlify.com/v4-neon-updates
[... cut ...]
[ERROR] Cannot resolve dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

jantje, 2018-01-14:
>[ERROR] Missing requirement: io.sloeber.feature.feature.group 4.2.0.qualifier requires 'ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.group 0.0.0' but it could not be found

as I said gnuarmeclipse has been renamed to gnumcueclipse

von13, 2018-01-14:
Thanks for the explanation, but I can't understand how to fix the issue. Could you please help with the fix? What do I need to change?

jantje, 2018-01-14:
It is not my strongest point but there is a dependency on ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.group which no longer exists.
I assume the dependency has to be to ilg.gnumcueclipse.debug.gdbjtag.openocd.feature.group but that name may have changed as developers like to group incompatibilities.

I'm mostly working in the gui (eclipse) and not with maven. In eclipse I would say
1) deinstall gnuarm eclipse
2) remove gnuarm update site
3) add gnumcu update site
4) install gnumcu (full install is easiest)
5) fix the errors in eclipse
6) pray it will work in mvn

wimjongman, 2018-01-14:
You can run the build locally for any platform:

      mvn clean verify -Pmac64 -DskipTests=true

It builds fine on my machine.

jantje, 2018-01-14:
>It builds fine on my machine
@wimjongman but probably with the same warning as on the build machine.
gnuarmeclipse  changed name and update site. So some thkngs must change in the plugin/build script

wimjongman, 2018-01-14:
Nothing to do with that. The build needed an explicit profile after my SDK introduction (#886). I fixed it and also added the `skipTests `to the `build_then_launch.sh `

I'm not sure if this script is working on mac though. 

Anyway, building locally works again with:

    mvn clean verify -DskipTests

von13, 2018-01-15:
Hello @wimjongman 
Unfortunately it didn't help. I can't build from command line on my mac os x high sierra. version is git a53aafa25e77ebcbfba72163fbc332ae19eb0371

bash-3.2$ mvn clean verify -DskipTests
[INFO] Scanning for projects...
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[INFO] Computing target platform for MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] Fetching p2.index from http://download.eclipse.org/nebula/snapshot/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/nebula/snapshot
[INFO] Adding repository http://download.eclipse.org/tools/cdt/releases/9.3
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.3/cdt9.3.3/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.3/cdt9.3.1/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.3/cdt9.3.0/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/launchbar/oxygen.0.1/ (0B at 0B/s)
[INFO] Fetching p2.index from http://gnuarmeclipse.sourceforge.net/updates/ (0B of 173B at 0B/s)
[INFO] Adding repository http://gnuarmeclipse.sourceforge.net/updates
[INFO] Resolving dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] {osgi.os=macosx, osgi.ws=cocoa, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: io.sloeber.feature.feature.group 4.2.0.qualifier
[ERROR]   Missing requirement: ilg.gnuarmeclipse.core 3.4.1.201704251808 requires 'bundle org.eclipse.ui.workbench 3.105.0' but it could not be found
[ERROR]   Cannot satisfy dependency: ilg.gnuarmeclipse.debug.gdbjtag.openocd 4.1.4.201704251808 depends on: bundle ilg.gnuarmeclipse.core 0.0.0
[ERROR]   Cannot satisfy dependency: ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.group 4.1.4.201704251808 depends on: ilg.gnuarmeclipse.debug.gdbjtag.openocd [4.1.4.201704251808]
[ERROR]   Cannot satisfy dependency: io.sloeber.feature.feature.group 4.2.0.qualifier depends on: ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.group 0.0.0
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException
bash-3.2$ mvn clean verify -Pmac64 -DskipTests
[INFO] Scanning for projects...
[INFO] Computing target platform for MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] Fetching p2.index from http://download.eclipse.org/nebula/snapshot/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/nebula/snapshot
[INFO] Adding repository http://download.eclipse.org/tools/cdt/releases/9.3
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.3/cdt9.3.3/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.3/cdt9.3.1/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.3/cdt9.3.0/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/launchbar/oxygen.0.1/ (0B at 0B/s)
[INFO] Fetching p2.index from http://gnuarmeclipse.sourceforge.net/updates/ (0B of 173B at 0B/s)
[INFO] Adding repository http://gnuarmeclipse.sourceforge.net/updates
[INFO] Resolving dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] {osgi.os=macosx, osgi.ws=cocoa, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: io.sloeber.feature.feature.group 4.2.0.qualifier
[ERROR]   Missing requirement: ilg.gnuarmeclipse.core 3.4.1.201704251808 requires 'bundle org.eclipse.ui.workbench 3.105.0' but it could not be found
[... cut ...]
[ERROR] Cannot resolve dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException

jantje, 2018-01-16:
My build server fails now as well
`[WARNING] The following locally built units have been used to resolve project dependencies:
[WARNING]   ilg.gnuarmeclipse.debug.gdbjtag.openocd/4.1.4.201707110257
[WARNING]   ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.group/4.1.4.201707110257
[WARNING]   ilg.gnuarmeclipse.debug.core/1.2.1.201707110257
[WARNING]   ilg.gnuarmeclipse.debug.gdbjtag.restart/1.3.1.201707110257
[WARNING]   ilg.gnuarmeclipse.debug.gdbjtag.openocd.feature.jar/4.1.4.201707110257
[WARNING]   ilg.gnuarmeclipse.debug.gdbjtag/3.2.1.201707110257
[WARNING]   ilg.gnuarmeclipse.packs.core/1.1.3.201707110257
[WARNING]   ilg.gnuarmeclipse.core/3.4.1.201707110257`

`[ERROR] Failed to execute goal org.eclipse.tycho:tycho-packaging-plugin:0.24.0:package-plugin (default-package-plugin) on project io.sloeber.application: /share/homes/admin/.jenkins/jobs/arduino eclipse plugin build/workspace/io.sloeber.application/build.properties: bin.includes value(s) [plugin_customization.ini] do not match any files. -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MojoExecutionException
[ERROR] 
[ERROR] After correcting the problems, you can resume the build with the command
[ERROR]   mvn <goals> -rf :io.sloeber.application`

But this was me :-s

jantje, 2018-01-16:
I'm still getting a warning from the update site (Which I do not understand) but no more warnings about gnuarm
Here is the update site warning
`[INFO] --- tycho-p2-repository-plugin:0.24.0:assemble-repository (default-assemble-repository) @ io.sloeber.updatesite ---
[WARNING] Mirror tool: Problems resolving provisioning plan.: [Unable to satisfy dependency from 201801161003.Dependencies 1.0.0.8H7g5cLSvvE7kMZlQsT2XLz0i to org.eclipse.cdt.feature.group [9.3.2.201709131603].; Unable to satisfy dependency from 201801161003.Dependencies 1.0.0.8H7g5cLSvvE7kMZlQsT2XLz0i to org.eclipse.cdt.feature.group [9.4.0.201712020452].; Unable to satisfy dependency from 201801161003.Dependencies 1.0.0.8H7g5cLSvvE7kMZlQsT2XLz0i to org.eclipse.cdt.feature.group [9.4.0.201712051550].]
[WARNING] More information on the preceding warning(s) can be found here:
[WARNING] - http://wiki.eclipse.org/Tycho_Messages_Explained#Mirror_tool`

wimjongman, 2018-01-16:
See comments on commit

wimjongman, 2018-01-16:
@von13 you are not on the latest. Please pull or fetch/rebase.

jantje, 2018-01-16:
@wimjongman I hope this is what you wanted

von13, 2018-01-16:
Still doesn't work. git version is 4abf30b6ef32abd29b9dc2ad1e09737c95423dab

bash-3.2$ git clone https://github.com/Sloeber/arduino-eclipse-plugin.git
Cloning into 'arduino-eclipse-plugin'...
remote: Counting objects: 16736, done.
remote: Compressing objects: 100% (82/82), done.
remote: Total 16736 (delta 42), reused 92 (delta 22), pack-reused 16622
Receiving objects: 100% (16736/16736), 14.33 MiB | 882.00 KiB/s, done.
Resolving deltas: 100% (9239/9239), done.
bash-3.2$ cd arduino-eclipse-plugin/
bash-3.2$ ./build_then_launch.sh 
Trying to build and then launch the Arduino Eclipse IDE
First we build with Maven. This may download a lot and take some time (it is downloading all Eclipse CDT for you)
Apache Maven 3.5.2 (138edd61fd100ec658bfa2d307c43b76940a5d7d; 2017-10-18T10:58:13+03:00)
Maven home: /usr/local/Cellar/maven/3.5.2/libexec
Java version: 1.8.0_31, vendor: Oracle Corporation
Java home: /Library/Java/JavaVirtualMachines/jdk1.8.0_31.jdk/Contents/Home/jre
Default locale: en_US, platform encoding: UTF-8
OS name: "mac os x", version: "10.13.2", arch: "x86_64", family: "mac"
[INFO] Scanning for projects...
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[WARNING] No explicit target runtime environment configuration. Build is platform dependent.
[INFO] Computing target platform for MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] Fetching p2.index from http://download.eclipse.org/nebula/snapshot/ (0B at 0B/s)
[INFO] Adding repository http://download.eclipse.org/nebula/snapshot
[INFO] Adding repository http://download.eclipse.org/tools/cdt/releases/9.4
[INFO] Fetching compositeContent.xml from http://download.eclipse.org/tools/cdt/releases/9.4/ (0B at 0B/s)
[INFO] Fetching compositeContent.xml from http://download.eclipse.org/tools/cdt/launchbar/oxygen.2/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/launchbar/oxygen.2/launchbar-2.2.0/ (0B at 0B/s)
[INFO] Fetching content.xml.xz from http://download.eclipse.org/tools/cdt/launchbar/oxygen.2/launchbar-2.2.0/ (0B at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.4/cdt-9.4.0/ (0B at 0B/s)
[INFO] Fetching content.jar from http://download.eclipse.org/tools/cdt/releases/9.4/cdt-9.4.0/ (0B of 68.49kB at 0B/s)
[INFO] Fetching p2.index from http://download.eclipse.org/tools/cdt/releases/9.4/cdt-9.4.1/ (0B at 0B/s)
[INFO] Fetching content.jar from http://download.eclipse.org/tools/cdt/releases/9.4/cdt-9.4.1/ (0B of 70.8kB at 0B/s)
[INFO] Fetching p2.index from http://gnu-mcu-eclipse.netlify.com/v4-neon-updates/ (0B of 172B at 0B/s)
[INFO] Adding repository http://gnu-mcu-eclipse.netlify.com/v4-neon-updates
[INFO] Resolving dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml
[INFO] {osgi.os=macosx, osgi.ws=cocoa, org.eclipse.update.install.features=true, osgi.arch=x86_64}
[ERROR] Cannot resolve project dependencies:
[ERROR]   Software being installed: io.sloeber.feature.feature.group 4.2.0.qualifier
[ERROR]   Missing requirement: ilg.gnumcueclipse.core 4.3.1.201801092051 requires 'bundle org.eclipse.core.filesystem 1.6.1' but it could not be found
[ERROR]   Cannot satisfy dependency: ilg.gnumcueclipse.debug.gdbjtag.openocd 4.3.1.201801092051 depends on: bundle ilg.gnumcueclipse.core 0.0.0
[ERROR]   Cannot satisfy dependency: io.sloeber.feature.feature.group 4.2.0.qualifier depends on: ilg.gnumcueclipse.debug.gdbjtag.openocd 0.0.0
[ERROR] 
[ERROR] See http://wiki.eclipse.org/Tycho/Dependency_Resolution_Troubleshooting for help.
[ERROR] Cannot resolve dependencies of MavenProject: com.github.brodykenrick.arduino-eclipse-plugin:io.sloeber.feature:4.2.0-SNAPSHOT @ /Users/sergey/arduino-eclipse-plugin/io.sloeber.feature/pom.xml: See log for details -> [Help 1]
[ERROR] 
[ERROR] To see the full stack trace of the errors, re-run Maven with the -e switch.
[ERROR] Re-run Maven using the -X switch to enable full debug logging.
[ERROR] 
[ERROR] For more information about the errors and possible solutions, please read the following articles:
[ERROR] [Help 1] http://cwiki.apache.org/confluence/display/MAVEN/MavenExecutionException
Problem in build

wimjongman, 2018-01-16:
Please checkout ddbe36fc3929f34181d786cda171c38ae7b78def

@jantje has reopened this issue (while he should have started a new one) and broke the build. Maybe he wants to revert his commits? ;)

jantje, 2018-01-16:
@wimjongman 
I reopened because there was a problem with the gnuarm to gnumcu in the warning which needed to be fixed.

The new problem seems to be related to the fact that gnumcu seems to force oxygen.
The following command works fine: 
`mvn clean verify -DskipTests -Poxygen `

wimjongman, 2018-01-16:
Yes, I forgot to push my last commit.

Now you can really build with:

     mvn clean verify -DskipTests

.. without having to put in a profile.

jantje, 2018-01-17:
@wimjongman 
cool, thanks

@von13 does it work for you now?

closed 2018-01-18 by wimjongman

von13, 2018-01-20:
thx, now it works fine