Sloeber/arduino-eclipse-plugin #902: Linked "personal" Libraries are linked in, but are not found in the build phase other libraries in sloeber.app are!

Skammi, 2018-01-16:
Hi I try to switch from the Eclipse cdt plugin to sloeber, because sloeber seems to support 3rd party boards.
I have some issues with the use of (personal) libraries. Attached an example using one of my libraries. The library works perfect in Arduino IDE and the Eclipse cdt plugin. But not with sloeber.
The Library is found and visible although all links show a strike through sign.

I'm using a Mac with MacOS Version 10.13.2

Best regards,
Jacob

[SloeberPersonalLibraryError.pdf](https://github.com/Sloeber/arduino-eclipse-plugin/files/1636311/SloeberPersonalLibraryError.pdf)

jantje, 2018-01-16:
The utils lib you are using does not follow the library guidelines.
The folder 1.0.7 should not be there.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

Skammi, 2018-01-16:
HI Jan, Thanks for the quick response.
It didn't solve the issue, although the library no longer has a strike through sign. It still generates the build error. I used both include with "" and <>. No difference.

See attachements

[SloeberPersonalErrorV2-1.pdf](https://github.com/Sloeber/arduino-eclipse-plugin/files/1636413/SloeberPersonalErrorV2-1.pdf)
[SloeberPersonalErrorV2-2.pdf](https://github.com/Sloeber/arduino-eclipse-plugin/files/1636414/SloeberPersonalErrorV2-2.pdf)
[SloeberPersonalErrorV2-3.pdf](https://github.com/Sloeber/arduino-eclipse-plugin/files/1636415/SloeberPersonalErrorV2-3.pdf)

Skammi, 2018-01-16:
Try to reopen

Skammi, 2018-01-16:
Hi Jan,
I got a workaround. The Library is recognised when I move it in the Sloeber.app path.
See attached.

[SloeberPersonalLibWorksInSloeberAppPath.pdf](https://github.com/Sloeber/arduino-eclipse-plugin/files/1636448/SloeberPersonalLibWorksInSloeberAppPath.pdf)

Would be good to get the personal path fixed as I use the Library and source construction on several computers using Dropbox. 

Best regards,
Jacob.

jantje, 2018-01-16:
>Would be good to get the personal path fixed as I use the Library and source construction on several computers using Dropbox.

I would say: try again with a new project without the version number in the path from the start.
It is complicated but simply put following steps won't work
1) add library with path lib/version/src/include.
2) change lib path to lib/src/include.
where the following will
1) change lib path to lib/src/include.
2) add library with path lib/src/include.

Skammi, 2018-01-16:
Hi Jan,

Thanks again, just back from lunch.
That did the trick, I moved the lib. back to the personal library folder and created a new project using the lib..
The project is now compiled without errors.
See attached for completeness.

[SloeberPersonalLibraryOK.pdf](https://github.com/Sloeber/arduino-eclipse-plugin/files/1636660/SloeberPersonalLibraryOK.pdf)

Best regards
Jacob

closed 2018-01-16 by Skammi