Sloeber/arduino-eclipse-plugin #904: int verus int16_t

JupiterMoll, 2018-01-17:
I'm trying to execute this code in Teensy 3.6 as .ino:
```
#include "Arduino.h"

int searchList[200];
 
void setup() {

	Serial.begin(9600);
        while (!Serial) { ;}

        Serial.println(sizeof(searchList));
}

void loop() {}
```
I got as result a wrong Value:
`800`

But if i use int16_t i got the right value I'm expecting;

```
#include "Arduino.h"

int16_t searchList[200];
 
void setup() {

	Serial.begin(9600);
        while (!Serial) { ;}

        Serial.println(sizeof(searchList));
}

void loop() {}
```

Why?

uzi18, 2018-01-17:
What is Your result?

JupiterMoll, 2018-01-17:
Using `int searchList[200];` i got 800.
Using `int16_t searchList[200];` i got 400.

uzi18, 2018-01-17:
It's not Sloeber error for me,  try it on ArduinoIDE if same,  report it on
teensy support.

JupiterMoll, 2018-01-17:
By default when doing arithmetic operations, short/byte types are converted to int to do the operation. In the case of the AVR processors, int is 16-bits. In the case of ARM processors, int is 32-bits. So presumably the shift/or is converted to 32-bits and a sign-extension is not done. Doing the cast around the whole expression, will cause the compiler to convert the 32-bit value to 16-bits, and then convert it back to 32-bits via sign extension, which is what you want. I originally assumed in my first reply that the values were essentially unsigned 16-bit values.

Using int16_t, int32_t, etc. is just a short hand, as these types are synonyms for the underlying basic types. In the case of the AVR:

    uint8_t is unsigned char;
    int8_t is signed char;
    uint16_t is unsigned int;
    int16_t is int;
    uint32_t is unsigned long;
    int32_t is long.

In the case of the ARM:

    uint8_t is unsigned char;
    int8_t is signed char;
    uint16_t is unsigned short;
    int16_t is short;
    uint32_t is unsigned int;
    int32_t is int.

jantje, 2018-01-17:
This is standard C stuff not related to sloeber.
sizeof gives you the size of the object in bytes. int is not a fixed size object http://en.cppreference.com/w/cpp/language/types
a uno is a 8 bit cpu. As such C uses 16 bit for int so 2 bytes for each int so 2 * 200 = 400 in total
Teensy 3.6 is a 32 bitter. As such C uses 32 bit for int so 4 bytes for each int so 4 * 200 = 800 in total

This explains why there is such a thing like uint32_t and sizeof

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2018-01-17 by jantje

JupiterMoll, 2018-01-17:
I appreciate your work for this arduino plugin.....
I dediced to support your work right now on
 https://www.patreon.com/jantje