Sloeber/arduino-eclipse-plugin #905: "Check for updates" with nightly installed failes

liebman, 2018-01-19:
Error box is below available updates window:
[image]

Info:
[image]

jantje, 2018-01-20:
Hi @liebman 
First of all thanks for being a patron :-) It really is appreciated.
Yesterday I bumped into some upgrade issues myself and investigated I have not yet resolved the issue fully.
In your first image the details section is actually a multiline. So I can't be sure what it says, but I assume it complains about a dependency on gnumcueclipse and/or cdt version 9.4 for which no update sites are found.
The update sites are known by the Sloeber nightly update site but apparently not taken into account.
These update sites are
http://gnu-mcu-eclipse.netlify.com/v4-neon-updates
http://download.eclipse.org/tools/cdt/releases/9.4

Add these  2 update sites to your available update sites.

On my system it still didn't want to upgrade due to a dependency from sloeber.ide to (If I recall correctly 4.0 to a cdt version)
So I went for a install which told me I had to upgrade and it worked fine ... except for it is the latest nightly -1. I have not found why this is so.

Upgrade cdt https://github.com/Sloeber/arduino-eclipse-plugin/commit/4abf30b6ef32abd29b9dc2ad1e09737c95423dab
Upgrade gnuarm to gnumcu #898

liebman, 2018-01-20:
Details say: "no remedy found" that's why I did not expand it.

Oh - my install was done installing CDT first, then adding the plugin. It's the same install I've been using for months.

Adding http://gnu-mcu-eclipse.netlify.com/v4-neon-updates as an update site fixed the issue.

closed 2018-01-20 by liebman

jantje, 2018-01-20:
I actually meant the other details :-)
[image]