Sloeber/arduino-eclipse-plugin #906: Editing Thrid party  index  url's  give  error   ( Malformed url  no protocol:  )

miky2k, 2018-01-22:
After delete of a line and leave empty  in  Thrid party  index  url's   it give the error Malformed url  no protocol: .
I try to restore to default or empty all but error persist.

jantje, 2018-01-22:
I do not get this issue when restoring to default.
Malformed url no protocol means there is no http: in front of a url
Please provide more details like your list of url's
sloeber version

miky2k, 2018-01-22:
Sorry but seems a race condition when changing something(i remove and re-add esp8266 json to force 2.4 version download) and give apply,

pheraps network error or slow net appling json, 

did not see any activity indicator during json downloading  
but now i see esp8266 2.4 so it load updated json and did not give me any error.

I cannot reproduce.

closed 2018-01-22 by miky2k

jantje, 2018-01-22:
thanks for the feedback