Sloeber/arduino-eclipse-plugin #907: Serial upload speed don't match with board setting

EddyAtPessac, 2018-01-25:
Hello,
I just discovered Sloebert and I think it's great! How comfortable compared to the other IDE!
On the other hand, I have set the upload speed to 921600  (see the attached  file) but when esptool is launched, the load speed selected is 19200. Why ? 
The proposed speed are 9600, 57600, 115200 and 921600. There is no intermediate speed between 115200 and 921600 ? 

[image]

`\sloeber\/arduinoPlugin/packages/esp8266/tools/esptool/0.4.12/esptool.exe` -cd ck -cb 19200 -cp COM1 -ca 0x00000 -cf `

jantje, 2018-01-25:
This problem is seems not to be related to Sloeber but to esp V2.4.0.
It is fixed in 2.4.0-rc2

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2018-01-25:
Strange ... the upload command was fine for me and then it no longer is.
Is this fixed for you?

jantje, 2018-02-02:
The problem is related to the boards.txt having double entries for the uploadspeed
For instance generic board contains

Default setting
`generic.upload.speed=115200`

Menu settings
```
generic.menu.UploadSpeed.115200=115200
generic.menu.UploadSpeed.115200.upload.speed=115200
generic.menu.UploadSpeed.9600=9600
generic.menu.UploadSpeed.9600.upload.speed=9600
generic.menu.UploadSpeed.57600=57600
generic.menu.UploadSpeed.57600.upload.speed=57600
generic.menu.UploadSpeed.256000.windows=256000
generic.menu.UploadSpeed.256000.upload.speed=256000
generic.menu.UploadSpeed.230400.linux=230400
generic.menu.UploadSpeed.230400.macosx=230400
generic.menu.UploadSpeed.230400.upload.speed=230400
generic.menu.UploadSpeed.460800.linux=460800
generic.menu.UploadSpeed.460800.macosx=460800
generic.menu.UploadSpeed.460800.upload.speed=460800
generic.menu.UploadSpeed.512000.windows=512000
generic.menu.UploadSpeed.512000.upload.speed=512000
generic.menu.UploadSpeed.921600=921600
generic.menu.UploadSpeed.921600.upload.speed=921600
```

If you remove the default settings in boards.txt and reselect the board again in project properties the upload speed should be ok
Settings to be removed:
`[myboard].upload.speed=115200`

eXtatic-GitHub, 2018-02-02:
Hi jantje, thanks for linking the issues and your suggestions.
Meanwhile I tried some combinations but no success for me.

I tried:
1. [myboard].upload.speed=115200:  
only 115200 selectable but falls back to 19200 when flashing

2. [myboard].menu.UploadSpeed.115200.upload.speed=115200:  
upload speed not configurable in properties, falls back to 19200 when flashing

3. both entries:
only 115200 selectable but falls back to 19200 when flashing

4. several entries of [myboard].menu.UploadSpeed.<baudrate>.upload.speed=<baudrate>:  
upload speed not configurable in properties, falls back to 19200 when flashing

5. several entries of [myboard].upload.speed=<baudrate>:  
<baudrate> selectable but falls back to 19200 when flashing

jantje, 2018-02-02:
Simply remove the lines
[myboard].upload.speed=XXXXX

You need to make sure sloeber reconfigures the project otherwise your changes will not be taken into account.

eXtatic-GitHub, 2018-02-02:
Hmm, maybe we don't understand each other. I still think you're referring to my 2. case thus I put here some more details: (afaik you don't have an ESP8266 for verification)

**initial state:**  
generic.menu.UploadSpeed.115200=115200  
generic.menu.UploadSpeed.115200.upload.speed=115200  
generic.menu.UploadSpeed.9600=9600  
generic.menu.UploadSpeed.9600.upload.speed=9600  
generic.menu.UploadSpeed.57600=57600  
generic.menu.UploadSpeed.57600.upload.speed=57600  
generic.menu.UploadSpeed.230400.linux=230400  
generic.menu.UploadSpeed.230400.macosx=230400  
generic.menu.UploadSpeed.230400.upload.speed=230400  
generic.menu.UploadSpeed.256000.windows=256000  
generic.menu.UploadSpeed.256000.upload.speed=256000  
generic.menu.UploadSpeed.460800.linux=460800  
generic.menu.UploadSpeed.460800.macosx=460800  
generic.menu.UploadSpeed.460800.upload.speed=460800  
generic.menu.UploadSpeed.512000.windows=512000  
generic.menu.UploadSpeed.512000.upload.speed=512000  
generic.menu.UploadSpeed.921600=921600  
generic.menu.UploadSpeed.921600.upload.speed=921600

**changed to:**
generic.menu.UploadSpeed.115200.upload.speed=115200  
generic.menu.UploadSpeed.9600.upload.speed=9600  
generic.menu.UploadSpeed.57600.upload.speed=57600  
generic.menu.UploadSpeed.230400.upload.speed=230400  
generic.menu.UploadSpeed.256000.windows=256000  
generic.menu.UploadSpeed.256000.upload.speed=256000  
generic.menu.UploadSpeed.460800.upload.speed=460800  
generic.menu.UploadSpeed.512000.windows=512000  
generic.menu.UploadSpeed.512000.upload.speed=512000  
generic.menu.UploadSpeed.921600.upload.speed=921600  

**makes baudrate unavailable in proprties:**
[image]

**results in:**
[image]

jantje, 2018-02-02:
There is no menu in

> Simply remove the lines
> [myboard].upload.speed=XXXXX

eXtatic-GitHub, 2018-02-03:
OK, I'm close to give up, there is no line  
*[myboard].upload.speed=XXXXX*  
in *boards.txt*  
There isn't even one single line in *boards.txt* that contains the word *speed* but not the word *menu*.
Neither in the sloeber v4.2 delivery nor in the Arduino 1.8.5 one.

Maybe you can refer to the line number in *boards.txt*?

I'm willing to spend time on this but I do not know how Sloeber/Eclipse makes use of *boards.txt* and how this is processed to build the arguments for the flash process.

If someone else got it working, please let me know since it then seems to be a personal issue on my side.
@EddyAtPessac did you have success?

jantje, 2018-02-03:
looks like you are using 2.4.0 and not 2.4.0-rc2

eXtatic-GitHub, 2018-02-03:
Yes I do. [2.4.0.](https://github.com/esp8266/Arduino/releases/tag/2.4.0) is current release and newer than 2.4.0-rc2  
Nevertheless I did the exact opposite and **added** *generic.upload.speed=115200* to my 2.4.0 *boards.txt* and now it always flashes with that speed, regardless what is selected in the properties.

EddyAtPessac, 2018-02-05:
Yes, it is ok. I change my esp 2.4.0 to  2.4.0-rc2 and it work fine. 
But, it is difficult to change. I had a lot of compilation errors. 
Finaly, I deleted the  buid directory, reload the library,  rebuild, and it work. 
Thanks !

Kami-Rotschild, 2018-02-08:
I had the same problem with incorrect upload speed and used @eXtatic-GitHub 's solution, as my boards.txt was missing 'generic.upload.speed' amongst others.
Any idea, what that source of that problem might be or how to get working GUI settings back?

eXtatic-GitHub, 2018-02-08:
Hey there, I have another suggestion.

I was just having a look at the build environment settings to somehow understand what sloeber does there.
Then I saw that my `JANTJE.PLATFORM_FILE` was pointing to  
`${eclipse_home}/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/platform.txt`
eventhough I didn't even had 2.3.0 version installed anymore.

After changing this to point to the 2.4.0/platform.txt file I was able to change upload speed via menu again. I didn't even remove my previous fix to always use 115200.
[image]
[image]

While changing back and forth between platforms I saw that this variable is not auromatically change like the others.  
Here you'll see the build environment after changing from 2.4.0 back to 2.4.0-rc2
[image]  

`JANTJE.PLATFORM_FILE` still pointing to 2.4.0

I'm not sure if my installation is broken or if it is a general issue.
Maybe one can confirm?

Regards,
eXtatic

Kami-Rotschild, 2018-02-08:
Sorry to disappoint you, but I could not confirm your suggestion on that platform.txt file.

If I open up a new arduino project, I get fresh settings, and that platform.txt is missing at `properties->C/C++ Build` alltogether. Adding the entry does not help.

But I noticed, that, when switching between 2.3 and 2.4, the settings for `JANTJE_MENU` are changing from
`CpuFrequency=80
Debug=Serial
DebugLevel=None____
FlashFreq=40
FlashMode=dio
FlashSize=4M3M
ResetMethod=nodemcu
UploadSpeed=921600`
 to
`CpuFrequency=80
CrystalFreq=26
Debug=Serial
DebugLevel=None____
FlashFreq=40
FlashMode=dio
FlashSize=4M3M
LwIPVariant=Prebuilt
ResetMethod=nodemcu
UploadSpeed=921600.upload.speed`

Deleting the `.upload.speed` at the end is showing a positive result. Both versions of Boards.txt have the same entries.

Could someone please confirm this observation?

Kami-Rotschild, 2018-02-08:
@jantje is there any chance, that the issue above shares similarities to this
[image]?

jantje, 2018-02-08:
@Kami-Rotschild 
I don't understand your question

Kami-Rotschild, 2018-02-08:
I was stating the missing upload speed settings, as found in Boards.txt

`generic.menu.UploadSpeed.115200=115200
generic.menu.UploadSpeed.115200.upload.speed=115200
generic.menu.UploadSpeed.9600=9600
generic.menu.UploadSpeed.9600.upload.speed=9600
generic.menu.UploadSpeed.57600=57600
generic.menu.UploadSpeed.57600.upload.speed=57600
generic.menu.UploadSpeed.230400.linux=230400
generic.menu.UploadSpeed.230400.macosx=230400
generic.menu.UploadSpeed.230400.upload.speed=230400
generic.menu.UploadSpeed.256000.windows=256000
generic.menu.UploadSpeed.256000.upload.speed=256000
generic.menu.UploadSpeed.460800.linux=460800
generic.menu.UploadSpeed.460800.macosx=460800
generic.menu.UploadSpeed.460800.upload.speed=460800
generic.menu.UploadSpeed.512000.windows=512000
generic.menu.UploadSpeed.512000.upload.speed=512000
generic.menu.UploadSpeed.921600=921600
generic.menu.UploadSpeed.921600.upload.speed=921600`

Only those are visible, that do not have any OS specific values.

jantje, 2018-02-08:
I hadn't noticed this......
I'm simply stunned....
What is this good for?

Kami-Rotschild, 2018-02-08:
I assume that there are baud rate transfer differences between *nix and *dos based operating systems and hence this needs to be reflected in matched values.

I hoped, you could help me a bit further 😸

jantje, 2018-02-09:
I'll take a look but I'm currently not at home.

dhzl84, 2018-02-10:
Did some further testing.
My `JANTJE.PLATFORM_FILE` environment variable seems to be a relic from Sloeber 4.1.
Setting up new projects with all kinds of ESP packages (2.3.0, 2.4.0,2.4.0_rc2 and latest git) showed that `JANTJE.PLATFORM_FILE`  is no longer used in 4.2 so I guess we can forget about it. Since I did only test `57600`, which seems to always work for me, I was making wrong assumptions here.

Also I was not able to flash with 115200 (or 9600) with @Kami-Rotschild suggestions.

My obervations:
`A.UPLOAD.SPEED` is the upload speed passed to esptool.

Eventhough I change 
`JANTJE.MENU.UPLOADSPEED` to `115200` and  
`JANTJE.MENU` to `CpuFrequency=80
CrystalFreq=26
Debug=Disabled
DebugLevel=None____
FlashFreq=40
FlashMode=qio
FlashSize=1M64
LwIPVariant=v2mss536
ResetMethod=ck
UploadSpeed=115200`
`A.UPLOAD.SPEED` defaults back to `19200` and esptool is using this.

There is just one exception for me, `57600` works straight away.
If I add `generic.upload.speed=115200` to *boards.txt* this seems to set the default upload speed if anything unforseen happens. As a consequence I'm able to flash with `57600` if selected in the properties and `115200` if I select anything else.

So the main question for me is, how is `A.UPLOAD.SPEED` build?

Kami-Rotschild, 2018-02-11:
Yes, my suggestion was a shot in the dark. The `JANTJE.MENU` change I suggested simply ends in an error due to missing parameters.
But on the other hand, you confirmed another thought that I had: 57600 Baud is the only one working in the menu. Let me guess: its the first in your drop down menu, too, @dhzl84 ?

About `A.UPLOAD.SPEED`: that one is working just fine with any manual setting. It even overwrites the `generic.upload.speed` at `boards.txt`. Thanks for the hint with that, @dhzl84 ! I can live with that for now.

jantje, 2018-02-12:
I found and fixed a but in the conversion of menu names to menu id's that caused the orinibal issue.

dhzl84, 2018-02-13:
Great to hear, thank you @jantje

closed 2018-02-16 by jantje

dhzl84, 2018-02-17:
@Kami-Rotschild: confirmed 57600 is the first entry also for me

Kami-Rotschild, 2018-02-17:
Thanks for solving the issue @jantje

hreintke, 2018-04-15:
Hi,

Now I also have the issue with upload speed.
Am I correct that it is not yet solved in a 4.x version but is in the nightly ?

If so, Can you point me to the location/link where I can download that ?
The https://oss.sonatype.org/content/repositories/snapshots/com/github/brodykenrick/arduino-eclipse-plugin/io.sloeber.product/4.2.0-SNAPSHOT/ link on the Sloeber website points to a location only containing builds from januari.

jantje, 2018-04-15:
@hreintke There was indeed a build engine issue. It it fixed now. For this type f issue a new issue should be made.
The fix is of feb so the latest nightly did not contain the fix.

hreintke, 2018-04-15:
Like to check first before creating a new issue.

Should it be solved by https://github.com/Sloeber/arduino-eclipse-plugin/commit/ae4e878d91305ff4ff3112ad7cac6861d779fa30? 

If so, I will build from git.

jantje, 2018-04-15:
no need the nightly from yesterday is available now at the download location

hreintke, 2018-04-17:
Thanks,

This worked for me.
See #963 for remark on upgrade