Sloeber/arduino-eclipse-plugin #909: Can't select platforms and boards

sk8nfool, 2018-01-29:
I just installed Sloeber V4.2 64 bit on a brand new Ubuntu 17.10 system with Java 9. I get a boatload of errors. I can't select any of the packages in Arduino->preferences->Arduino->Platforms and Boards.
[SLOEBER.TXT](https://github.com/Sloeber/arduino-eclipse-plugin/files/1674375/SLOEBER.TXT)

jantje, 2018-01-29:
It looks as if you do not have network access

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

sk8nfool, 2018-01-29:
How can that be? I just watched it download a bunch of stuff from the net
on first start. Firefox works. I'm confused.

jantje, 2018-01-29:
I don't know. These errors do not sound sloeberish to me.
I have not seen similar errors.
The first Sloeber logging is "arduinoPlugin folders created"
Which is at the very startup of the plugin
https://github.com/Sloeber/arduino-eclipse-plugin/blob/master/io.sloeber.core/src/io/sloeber/core/Activator.java#L89

Internet access is organized by application. Maybe sloeber is not allowed.
Is there content in [eclipseinstall]/arduinoPlugin/downloads?
If you can you open the files?

sk8nfool, 2018-02-01:
> master/io.sloeber.core/src/io/sloeber/core/Activator.java#L89

First question - who is organizing Internet access? I am running V4.2 on
Ubuntu 16.04 64 bit. I've never heard of such a thing.

Second - There are files in [eclipseinstall]/arduinoPlugin/downloads (17
actually) and they are all readable (at least md5sum can read them and
unzip -t says everything is OK).

sk8nfool, 2018-02-01:
Java crashed while I wasn't looking - tried to send crash report but I
don't know what happened.

Fired sloeber up again and got the following

$ sloeber-ide
arduinoPlugin folders created
log4j:WARN No appenders could be found for logger
(javax.jmdns.impl.JmDNSImpl).
log4j:WARN Please initialize the log4j system properly.

Not being a Java programmer I don't know what how to "initialize the log4j
system properly".

Still can't select any board/platforms yet. Note - I haven't started to
create any sketches yet.

Ok - I think I found the problem - It's sort of an Eclipse Tradition to not
guide a newbie like me. Apparently I was using the wrong Platform. I had
selected the .../arduino/hardware/avr/1.6.20 platform folder because that
was the only one defined in the drop-down list. This was even after I'd
"applied" the addition of
https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
to the 3-rd party index url list.

In particular I did the following

   1. unpacked V4.2_linux64.2017-11-20_02-16-52.tar.gz into
   $HOME/.local/share/sloeber/V4.2_linux64.2017-11-20_02-16-52
   2. Created a symbolic link in $HOME/bin to
   $HOME/.local/share/sloeber/V4.2_linux64.2017-11-20_02-16-52/sloeber-ide
   3. Started sloeber with the symbolic link
   4. launched the default workspace
   5. It said "arduinoPlugin folders created" in the terminal window where
   I started it.
   6. It complained about a couple of indexer options - I said OK
   7. It presented a Welcome window and started to "finishing the
   installation" by downloading a bunch of stuff and processing it - I let
   that complete. I think sometime during this it said

log4j:WARN No appenders could be found for logger
(javax.jmdns.impl.JmDNSImpl).
log4j:WARN Please initialize the log4j system properly

   1. I entered Window->Preferences and fixed the two indexer options it
   complained about and poked "Apply and close
   2. I opened Arduino->Preferences->Platforms and Boards and noticed that
   Adafruit was not there.
   3. I added
   https://adafruit.github.io/arduino-board-index/package_adafruit_index.json
   to the Third party index url;s list
   4. I noticed that this did nothing to the Platforms and Boards list so I
   poked Apply and close
   5. Now I see Adafruit in the Platforms and Boards list but still connot
   select Adafruit.

At this point I seem to have run into the Eclipse Tradition of confusing
new users. I poked around a LOT and finally discovered that I had to expand
the Adafruit list and the Adafruit AVR boards list within to see a list of
- what - versions? It doesn't say so in ignorance I picked the last
(presumably lates) one.

Now - the reason this was a problem was that all the levels above that
version box had selection boxes next to them that didn't work. Why? BECAUSE
THEY WEREN'T SELECTION BOXES!!!! WTF! If they're not selection boxes DON'T
MAKE THEM LOOK LIKE SELECTION BOXES! Make them visible/invisible
something-or-others (like a check mark or X or asterisk or something).

Dave

Oh - BTW - This is the second time I've stumbled over this. Really
annoying. BUT once past this I really like this addition to the Arduino
tool set.

jantje, 2018-02-02:
>log4j:WARN No appenders could be found for logger (javax.jmdns.impl.JmDNSImpl). log4j:WARN Please initialize the log4j system properly.

That warning I get myself all the time. It's nothing to be worried about.

>At this point I seem to have run into the Eclipse Tradition of confusing new users. 

I guess you are not aware there are a lot of video's around for this project
https://www.youtube.com/playlist?list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF

If I understand you correctly your problem is fixed and this issue can be closed.

sk8nfool, 2018-02-02:
I found a work-around - Ignore the check boxes - they're bogus and don't
work the way check boxes are supposed to work. If that is a solution then
yes you can close this issue.

jantje, 2018-02-02:
adapting is indeed the solution.

closed 2018-02-02 by jantje