Sloeber/arduino-eclipse-plugin #910: Refactoring across (.ino) files does not work

Dlrk, 2018-01-29:
Hi,

it seems to me that when I do a right klick on a function-name and select [Refactor] -> [Rename...], it only renames the function (or variable) in that particular file (and in the sloeber.ino.cpp), not in any of the other .ino files that function (or variable) is used.

In the [eclipse Help](http://help.eclipse.org/oxygen/index.jsp) under " Rename Refactoring" it states:
> [Rename...] Renames selected object (variable, method, etc...) and propagates changes **to other files** in project.

It does not matter if I rename a function in the [projectname].ino or any other .ino file. (I can't think of anything else to try to narrow it down more, sorry.)
I would be happy to test if anyone has a suggestion.
I am using the Nightly on debian, Linux.

All the best,
Dirk

jantje, 2018-01-29:
I tried it here. Seems to work.
I tried with function names and variables both with files that have pending changes and saved files.
It all seems to work for me.

wimjongman, 2018-01-30:
Where do you start the rename from. Can you supply some screenshots?

Dlrk, 2018-01-30:
thanks a lot for getting back to me.
It seems to only happen when I start from the function declaration. It does work when starting from a function call.

**from declaration:**
[image]

**from function call:**
[image]

jantje, 2018-01-30:
I don't get the dialog box and it works in both cases.
See video here https://youtu.be/1F0bitbYyOk

Dlrk, 2018-01-30:
Thanks for checking and going through the hassle to even record a video.

I used the preview in the dialog box because it is very clear which files will be altered. It can be accessed by klicking the little arrow in the yellow box that appears after you clicked [Rename...] or by selecting the function and typing in [Shift+Alt+r] **twice**.
[image]

I have downloaded a fresh nightly and tried it again with the same result, it doesn't work from the implementation. Maybe I'm missing something!?!
I thought it might have something to do with the Alphabetic order of the files bit renaming the file to 01_file.ino changed nothing.
If the function is called in the same file of the implementation, that function called is renamed...

You include a demo.h which I don't have, could that be a problem?

Theremingenieur, 2018-01-30:
If I read the Arduino documentation and development standards well (was several years ago, I admit), there should be only one .ino file per project, other included files in the "sketchbook" (as they call it instead of "project") should have .c, .cpp, or .h extensions. I'm not sure if the fact of having multiple .ino files contributes to the problem. I've never encountered refactoring problems with Sloeber nightly under macOS.

jantje, 2018-01-30:
One ino file used to be a constraint. Arduino ide supports multiple ino files for a long time now.
Again ino is not my preferred choice but Arduino ide supports it so did I.

Indexing, refactoring are pure cdt functionality. Sloeber tells cdt .ino files are source files. So I would not expect any differences between .ino files and .cpp files. I also never spotted a difference.
So I think op has some special case of a bug in something cdt related. If we find it I can fix it and contribute it to cdt.

Dlrk, 2018-01-30:
ok, I tried a few things. (the "header index" in the preferencec and renaming the .ino files to .cpp, which did nothing)

 It does work if I include the .ino file in the other files. But then it won't compile any more because of:
../AA_otherFile.ino:1:6: error: redefinition of 'void otherFunction()'
[image]

**The fix:**
It turns out that "Exaustive (slow) file search:" must be enabled in the renaming dialog which was not enabled in either of my installations. Now it works fine (even without the includes).
[image]

Thanks a lot for your help!
Please tell me if I can test anything else to help that others will not face that issue.

jantje, 2018-01-30:
This option is not on here
[image]

jantje, 2018-01-30:
How are your cdt indexer setting set?

Dlrk, 2018-01-30:
[image]

jantje, 2018-01-30:
looks ok

Dlrk, 2018-01-30:
Are there settings somewhere that all eclipse instances share? (that I might have changed before in a different instance)
I searched my hard drive but I did not find a .config file or folder.

jantje, 2018-01-30:
>Are there settings somewhere that all eclipse instances share? (that I might have changed before in a different instance)

There is oomph but I don't think that is at play here as oomph changes are visible in the gui like normal changes.

>If you "don't care" because the problem only affects me, that would be totally fine! :)

The thing is... I'm a quality man so I want to know what is going on. I think I'll test with the exact same code as you have. But I'm a bit short of time in the next weeks.

jantje, 2018-02-16:
I tried here with your code and I get "strange results" 
So I assume this is a special case where the refactoring fails

Dlrk, 2018-02-16:
Hey, thanks a lot for getting back to this.
That is wired. I watched your video again but that is totally not what I get... 

For me it is a perfectly fine workaround to enable "Exaustive (slow) file search:" so personally I would not mind if this issue would be closed.
In case you "want" to investigate this further please tell me if I can help!

jantje, 2018-02-16:
I have played around a bit more. The same behaviour exists when using plain cpp files. Also the "Open call hierarchy" does not find the called methods. Which is consistent with the refactoring behaviour.
As you found the "Exhaustive (slow) file search:" makes refactoring work as expected. However the call hierarchy" still misses the link.

I think the indexer behaves this way because in C/C++ normal behaviour is "first define then use".
The arduino Ino behaviour goes against this rule (by extracting the methods and appending then like Sloeber does in sloeber.ino.cpp) and so do/did other c/c++ based tools which caused the implementation of  "Exhaustive (slow) file search:" 

It may be a idea to split sloeber.ino.cpp in sloeber.ino.h and sloeber.ino.cpp so sloeber.ino.h can be included in the .ino files.
However this works against arduino ide compatibility and using .cpp files is my advised way to go. 

So I'm closing this as a special case which requires a special solution.

closed 2018-02-16 by jantje