Sloeber/arduino-eclipse-plugin #911: Third Party index url's : cannot undo an error once the error is made

ghislainborremans, 2018-01-31:
I added a reference to a local Json file as "D:\Electronica\...."
When you push "Apply" the message  "Malformed url D:\Electronica\Libraries\hardware\package_jantje_index.json
unknown protocol: d"  appears.  Even when pressing "Restore Defaults" you keep getting the same message. 
I tried to edit the configuration file "io.sloeber.arduino.prefs" and discarded the reference but the error remains. It must be kept else where but i do not know where.

ghislainborremans, 2018-01-31:
I reinstalled sloeber and that solved the problem.

jantje, 2018-01-31:
The newest version includes package_jantje_index.json
File references should be given as file://D:\Electronica\Libraries\hardware\package_jantje_index.json
Clicking defaults should fix it. I'll try to reproduce.

jantje, 2018-01-31:
It is very reproducible.
Needs to be fixed

jantje, 2018-02-12:
Now available with the nightly and fixed

closed 2018-02-12 by jantje

lewiss66, 2018-04-05:
On version 4.2 It's still a problem.
I miss copy a link in the third party index url's:
(c:\user.....) instead of (https://raw.githubu....) and then I can't get rid of the error even though I supress the link. I click apply and I gives an error:
"Malformed url C:\Users\gilles\git\ds3231
unknown protocol: c"

jantje, 2018-04-05:
>On version 4.2 It's still a problem.

indeed it is fixed in the nightly only for now.