Sloeber/arduino-eclipse-plugin #912: Maven build no longer works

jantje, 2018-01-31:
Only root and parent are build.
As the last nightly on the nightly site is from 16/01/2018 I guess this happened on the 15th jan
[image]

jantje, 2018-02-01:
apparently I need to add NOSDK or SDK
Still doesn't work but I'm back to gnumcu repository problems

jantje, 2018-02-01:
@wimjongman 
Can you have a look. Something must have gone wrong with https://github.com/Sloeber/arduino-eclipse-plugin/commit/d299833490c46d7f932d4e273f444212eeaf49b9
`mvn clean verify -DskipTests`
no longer  works.
It is very similar to the oxygen error I got but oxygen is no longer a solution. https://github.com/Sloeber/arduino-eclipse-plugin/issues/898#issuecomment-358063250

I also get another error when using 
`mvn clean -Pwin64,SDK -Dtest=RegressionTest verify`

 

> Missing requirement: io.sloeber.product 4.2.0.qualifier requires 'ilg.gnumcueclipse.debug.gdbjtag.openocd.feature.feature.group [4.3.1.201801092051]' but it could not be found

versus

>  Missing requirement: ilg.gnumcueclipse.core 4.3.2.201801250917 requires 'bundle org.eclipse.core.filesystem 1.6.1' but it could not be found

jantje, 2018-02-06:
@wimjongman can you please take a look?
The update site is still from 16 jan due to this issue

wimjongman, 2018-02-08:
Sorry I was busy. Slack does not notify me on mentions anymore. I will take a look,

closed 2018-02-08 by wimjongman

jantje, 2018-02-09:
Thanks for fixing :-)
No worries. If it was urgent I would have been more proactive.
I can add you to the mailinglist of failed builds of my jenkins server :-)

wimjongman, 2018-02-09:
Sure, must be a pretty lonely list :)