Sloeber/arduino-eclipse-plugin #915: Failed to run the Upload recipe

JBerg60, 2018-02-12:
I am using 4.2 on a Mac, added some boards from `http://dan.drown.org/stm32duino/package_STM32duino_index.json`

Then building the blink sketch for a Maple Mini gives `Failed to run the Upload recipe `

Console shows
`Launching/Applications/sloeber.app/Contents/Eclipse//arduinoPlugin/packages/stm32duino/tools/stm32tools/2018.2.11/win/maple_upload`

Obviousy the win folder does not exists on my Mac, the maple_upload is in the macosx folder

How to fix this?

jantje, 2018-02-12:
Are you sure the package supports mac?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2018-03-04:
closing due to no input

closed 2018-03-04 by jantje