Sloeber/arduino-eclipse-plugin #917: possibly invalid hex file

major-kellogs, 2018-02-14:
Hello,

I tried to upload a sketch to an Arduino Uno (original) and an Arduino Nano (clone). The output from avrdude seems to be ok (except the written bytes), but neither the Uno nor the Nano starts working. Compiling and uploading the sketch by Arduino IDE works. So I tried to take the command of avrdude (including params and options) from sloeber and used the .hex from Arduin IDE ... it works.

I'm using:
* OS: Linux 4.4.0
* Sloeber: 4.2.0.201711210114
  * Platform: 1.6.11

Here is the output of avrdude (Eclipse/Sloeber - .hex):
```
Starting upload
using arduino loader
Starting reset using DTR toggle process
Toggling DTR
Continuing to use/dev/ttyACM0
Ending reset

Launching/home/robd/.eclipse-sloeber/sloeber//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C/home/robd/.eclipse-sloeber/sloeber//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega328p -carduino -P/dev/ttyACM0 -b115200 -D -Uflash:w:/home/robd/workspace-sloeber/Nano_DGPS/Release/Nano_DGPS.hex:i 
Output:

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "/home/robd/workspace-sloeber/Nano_DGPS/Release/Nano_DGPS.hex"
avrdude: writing flash (444 bytes):

Writing | ################################################## | 100% 0.08s

avrdude: 444 bytes of flash written
avrdude: verifying flash memory against /home/robd/workspace-sloeber/Nano_DGPS/Release/Nano_DGPS.hex:
avrdude: load data flash data from input file /home/robd/workspace-sloeber/Nano_DGPS/Release/Nano_DGPS.hex:
avrdude: input file /home/robd/workspace-sloeber/Nano_DGPS/Release/Nano_DGPS.hex contains 444 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.07s

avrdude: verifying ...
avrdude: 444 bytes of flash verified

avrdude done.  Thank you.

avrdude finished
upload done
```

Here is the output of avrdude (using the .hex from Arduino IDE):
```
./avrdude -C/home/robd/.eclipse-sloeber/sloeber//arduinoPlugin/packages/arduino/tools/avrdude/6.3.0-arduino9/etc/avrdude.conf -patmega328p -carduino -P/dev/ttyACM0 -b115200 -D -Uflash:w:/tmp/build960fa50d6a7307f3d9c0063f8f16454f.tmp/sketch_feb14a.ino.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: reading input file "/tmp/build960fa50d6a7307f3d9c0063f8f16454f.tmp/sketch_feb14a.ino.hex"
avrdude: writing flash (1434 bytes):

Writing | ################################################## | 100% 0.25s

avrdude: 1434 bytes of flash written
avrdude: verifying flash memory against /tmp/build960fa50d6a7307f3d9c0063f8f16454f.tmp/sketch_feb14a.ino.hex:
avrdude: load data flash data from input file /tmp/build960fa50d6a7307f3d9c0063f8f16454f.tmp/sketch_feb14a.ino.hex:
avrdude: input file /tmp/build960fa50d6a7307f3d9c0063f8f16454f.tmp/sketch_feb14a.ino.hex contains 1434 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.20s

avrdude: verifying ...
avrdude: 1434 bytes of flash verified

avrdude: safemode: Fuses OK (E:00, H:00, L:00)

avrdude done.  Thank you.
```

Further more everything works fine with an Arduino Due. So I don't know what exactly is going wrong. Any advices?

Thanks

jantje, 2018-02-16:
The Sloeber hex file seems lots smaller than the Arduino IDE hex files.
It sure can not be the same source code

major-kellogs, 2018-02-16:
Well .. it is. It's just a small code for testing:

```C++
void setup() {
  // put your setup code here, to run once:
Serial.begin(9600);
}

void loop() {
  // put your main code here, to run repeatedly:
Serial.println("Es Lebt!");
}
```

jantje, 2018-02-16:
I build your sketch here on the arduino avr 1.6.11 platform and the file is bigger than 4K.
Something is/went wrong during the build.
Please check the build console ouput

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2018-03-04 by jantje