Sloeber/arduino-eclipse-plugin #918: Arduino Libraries "FILTERED" when added to project

mcgovernr, 2018-02-16:
1. This add-in is absolutely amazing. 
2. I recently found a bug when adding external libraries to my project. Using the Arduino library download GUI, I downloaded a number of libraries to "arduinoPlugin/libraries". I then added them to my project using the Arduino gui command. A filter was created to with the directories.

jantje, 2018-02-16:
It is really unclear to me what you are doing.

mcgovernr, 2018-02-23:
Forgive my previous post, it was lacking details. Essentially, when I loaded an Arduino library through the Sloeber Arduino gui, a corresponding filter was also created. This filter was under _Paths and Symbols_ in on the Source Location tab. The filter had exclusion paths for the library *.h and *.cpp files that I had added. I removed those two filters and it compiled. Not sure if it was a bug or not, but it was interesting how it created this filter. I'm assuming the filter is there to filter files not needed for compile, such as examples, readme, etc.  

Nonetheless, the solution was to remove the filters and it's working fine now.

uzi18, 2018-02-23:
please attach here problematic library

mcgovernr, 2018-02-23:
It was either the Adafruit_GFX_Library and/or the Adafruit_SSD1306.

jantje, 2018-03-04:
@mcgovernr 
As I still do not understand the issue I'm closing it.
Please reopen with a clear issue statement if you disagree.
Best regards
Jantje

closed 2018-03-04 by jantje