Sloeber/arduino-eclipse-plugin #919: Plotter doesn't work (Eclipse v. 4.7.2, Sloeber v. 4.2.0.201711210114)

flo2k, 2018-02-18:
Hello,

I'm just new to the Sloeber Eclipse plugin. As far is I saw, everything besides the Plotter works fine. I can program and use the "Serial monitor view" to see the serial output of my Arduino Ethernet.

[image]

In my test, I just write a boolean value to the serial output. With the Arduino IDE (v. 1.8.5) and the internal Plotter my data can be visualised - therefore the output should be fine.

[image]

But within the eclipse and the Sloeber Plotter it just shows a black area with a strange yellow line (see attached Screenshot).

[image]

I use:
- Eclipse v. 4.7.2
- Sloeber v. 4.2.0.201711210114
- Linux Mint 18.1 Serena
- Kernel v. 4.8.0-46

Are some dependencies missing on my system or is something not correctly configured?  

Best,

Florian

jantje, 2018-02-18:
Hi Florian
Thanks for the nice words
The plotter in sloeber works different from the plotter in Arduino IDE.
Use a simplot example to see how it works.
And do not forget this is all unpaid volunteer work. Support us at https://www.patreon.com/jantje

flo2k, 2018-02-18:
Hello jantje,

thank you for this quick response. Now I understand how it works. I have to include simplot and use the provided plot functions:

```c++
#include "simplot.h"
...
plot2(Serial, values[i], counters[i]);
}
```

Unfortunately it does not satisfy my requirements. Especially for small values, I can not zoom enough into the data:

[image]

As a comparison, the Arduino IDE Plotter:

[image]

That means for me that I will use the Arduino IDE Plotter for this analysis.

Best,

Florian

closed 2018-02-18 by flo2k

jantje, 2018-02-18:
you can zoom with the mouse drag and drop left and right. one is to zoom other to move the axis.
Sloeber does not allow a zoom bigger than 1=1pixel.
You can also zoom in software
`plot2(Serial, values[i]255*, 255*counters[i]);`
But choose is a good thing so feel free to use whatever suits your needs the most