Sloeber/arduino-eclipse-plugin #920: Win 10 Unable to get sloeber working in newest version

billiho, 2018-02-19:
This new version gave me a warning that the directory paths were too long when it was in my downloads directory, so I moved it to ../users/bill/sloeber/.  My Arduino ide works fine as doe the sketches. Sloeber gives me a warning that says my arduino sketches aren't sketches?

[image]

It starts super slow and says it can't finish the install. Tried updating to latest Java, tried deleting sloeber directory ( I think that's everything?), then downloading and reinstalling back to ../users/bill. Fired it up and after several minutes it comes up with the following error
[image]

When I try to run the software update I get this message.

[image]

So basically I can't use it at all. Is everything in 1 directory, and if how can I delete all references of it and try a reinstall?

jantje, 2018-02-19:
Did you install the product or the plugin?
Have you ever used sloeber (or eclipse) before?
Did you allow Sloeber to get on the internet?

One of the nice things in eclipse is everything is in one folder so simply deleting the eclipse folder erases everything from your system.

billiho, 2018-02-19:
Hi,

Thanks for your response.

I got the package from
http://eclipse.baeyens.it/

64 bit bundle. unzipped and moved sloeber directory to
C:\Users\Bill\

Just deleted the jre directory. Still having Java errors.

Workspace location
C:\Users\Bill\Documents\sloeber-workspace

Fires up as with welcome screen that says Eclipse Oxygen 2, December 2017.
Takes about 5-10 minutes to load on i7-8gb. Lots of errors when loaded.

I've deleted and installed about 3-4 times without success. Fires up but
not working, won't upload, lots of errors. I think it's trying to finalize
the install but bombing out, possibly a java issue?

Tried all the updating several times.

Looks like eclipse makes a few directories under ..users/bill/
.eclipse
.jubala
.p2
.tooling

Also regardless of if I open an existing .ino or create a new sketch it
says it's not an arduino file when I try to upload.

Unsure what to do at this point. Tried Sloeber before last year and I think
it worked fine, but haven't touched arduino stuff in the last year, so I'm
a bit rusty.

Bill Harding
Bizcom Telecom
4754 Elk Road
Victoria, BC V9E 2E6
www.bizcom.ca

250-479-8585
1-877-479-8585

jantje, 2018-02-19:
I just tried here and it works fine here

>Fires up as with welcome screen that says Eclipse Oxygen 2, December 2017.

I get
[image]

Again it looks like Sloeber can not access the internet.
What is in the [eclipse install]/arduinoPlugin/ and [eclipse install]/arduinoPlugin/downloads folder

billiho, 2018-02-19:
See attached screenshot. I gather when you mean [eclipse install] you mean
the main directory that everything's in, which is c:/users/sloeber. BTW I
am happy to Patreon if that's an incentive to help(wink,wink, nod,nod:)
[image: Inline image 1]

Bill Harding
Bizcom Telecom
4754 Elk Road
Victoria, BC V9E 2E6
www.bizcom.ca

250-479-8585
1-877-479-8585

jantje, 2018-02-19:
>[eclipse install] you mean the main directory that everything's in,

yes

>See attached screenshot. 

Something went wrong no screenshot is visible here

>BTW I am happy to Patreon if that's an incentive to help(wink,wink, nod,nod:)

Are you suggesting me stopping to help you untill you are a patreon?

billiho, 2018-02-19:
Heck no!

Anyway, here's a zip file with pics for both the arduinoplugins directory
and the downloads directory. They are populated with files.

Bill Harding
Bizcom Telecom
4754 Elk Road
Victoria, BC V9E 2E6
www.bizcom.ca

250-479-8585
1-877-479-8585

billiho, 2018-02-20:
> 250-479-8585 <(250)%20479-8585>
> 1-877-479-8585 <(877)%20479-8585>

Hi,

I was able to get it working better. I deleted everything again including
all the .blah files in the users directory and redownloaded, then copied it
into c:\users\bill\sloeber again, and started it but this time i didn't do
any updates. It doesn't give me the oxygen 2 splash screen, I get the
sloeber screen. I think it was installing the full eclipse ide and then was
bombing out on some part of the eclipse setup.

I am now trying to figure out how to set the upload settings as I was able
to up[load a file but I had issues. I dumped in an arduino build and will
try again tomorrow.

Thanks,

Bill Harding
Bizcom Telecom
4754 Elk Road
Victoria, BC V9E 2E6
www.bizcom.ca

250-479-8585
1-877-479-8585

jantje, 2018-02-20:
@billiho 
I guess you are not aware you are responding via forum software and as such exposing your address and telephone number to the whole world.

I tried the upgrade scenario but I can not reproduce the behaviour your are experiencing.
Did you install the nightly or the stable?
Did you watch some of my video's?
Can you make a video on how you install and startup?

billiho, 2018-02-20:
haha, Thanks, Wasn't aware of that but I am now!

This message comes up when I just fired up Sloeber.

Unable to download http://arduino.esp8266.com/stable/package_esp8266com_index.json
Connection refused: connect
[image]

?? Can I manually download that? 

When I go into update then it runs and I end up with this message
[image]

So I think it was having a cow trying to download stuff as you thought. I am running the update right now. We'll see what happens. I just got this message. I clicked install anyway

[image]

After the update and restart I get this message as well as the message about being unable to download from arduino.esp8266.com
[image]

So it looks like for whatever reason it can't get those packages. Tried with my firewall off but the same result.

This is what I get after firing up sloeber, and some detail on the first error. All the other errors say "an exception stack tarce is not available.

[image]

Install was literally download the 64 bit package from your site, unzip using winrar, copy the sloeber directory to c:/users/bill. click on sloeber-ide.exe and click on help/check for updates. No other magic. I have also downloaded atom and platformio and they installed without errors, so I'm not being a total bonehead. They are not my preference because I think your package is going to do more, and I like the layout.

I am going to try on a couple other pc's to see what happens.

jantje, 2018-02-20:
>http://arduino.esp8266.com/stable/package_esp8266com_index.json

The platform providers are out of my control. The download of esp8266 has given problems in the past and apparently you seem to have problems as well.
You can download it manually and place it in [eclipse folder]/arduinoPlugin or remove it from the list of json's if you do not use esp8266 based boards

If you want to update why don't you install the nightly? Quite a bit of things have changed (like gnuarm has renamed into gnumcu) so at this point it is safer/easier to install the nightly or stick with V4.2.

The not signed perfectly normal. Signing costs money.

I see nothing abnormal for now except for, why you wan't  to update V4.2 (use nightly) esp8266 json file download is bad again and gnuarm has changed update site.

billiho, 2018-02-21:
I use and like the esp8266, so I'll try a manual download. I've setup nightly builds in the check for updates.

Do I need to create a new project, and copy in the the .ino contents of an existing sketch to the new project so it'll upload properly? Whenever I try to upload an existing arduino sketch I get a "this is not an arduino file" error, but it works fine when I upload from a sloeber project.

Is there a setup for uploads? Cpu MHz, clock speed, MIPS etc? Haven't found that yet. 

Thanks for your help!

billiho, 2018-02-21:
One thing I seem to be having an issue with is the serial port speed using esp8266's. I have set serial output as 115200 in my sketch.
Serial.begin(115200);

I upload, then check my serial monitor at 115200 and get junk. I set at 76800 and see my serial output. 

Same sketch built in arduino outputs at 115200, which is what it's supposed to do. 

I think the esp8266's default on port is 76800.  I'm puzzled by this?

jantje, 2018-02-21:
>Do I need to create a new project, and copy in the the .ino contents of an existing sketch to the new project so it'll upload properly? Whenever I try to upload an existing arduino sketch I get a "this is not an arduino file" error, but it works fine when I upload from a sloeber project.

I don't understand what you are saying.

>I upload, then check my serial monitor at 115200 and get junk. I set at 76800 and see my serial output.

Strange. Are you now on the nightly or V4.2 stable? #931

Have you watched some of my videos? arduino->videos

billiho, 2018-02-21:
I've watched a few videos that have answered several questions. I am using the nightly. 

My guess is the serial thing is sloeber is either ignoring the serial.begin(speed) statement, somehow using the default speed, or I'm using the wrong magical pixie dust. This happens regardless of what upload speed I've tried. When uploading the same code using the arduino ide, it seems to work fine.

By the looks of it you can't just open a sketch built in arduino ide and upload it. It needs to be either created in Sloeber, or the code copied into a Sloeber created project, correct? If I just open a sketch created in arduino ide and try to upload it I get the "this is not an Arduino" error message.

billiho, 2018-02-22:
switched back to esp2866 platform  2.3.0 and the serial communications appears to be working correctly now.  Must be some issue with 2.4. Tried 2.0 and it worked fine as well. Earlier versions didn't like my code.

The sloeber seems to be working now, yay! This was more work then I thought. I hope things will run smoothly now. Thanks for your help.

jantje, 2018-02-26:
I'll check the esp8266 serial baud rate when I'm done with testing #243 which is a real pain

jantje, 2018-05-18:
Serial baud rate works fine for me

closed 2018-05-18 by jantje