Sloeber/arduino-eclipse-plugin #921: ESP32 and library path

ToninoTarsi, 2018-02-19:
Hi,
I'm trying to use Sloeber with ESP32 and some sensors ( LoRa, BME280 .. )
I think installation is OK as I can compile correctly standard ESP32 examples.
But when I add sensor library I got error in include files. 

For example for Lora:
The library is correctly installed in .\arduinoPlugin\libraries\LoRa\0.3.0 and included into the project, it is found by the editor but not included into the compiler 
..\sloeber.ino.cpp:9:18: fatal error: LoRa.h: No such file or directory

See picture below.

[image]

I'm sure I'm missing something but .. what ?

Thanks

ToninoTarsi, 2018-02-19:
Looking at compiler include : 

`-I"C:\Appl\sloeber\arduinoPlugin\libraries\LoRa"`

So if I move LoRa.h and Lora.cpp in the main library folder ( arduinoPlugin\libraries\LoRa\ ) it compile correctly.
But library are not installed there by the Library manager  ( arduinoPlugin\libraries\LoRa\0.3.0\src )

[image]

jantje, 2018-02-19:
Have you tried arduino->Reattach libraries?
If that doesn't help please provide a image like the first one but with the lora folder opened and the text from the console.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

ToninoTarsi, 2018-02-19:
Thanks for your time.
arduino->Reattach libraries does not help

[image]

`14:04:59 **** Incremental Build of configuration Release for project test ****
"C:\\Appl\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/config" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/bluedroid" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/app_trace" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/app_update" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/bootloader_support" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/bt" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/driver" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/esp32" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/esp_adc_cal" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/ethernet" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/fatfs" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/freertos" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/heap" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/jsmn" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/log" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/mdns" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/mbedtls" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/mbedtls_port" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/newlib" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/nvs_flash" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/openssl" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/spi_flash" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/sdmmc" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/spiffs" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/tcpip_adapter" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/ulp" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/vfs" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/wear_levelling" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/coap" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/console" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/expat" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/json" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/lwip" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/newlib" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/nghttp" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/soc" "-IC:\Appl\sloeber\/arduinoPlugin/packages/esp32/tools/sdk/include/wpa_supplicant" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -Wall -Werror=all -Wextra -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti -c -DF_CPU=240000000L -DARDUINO=10802 -DARDUINO_ESP32_DEV -DARDUINO_ARCH_ESP32 "-DARDUINO_BOARD=\"ESP32_DEV\"" "-DARDUINO_VARIANT=\"doitESP32devkitV1\"" -DESP32 -DCORE_DEBUG_LEVEL=1  -I"C:\Appl\sloeber\arduinoPlugin\packages\esp32\cores\esp32" -I"C:\Appl\sloeber\arduinoPlugin\packages\esp32\variants\doitESP32devkitV1" -I"C:\Appl\sloeber\arduinoPlugin\libraries\LoRa" -I"C:\Appl\sloeber\arduinoPlugin\packages\esp32\libraries\SPI" -I"C:\Appl\sloeber\arduinoPlugin\packages\esp32\libraries\SPI\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
..\sloeber.ino.cpp:9:18: fatal error: LoRa.h: No such file or directory
compilation terminated.
subdir.mk:31: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1

14:05:00 Build Finished (took 939ms)`

uzi18, 2018-02-19:
Tested it with clean installation of Sloeber and build ok on Arduino Uno. 

Do You have other LoRa libraries installed?

ToninoTarsi, 2018-02-19:
Yes uzi18,
Yes uzi18 . I  tested a clean install with Arduino Uno Wifi and works ok!
Trying to reproduce it now !!
Thanks

ToninoTarsi, 2018-02-19:
I think I found the problem.

For the ESP32 toolchain installation I follow this one :  https://www.youtube.com/watch?v=xJyNLcoG7Q0
At minute 3:20 he modifies the private library path and private hardware path. 
I did the same : 
[image]
If you do this you got that problem. 
May be private library path shod pint to something different than standard one.
I can close this and continue testing the nice plug-in .. Arduino IDE is awful :-(
Thanks to all.
Tony

uzi18, 2018-02-19:
Maybe true, I did not change any private path.

closed 2018-02-19 by ToninoTarsi

jantje, 2018-02-19:
I saw that video so I decided to make my own due to ... well not fully correct https://youtu.be/szRrvJPvQz4
Always try to see whether I have a video out on the subject.

NipunaSL, 2020-01-29:
https://github.com/sandeepmistry/arduino-LoRa/blob/master/src/LoRa.h

Installed "arduino-LoRa" solved my issue