Sloeber/arduino-eclipse-plugin #922: Can't Upload to Adafruit Bluefruit nRF52 Feather

Douglas-5DT, 2018-02-19:
Hi, 

I created a test project using the nRF52 example blinky. The project compiles but does not upload.
I get the following issues:
Failed to run the Upload recipe 
Cannot run program "" (in directory "C:\Eclipse"): CreateProcess error=87, The parameter is incorrect

and

tools.nrfjprog.program.pattern : not found in the platform.txt file

I used the following settings for the board:
[image]

This is only my 1st step in trying to get everything to work. The next step would be to debug the code step by step using the j-link. If you could point to some sort of tutorial to achieve this it would be much appreciated.

jantje, 2018-02-19:
Does default upload protocol work?
I do not own a nRF52 so I don't know how to debug with that one. I have some video's on hardware debugging with the zero and esp8266 that could get you started.
https://youtu.be/WFzVzgEA8Fo
https://youtu.be/DXTo4-0pWKA

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

Douglas-5DT, 2018-02-20:
Hi, 

The upload works when I use default, but that uses the USB and not the j-link.

Douglas-5DT, 2018-02-20:
It seems that the issue was with the J-Link drivers, I reinstalled them and now programming using the J-link works.

closed 2018-02-20 by Douglas-5DT