Sloeber/arduino-eclipse-plugin #923: mvn fast build

uzi18, 2018-02-19:
Is it possible to build sloeber without downloading/fetching everything from internet? 
And also without creating archive at the end? 

I use symlink on arduinoPlugin dir to not download all every run. 
I need to have fast build and launch to fix some issues. 

@jentje please help with this :)

jantje, 2018-02-19:
install eclipse and debug? https://youtu.be/ZkJNrSjKSW0

uzi18, 2018-02-26:
Thank You Roberto. 

I'm able to build sloeber with maven and trying to understand how it work. 

@jentje how to debug plugin,  i have use System.out.println() to get info to terminal,  maybe there is better way? 
Did not try to build eclipse plugin in another eclipse...  is it a way?

I'm investigating how sloeber search libraries, because want to add project private libraries (in lib or libraries dir). 
After added it encounter strange behaviour, sloeber didn't find libs in hardware dir.

uzi18, 2018-02-26:
Need to take a look in details on this Youtube clip from @jantje.

jantje, 2018-02-26:
@uzi18 I do all the Sloeber development, debugging and testing in eclipse.
The video shows how to install eclipse to develop Sloeber. One you have done that you can set breakpoints and step through the code for a better understanding.

uzi18, 2018-02-26:
If it is possible to do breakpoints it will realy help a lot. Thanks.

jantje, 2018-02-26:
off course it is. eclipse is a full featured development environment.

wimjongman, 2018-02-28:
You can use -P SDK in maven to build the SDK

jantje, 2018-02-28:
@wimjongman indeed great work of yours. Sorry I forgot about that. I should test this and make a new video :-)

uzi18, 2018-02-28:
Thanks for info @wimjongman

uzi18, 2018-03-01:
ok it works for me now, I'm able to debug sloeber now :)

jantje, 2018-03-01:
Next step is to contribute :-)

closed 2018-03-01 by jantje

uzi18, 2018-03-02:
I need to find myself in sloeber and try first fix what is not working for me.
Something have fixed but not sure if aproch is ok for @jantje to integrate it into master :)
Now testing ;)

jantje, 2018-03-02:
Best is to start by creating a issue to describe the something that needs fixing.