Sloeber/arduino-eclipse-plugin #924: Third party URLs breaks with empty line

rlogiacco, 2018-02-26:
If you insert an empty line in the third party JSON URLs list an exception is generated:
```
Malformed url 
no protocol: 
```

rlogiacco, 2018-02-26:
Is there a reason why we add the current list to the one saved in the preferences? It seems a little bit odd...

rlogiacco, 2018-02-26:
The fact we add the current selection to the ones previously saved puts eclipse into a non-recoverable mode where you have no way to remove the breaking line: any attempt will end up in the previously described error.
The only way to remove the error is by hand editing the file where such preference is saved `<install path>/configuration/.settings/io.sloeber.arduino.prefs`.
please note the fix above prevents this from happening only in case the breaking line is an empty line, but doesn't help for any other non URL content, like a line of dashes.

I believe we should actually avoid adding the stored content: I can perform that change in 5 minutes if that's ok.

jantje, 2018-02-26:
You add to the freferences as you are in the addJsonURLs method.
The other remark is fixed in #911 are you sure you are on the latest nightly version?

jantje, 2018-02-26:
Looking at the code I would prefer to deal with this situation in SetJsonUrl like
```
	public static void setJsonURL(String[] newJsonUrls) {

		String curJsons[] = getJsonURLList();
		HashSet<String> origJsons = new HashSet<>(Arrays.asList(curJsons));
		HashSet<String> currentSelectedJsons = new HashSet<>(Arrays.asList(newJsonUrls)); 
		currentSelectedJsons.remove(new String());//newLine be gone
```

jantje, 2018-03-04:
duplicate of #911

closed 2018-03-04 by jantje