Sloeber/arduino-eclipse-plugin #927: Add support for "Pre and post build hooks (since IDE 1.6.5)"

jantje, 2018-02-26:
The arduino hardware specification https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification specifies Pre and post build hooks (since IDE 1.6.5)
recipe.hooks.sketch.prebuild.NUMBER.pattern (called before sketch compilation) is used by nucleo making the build fail

Rabe0x, 2018-05-29:
[](url)

Rabe0x, 2018-05-29:
the issue comes form picking "v1.4 Compile from source" at project properties -> Arduino -> lwIP Variant. I've chanced it to "v2 low memory" and the error was gone.
(see picture below) 
[image]

jantje, 2018-05-29:
There are more boards/options that have this issue. Problem is that make is not meant to insert actions like this.
I really need a think/research about this one

jantje, 2018-06-05:
If the workaround as proposed in #984 really works that may be a good solution for most of the arduino hooks:

>     recipe.hooks.sketch.prebuild.NUMBER.pattern (called before sketch compilation)
>     recipe.hooks.sketch.postbuild.NUMBER.pattern (called after sketch compilation)
>     recipe.hooks.libraries.prebuild.NUMBER.pattern (called before libraries compilation)
>     recipe.hooks.libraries.postbuild.NUMBER.pattern (called after libraries compilation)
>     recipe.hooks.core.prebuild.NUMBER.pattern (called before core compilation)
>     recipe.hooks.core.postbuild.NUMBER.pattern (called after core compilation)
>     recipe.hooks.linking.prelink.NUMBER.pattern (called before linking)
>     recipe.hooks.linking.postlink.NUMBER.pattern (called after linking)
>     recipe.hooks.objcopy.preobjcopy.NUMBER.pattern (called before objcopy recipes execution)
>     recipe.hooks.objcopy.postobjcopy.NUMBER.pattern (called after objcopy recipes execution)
>     recipe.hooks.savehex.presavehex.NUMBER.pattern (called before savehex recipe execution)
>     recipe.hooks.savehex.postsavehex.NUMBER.pattern (called after savehex recipe execution)

For following hooks I see no real solution with the #984 workaround:
>     recipe.hooks.sketch.prebuild.NUMBER.pattern (called before sketch compilation)
>     recipe.hooks.sketch.postbuild.NUMBER.pattern (called after sketch compilation)
>     recipe.hooks.libraries.prebuild.NUMBER.pattern (called before libraries compilation)
>     recipe.hooks.libraries.postbuild.NUMBER.pattern (called after libraries compilation)
>     recipe.hooks.core.prebuild.NUMBER.pattern (called before core compilation)
>     recipe.hooks.core.postbuild.NUMBER.pattern (called after core compilation)

But the following can be fixed with the pre build cdt options because sloeber.ino.cpp is the very first file compiled
>     recipe.hooks.sketch.prebuild.NUMBER.pattern (called before sketch compilation)

The problem with the libraries and core hooks is that it assumes a order of compilation of the source files. CDT and make (and thus Sloeber) do not do this.

jantje, 2018-09-28:
Somebuild hooks have been implemented now.
The remaining ones are simply not easy to implement in a make file.
So I'm labelling this as a known issue and close it as probably no work will be done.

closed 2018-09-28 by jantje