Sloeber/arduino-eclipse-plugin #928: support platform.local.txt

liebman, 2018-02-28:
Arduino IDE allows overriding platform (and board) configuration with a "local" file (platform.local.txt & boards.local.txt). For instance I have need to use remote serial (rfc2217) to upload which is supported using esptool.py and not the esptool included with esp8266 Arduino so I override with a platform.local.txt that includes:
```
tools.esptool.cmd=esptool.py

tools.esptool.upload.protocol=esp
tools.esptool.upload.params.verbose=--trace
tools.esptool.upload.params.quiet=
tools.esptool.upload.pattern="{cmd}" --no-stub {upload.verbose} --baud {upload.speed} --port "{serial.port}" write_flash 0x00000 "{build.path}/{build.project_name}.bin"
```

Note that this still fails to do what I want in the Arduino IDE because it does not allow manual entry of the value for the serial port. But it DOES run the esptool.py command instead of the origional esptool provided with esp8266 Ardunio.  Sloeber on the other hand allows me to specify the serial port as `rfc2217://zod.local:2217` so this would work.....

jantje, 2018-02-28:
Why Don't you use a programmer?
Currently Sloeber supports externalprogrammers.txt.
But I'll add platform.local.txt and boards.local.txt and programmers.local.txt.
Do you know how these are processed by Arduino IDE in case of referenced cores/variants?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

liebman, 2018-02-28:
Is there documentation on how I would set up the file to generate a command like:

esptool.py --no-stub --baud {upload.speed} --port "{serial.port}" write_flash 0x00000 "{build.path}/{build.project_name}.bin"

If that will work then I won't need the .local support.

jantje, 2018-03-01:
There is probably documentation but with what you have there you are almost done :-)

liebman, 2018-03-01:
This is the error i'm getting:
```
tools.Program tool not properly configured.program.pattern : not found in the platform.txt file
```
I've tried a number of variations and have not figured out what i'm missing.  Hints would be appreciated!

```
#esptool.name=esptool
#esptool.program.tool=esptool.py
#esptool.protocol=esp
#esptool.program.protocol=esp
#esptool.py.program.pattern="{cmd}" --no-stub {upload.verbose} --baud {upload.speed} --port "{serial.port}" write_flash 0x00000 "{build.path}/{build.project_name}.bin"

#esptool.name=esptool
#esptool.protocol=esp
#tools.esptool.cmd.path=esptool.py
#tools.esptool.program.protocol=esp
#tools.esptool.program.pattern="{cmd.path} --no-stub --baud {upload.speed} --port "{serial.port}" write_flash 0x00000 "{build.path}/{build.project_name}.bin"

esptoolpy.name=esptoolpy
tools.esptoolpy.upload.protocol=esp
tools.esptoolpy.upload.params.verbose=
tools.esptoolpy.upload.params.quiet=
tools.esptoolpy.upload.pattern=esptool.py --no-stub --baud {upload.speed} --port "{serial.port}" write_flash 0x00000 "{build.path}/{build.project_name}.bin"

tools.esptoolpy.program.pattern=esptool.py --no-stub --baud {upload.speed} --port "{serial.port}" write_flash 0x00000 "{build.path}/{build.project_name}.bin"
```

jantje, 2018-03-01:
Add the following to the (normally empty) platform.txt
```

rfc2217.program.tool=pyesptool
rfc2217.name=remote serial (rfc2217)
rfc2217.upload.wait_for_upload_port=false

#add path here if needed
rfc2217.tools.pyesptool.program.pattern=esptool.py --no-stub {upload.verbose} --baud {upload.speed} --port "{serial.port}" write_flash 0x00000 "{build.path}/{build.project_name}.bin"
```

Selecting the upload protocol "remote serial (rfc2217)" should do the trick both in arduino IDE as in sloeber.

And thanks for being a patron :-)

liebman, 2018-03-01:
So not using externalprogrammers.txt?

jantje, 2018-03-01:
It doesn't matter for sloeber whether you put the info in externalprogrammers.txt or programmers.txt.
It is also not a change I would want to version management.

liebman, 2018-03-01:
Perfect!  Thanks you!

closed 2018-03-01 by liebman