Sloeber/arduino-eclipse-plugin #929: Serial ports on Linux

sk8nfool, 2018-02-28:
Sloeber has trouble managing serial/tty ports on Linux (at least Ubuntu 16.04). This is because of the way serial/tty devices are instantiated and removed. When the kernel sees a new device it lets udev, the Dynamic device manager, know something new has appeared. Udev then goes about creating a node in /dev for that device. For one of my Arduino devices this typically is /dev/ttyACM0. However if this device is already in used by some other process (like Eclipse/Sloeber serial monitor) udev selects the next name that's not in use like /dev/ttyACM1. This means I have to go reconfigure the serial device for downloading programs and reconfigure the serial device for the serial monitor. This is a bit of a bother since these two are no bonded together (as they shouldn't) and the sequence of button clicks is a bit involved. I succeeded in teaching udev to create a "node" (actually a symbolic link) in /dev with a constant name based on the serial number of the arduino. I can enter this name by typing it directly into the Port box in the project properties/arduino dialog. However the Serial monitor doesn't allow direct entry of the serial device. (BTW - neither list the tty's name as an option)

jantje, 2018-02-28:
Have you tried clicking the "open serial monitor" toolbar button?
That one should open the serial monitor with the serial port of the selected project and the baud rate as specified in the code 
`Serial.begin(BaudRate);`

sk8nfool, 2018-02-28:
That works! I'm not sure what you mean by "the baud rate as specified in the code" - does Sloeber sniff out the baud rate from the arduino code?

jantje, 2018-02-28:
>does Sloeber sniff out the baud rate from the arduino code?

yes

sk8nfool, 2018-02-28:
I'm impressed. Feel free to close this topic.

jantje, 2018-02-28:
Feeling free

closed 2018-02-28 by jantje

jantje, 2018-03-02:
s I'm planning to port my "upload tests" to a linux machine some day ... could you provide a link to a nice doc on how you set these "stable upload ports" in linux?

uzi18, 2018-03-02:
If we think about usb connection it is possible to address specific device

uzi18, 2018-03-02:
example for NodeMcu with CH340:   
```
$ ls -l /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 
lrwxrwxrwx 1 root root 13 03-02 23:33 /dev/serial/by-id/usb-1a86_USB2.0-Serial-if00-port0 -> ../../ttyUSB0
$ ls -l /dev/serial/by-path/pci-0000\:00\:1d.0-usb-0\:1.4\:1.0-port0 
lrwxrwxrwx 1 root root 13 03-02 23:33 /dev/serial/by-path/pci-0000:00:1d.0-usb-0:1.4:1.0-port0 -> ../../ttyUSB0

$ lsusb |grep 340
Bus 002 Device 007: ID 1a86:7523 QinHeng Electronics HL-340 USB-Serial adapter

```
for Original Arduino Mega:   
```
$ lsusb |grep 2560
Bus 002 Device 008: ID 2341:0010 Arduino SA Mega 2560 (CDC ACM)
$ ls -l /dev/serial/by-path/*
lrwxrwxrwx 1 root root 13 03-02 23:45 /dev/serial/by-path/pci-0000:00:1d.0-usb-0:1.2:1.0 -> ../../ttyACM0
$ ls -l /dev/serial/by-id/*
lrwxrwxrwx 1 root root 13 03-02 23:45 /dev/serial/by-id/usb-Arduino__www.arduino.cc__Arduino_Mega_2560_64932343938351705222-if00 -> ../../ttyACM0

```

So if You reset device it is possible to find its actual tty, by-path means how it is connected and where, by-id means how device descibes itself to computer.
For Arduino original product You can find also serial: 64932343938351705222

sk8nfool, 2018-03-03:
This actually wasn't all that easy to figure out. I can only speak to
Ubuntu 16.04. I suspect what I'm going to describe has changed over time. I
kind of thrashed around until I got a rule that worked.

The creation of device nodes (typically in /dev) is managed by a daemon
called systemd-udevd which is configured by a bunch of files containing
"rules". These files are currently located in /lib/udev/rules.d and
/etc/udev/rules.d. The syntax for these files can be found in the udev
manual page here
<http://manpages.ubuntu.com/manpages/xenial/man7/udev.7.html>. This page
describes the structure of rules files but doesn't provide enough detail so
one can write a rule. There are boundless pages describing how to write
rules half of which are either out of date or wrong. Kind of typical for
the web.

Pawing through all this I determined that the rule should look something
like

SUBSYSTEM="tty", ATTRS{idVendor}=="vvvv", ATTRS{idProduct}=="pppp",
ATTRS{serial}=="ssss", SYMLINK+="lllll"

The trick was finding the values for "vvvv', "pppp", and "ssss". The choice
for "llll" is arbitrary. I used "ttyARDx" where "x" is a sequence number I
pick and is unique to the device with serial number "ssss".

The "udevadm" command is your friend here. Once you've found the device
path for your Arduino or whatever you can use "udevadm" to discover the
above values.

The first thing is to find the device path in /dev of your Arduino. One way
to find the device path is to look at the output of "dmesg -w". The "-w"
tells "dmesg" to follow the end of the log. For my Arduino Uno I got this

$ dmesg -w
       .
       . <current contents of ring buffer - ignore this>
       .
[160265.009900] usb 2-1.7: new full-speed USB device number 8 using ehci-pci
[160265.122009] usb 2-1.7: New USB device found, idVendor=2341,
idProduct=0043
[160265.122012] usb 2-1.7: New USB device strings: Mfr=1, Product=2,
SerialNumber=220
[160265.122014] usb 2-1.7: Manufacturer: Arduino (www.arduino.cc)
[160265.122015] usb 2-1.7: SerialNumber: 95336333735351A05232
[160265.122389] cdc_acm 2-1.7:1.0: ttyACM0: USB ACM device

The last line says that the serial device name assigned by the kernel and
udev is "ttyACM0" so the system assigned path for my Arduino is
"/dev/ttyACM0". For my Seeeduino (an Arduino clone) I got this
$ dmesg -w
       .
       . <current contents of ring buffer - ignore this>
       .
[160612.199127] usb 1-1.3: new full-speed USB device number 11 using
ehci-pci
[160612.309166] usb 1-1.3: New USB device found, idVendor=2886,
idProduct=0004
[160612.309171] usb 1-1.3: New USB device strings: Mfr=1, Product=2,
SerialNumber=220
[160612.309173] usb 1-1.3: Product: Seeeduino v4
[160612.309176] usb 1-1.3: Manufacturer: Seeed (www.seeed.cc)
[160612.309178] usb 1-1.3: SerialNumber: 95635333131351F0A1B2
[160612.309564] cdc_acm 1-1.3:1.0: ttyACM2: USB ACM device

The name assigned is "ttyACM2". The name is somewhat arbitrary. If a name
is busy (i.e. some process/device is already using that name) the next name
is used. In this case ttyACM0 was assigned to my Arduino UNO (which was
still plugged in) and ttyACM1 was open by another process although the USB
device had been removed. This arbitrary-ness is what causes the problem.
Each time a serial device is plugged in a name is assigned that depends on
what's currently plugged in. Unlike some Windows configurations that have a
nasty habit of never reusing a name Linux always assigns the first unused
name. The "ACM" part is determined by the driver for that particular USB
serial device.

Back to the rule. Once you have the assigned pathname and while the device
is still plugged in execute the following command

$ udevadm info /dev/ttyACM0 | grep -e ID_VENDOR_ID -e ID_MODEL_ID -e
ID_SERIAL_SHORT
E: ID_MODEL_ID=0043
E: ID_SERIAL_SHORT=95336333735351A05232
E: ID_VENDOR_ID=2341
$

If you just enter "udevadm info /dev/ttyACM0" udevadm will print a bunch of
other stuff. The values after the "=" are the ones to substitute in the
rule above. It's unfortunate that the parameter/property names printed by
udevadm (e.g. ID_MODEL etc) are not the same as the names you need to use
in the rule (e.g.idProduct etc). There is a "-a" option  that prints
properties in a format compatible with that need for a rule but the values
are wrong and belong to all the devices from the Arduino to the USB host
adapter. Really worthless for this effort.

Once you have the rule you can add it to /etc/udev/rules.d/99-arduino.rules
using your favorite text editor. You'll have to be superuser to create/edit
this file.

Dave

sk8nfool, 2018-03-04:
uzi18 - What you describe lets you find the current name for your device but doesn't give it a fixed name.
Also the by-id path is not really how the device describes itself but how the driver for that device decided to call it typically by concatenating the USB text descriptors in some manner. In my case of an original Arduino as you call it the device says
    iManufacturer: Arduino (www.arduino.cc)
    idVendor: 0x2341
    idProduct: 0x0043
    iSerial: 95336333735351A05232
These values are laid out in the USB device descriptor retrieved by the driver. The names of the fields are not present in the device descriptor itself although the standard uses a consistent set of names. See http://www.beyondlogic.org/usbnutshell/usb5.shtml for a fairly good description. Software authors seem to take a lot of liberty in the names they use when printing out these fields. The "lsusb -v" command seems to use the names from the standard.

As any rate, unplugging the USB cable and plugging it in again causes the kernel/udev to emumerate the device again. If a process still has the previous device open the kernel/udev assigns a new name. This is the problem folk are trying to solve.