Sloeber/arduino-eclipse-plugin #930: subdir.mk:31: recipe for target 'sloeber.ino.cpp.o' failed

JaspervDalen, 2018-03-01:
The system cannot find the path specified.

I unpacked Sloeber in C:/Sloeber to make sure there are no problems with char limits however I get this build error and i can't find the answer to it. I have already added cmd.exe/c. I try to make it work on a windows 10 machine with java 8.

underneath you can find the message before i get an error about the path. When i look in the build variables the PATH variable is red with the following value and type string. 
${A.COMPILER.PATH}${PathDelimiter}${A.BUILD.GENERIC.PATH}${PathDelimiter}${SystemRoot}\system32${PathDelimiter}${SystemRoot}${PathDelimiter}${SystemRoot}\system32\Wbem${PathDelimiter}${sloeber_path_extension}

message before error:
"C:\ET\Ecl\V4.2\sloeber\/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -Wextra -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -Dprintf=iprintf -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10802 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\ET\Ecl\V4.2\sloeber\/arduinoPlugin/packages/arduino/hardware/sam/1.6.11/system/libsam" "-IC:\ET\Ecl\V4.2\sloeber\/arduinoPlugin/packages/arduino/hardware/sam/1.6.11/system/CMSIS/CMSIS/Include/" "-IC:\ET\Ecl\V4.2\sloeber\/arduinoPlugin/packages/arduino/hardware/sam/1.6.11/system/CMSIS/Device/ATMEL/"  -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\sam\1.6.11\cores\arduino" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\sam\1.6.11\variants\arduino_due_x" -I"C:\NMEA2000\EclipOntMap" -I"C:\sloeber\arduinoPlugin\libraries\due_can\2.0.1" -I"C:\NMEA2000\EclipOntMap\OngewLibraries" -I"C:\sloeber\arduinoPlugin\packages\arduino\hardware\sam\1.6.11\libraries\SPI" -I"C:\sloeber\arduinoPlugin\libraries\Time\1.5.0" -I"C:\sloeber\arduinoPlugin\libraries\ArduinoUnit\2.2.0" -I"C:\sloeber\arduinoPlugin\libraries\ArduinoUnit\2.2.0\src" -I"C:\sloeber\arduinoPlugin\libraries\DS1307RTC\1.4.0" -I"C:\sloeber\arduinoPlugin\libraries\Esplora\1.0.4" -I"C:\sloeber\arduinoPlugin\libraries\Esplora\1.0.4\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"

jantje, 2018-03-01:
you say you unpacked sloeber in C:/Sloeber but the command says C:\ET\Ecl\V4.2\sloeber and the libraries are in C:\sloeber\arduinoPlugin\libraries
any idea where the C:\ET\Ecl\V4.2 comes from?

JaspervDalen, 2018-03-01:
I have no clue. Those folders do not exist and i could not find where this path was made or how to change it. The libraries are indeed in arduinoplugin/libraries

jantje, 2018-03-02:
Well the problem you are experiencing is the reference to `C:\ET\Ecl\V4.2\ `because there is no "`C:\ET\Ecl\V4.2\sloeber/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++"` on your system it is at "`C:/sloeber/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++"`

for platform Arduino sam V1.6.11 the cpp recipe is
`recipe.cpp.o.pattern="{compiler.path}{compiler.cpp.cmd}" {compiler.cpp.flags} -mcpu={build.mcu} -mthumb -DF_CPU={build.f_cpu}  `
Compiler path is defined ion platform.txt as
`compiler.path={runtime.tools.arm-none-eabi-gcc.path}/bin/`
So on your os `{runtime.tools.arm-none-eabi-gcc.path}` translates to `C:\ET\Ecl\V4.2\sloeber/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/`
In sloeber environment variable this should translate to 
 `A.RUNTIME.TOOLS.ARM-NONE-EABI-GCC.PATH=C:\ET\Ecl\V4.2\sloeber/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/`
on My windows OS I have
`A.RUNTIME.TOOLS.ARM-NONE-EABI-GCC.PATH=${eclipse_home}/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1`
The library path is also defined from eclipse_home so the fact there is a inconsistency there is "really strange"
Please check you project properties->C/C++ build->Build variables. It should look like the one below
[image]
If that is ok and you still have the problem check project properties->C/C++ build->Environment variables for the value of A.RUNTIME.TOOLS.ARM-NONE-EABI-GCC.PATH
[image]

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

wimjongman, 2018-03-03:
In the file menu, do switch workspace and create a new workspace folder.

If this does not work:

Please remove Sloeber completely and freshly unpack it in c:/sloeber. 
It looks like you unpacked it first in c:/et/.. ran it and then moved it or that you tried sloeber before and the workspace is pointing to old data.

jantje, 2018-03-06:
@JaspervDalen did yo get it to work?
If so how?

JaspervDalen, 2018-03-06:
@jantje I was away for the weekend, I will try to get it to work tonight

JaspervDalen, 2018-03-06:
It had to do with me importing the project of somebody else. In the imported makefile the wrong path was given and this set the ECLIPSE_HOME folder to the wrong place

jantje, 2018-03-06:
When there is something wrong with a project these are the first steps to do.
Project propeties->arduinoe->apply and close
delete the release folder in your project
rebuild

closed 2018-03-06 by jantje