Sloeber/arduino-eclipse-plugin #931: Flashing of cpp files to an ESP8266 on macOS High Sierra causes Kernel Panic

tiirex9, 2018-03-02:
Sooo as summarized in the title I have a problem with flashing cpp files to a Wemos D1 mini from my macOS running High Sierra (10.13.3).

Here some key facts:

- It's irrelevant whether I use "the product" or just the plugin.
- I installed the official driver (1.4) from here: https://wiki.wemos.cc/downloads
- The patched 1.3 driver for Sierra didn't work either. https://github.com/MPParsley/ch340g-ch34g-ch34x-mac-os-x-driver
- If I want to flash an *.ino file everything works fine, only if setup() and loop() are in a cpp file the system crashes on upload.

I attached the Panic Log, maybe some of you have a smart idea what I can try: https://www.dropbox.com/s/qmidhktyjtircgz/Kernel_2018-03-01-192943_TiiBook.panic?dl=0

For now I try to workaround this issue by making my primary file an *.ino file and everything else cpp/h files. But this bugs me a lot to be honest. :/

jantje, 2018-03-02:
Strange.
*.ino files get included in a .cpp file. So with only 1.ino file I would assume the binary file to be nearly 100% the same as the .cpp version of a sketch.
I think it would be a good idea to contact the ESP8266 community and provide them a crashing file so they can investigate the root cause of the Panic. (I don't own a mac so I don't have a clear idea on what a Panic is)

Do you have this problem with any sketch or with a particular sketch only?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

liebman, 2018-03-02:
The panic message implies that there is a bug (use of memory after free or uninitialized/rogue pointer) in the driver.  I use a mac but not a a Wemos D1 mini and have not seen any panics.

tiirex9, 2018-03-02:
I wonder if it has something to do with the USB cable or the USB-C to USB-A adapter....

tiirex9, 2018-03-02:
Toggling DTR is the last message appearing in the Eclipse Console right before the kernel panic.
Btw: Kernel panic is what Windows users call a Bluescreen.

jantje, 2018-03-03:
>Toggling DTR is the last message appearing in the Eclipse Console right before the kernel panic.

Do you mean that Sloeber is not yet trying to upload?

tiirex9, 2018-03-03:
Seems like the problem had something to with an error in my cuircit...
I misunderstood how the internal pullup worked and applied voltage to the input pin that already got voltage from the internal pullup. So ... yeah... Feeling pretty stupid now. :D But I never assumed, that my MacBook would crash because of this... I just overlooked that small detail on my breadboard.

I'm smarter now. Thanks all for helping out!

closed 2018-03-03 by tiirex9

jantje, 2018-03-03:
Thanks for reporting :-)
We all grow smarter from our mistakes.