Sloeber/arduino-eclipse-plugin #933: Same code compiles in Arduino IDE, fails in Sloeber

jrobert-gh, 2018-03-04:
I wrote a small sketch in using Sloeber that gets compiler errors, but compiles, loads, and runs correctly with the Arduino IDE. Image showing error messages below.

Maybe relevant?: 
- When Sloeber launches it says it is finishing the installation for about 15 seconds. Shouldn't this happen only on the first launch after an update?

My Arduino IDE is at 1.6.12, while the latest version is 1.8.5. (I have not updated to 1.8.5 because Sloeber''s Properties > Arduino > Arduino AVR Boards only lists IDE versions 1.6.2 through 1.6.20. Is 1.8.x supported?)

Arduino IDE:  1.6.12
Sloeber: io.sloeber.product-4.2.0-20180116.154118-36-macosx.cocoa.x86_64.tar.gz

JRobert
(patron)

[image]

Code:
```

// Multi-LED blinker

#include <Arduino.h>

#define BLINKTIME_ms    10L

typedef struct {
   const uint8_t intvl_ds;      // blink period [0.1 sec]
   const uint8_t pin;           // led pin#
   uint8_t state;               // current state
   uint32_t next_ms;            // next action time [ms]
} WINK_t;

//              dSec pin st nx
WINK_t Led1 = { 10,  9, 0, 0L };
WINK_t Led2 = { 15, 10, 0, 0L };
WINK_t Led3 = { 20, 11, 0, 0L };

void runLed(WINK_t *pled);

void setup() {
   Serial.begin(115200);
   Serial.println("MultiBlink\n");

   pinMode(Led1.pin, OUTPUT);
   digitalWrite(Led1.pin, LOW);

   pinMode(Led2.pin, OUTPUT);
   digitalWrite(Led2.pin, LOW);

   pinMode(Led3.pin, OUTPUT);
   digitalWrite(Led3.pin, LOW);
}

void loop() {
   runLed(&Led1);
   runLed(&Led2);
   runLed(&Led3);
}

void runLed(WINK_t *pled){

   switch( pled->state ){
   case( 0 ):
      // Start blink
      digitalWrite(pled->pin, HIGH);            // led on
      pled->next_ms = millis() + BLINKTIME_ms;  // blink time
      ++pled->state;                            // advance
      break;

   case( 1 ):
      // WF end on-time
      if( millis() >= pled->next_ms ){
         // Stop blink
         digitalWrite(pled->pin, LOW);            // led off
         pled->next_ms = millis() + pled->intvl_ds*100L - BLINKTIME_ms;  // off-time
         ++pled->state;                            // advance
      }
      break;

   case( 2 ):
      // WF end off-time
      if( millis() >= pled->next_ms ){
         pled->state = 0;                       // start a new cycle
      }
      break;

   default:
      break;
   }
}

```

jantje, 2018-03-05:
try putting 
```
typedef struct {
   const uint8_t intvl_ds;      // blink period [0.1 sec]
   const uint8_t pin;           // led pin#
   uint8_t state;               // current state
   uint32_t next_ms;            // next action time [ms]
} WINK_t;
```
in a header file

jrobert-gh, 2018-03-05:
Thanks, Jantje - that worked.
I'd suggest leaving the issue open though, even if it has to become a "maybe/someday", to document the requirement, since typedefing within the one file is legitimate C/C++.

JRobert

PS: to my Arduino IDE version question above: can I replace it with the current IDE version and still be compatible with Sloeber?

jantje, 2018-03-05:
I'm closing it as this is a duplicate of known limitation #775. This is the list of all known limitatios https://github.com/Sloeber/arduino-eclipse-plugin/issues?q=is%3Aissue+is%3Aclosed+label%3A%22status%3A+known+limitation%22

As to arduino IDE: Arduino IDE and Sloeber are now completely decoupled. They both work with platforms and the platform versions are completely decoupled from the IDE (That is the way it should be).
The version number 1.6.20 is the avr arduino platform version. In both the latest release of arduino IDE and Sloeber you can work with this version or older.
I hope this makes it a bit clearer to you

closed 2018-03-05 by jantje

jrobert-gh, 2018-03-05:
> I'm closing it as this is a duplicate of known limitation #775[1].

So I can replace my 1.6 with 1.8?
Or do I need to keep the 1.6 around for Sloeber's use? 

Links:

  1. https://github.com/Sloeber/arduino-eclipse-plugin/issues/775
  2. https://github.com/Sloeber/arduino-eclipse-plugin/issues/933#issuecomment-370460694
  3. https://github.com/

jantje, 2018-03-05:
You can run Sloeber without having the  Arduino IDE installed.