Sloeber/arduino-eclipse-plugin #934: "How to create a library" tutorial is out of date

sk8nfool, 2018-03-07:
I tried to follow the "How to create a library with Arduino Eclipse plugin" ([here](http://eclipse.baeyens.it/how_to.shtml)) and ran into what looks like a version mismatch at the first step. 

Also the text of the tutorial is a bit confusing about what is explanatory text and what is tutorial text.

I tried to follow the discussion about library development but am too much of a newbie to understand what's being discussed.

jantje, 2018-03-07:
I just reread the document and (apart from a typo) all seems fin to me.
Can you tell me what exact command is confusing?

sk8nfool, 2018-03-08:
sure - right off the bat the tutorial makes a project in an empty workspace
resulting in

When I do that I get

This is almost the same with the exception of the "Arduino" folder. BUT
there a no "Libraries" folder. Also the "Varient" folder is missing. I
chalked these up to version differences and kept going. I tried to create
the "Libraries" fold and ran into my next problem. Eclipse has two folder
creation options: a plain folder and a source folder. I assume the
difference is whether or not the folder is included in the build, You also
don't specify which to use in the tutorial. However the next step

In most cases you will start by simply creating a class (right click the
> folder and select new class) inside a existing project.
>

Which folder? Eclipse allows me to right click all folders except
"includes" and create a new class in each of them. Since the next lines say
you want to create a new class at the project level which will get moved to
the right place later. You didn't describe how to create said class and
went on to creating a subfolder. So I assumed you meant for me to create a
new class at the project level by right clicking on the project
container(?). Which I did (it would have been useful to tell me what name
you used for the class so I wouldn't have to figure it out later). Plowing
on

The next part says

*Create a subfolder with the library name in you private library folder*

Um - what's a "private library folder"?

At this point I gave up and asked for help by complaining that the tutorial
seemed to be out of date. I still think that is true and hope that when you
re-do the tutorial you would do a bit more in the way of hand holding.

Dave

sk8nfool, 2018-03-08:
Being the stubborn donkey that I am I tried to figure out the next step and
realized that "*Create a subfolder with the library name in you private
library folder*
" was to be done OUTSIDE of eclipse - that's weird! So I did what I thought
you were suggesting and created /home/braun/Arduino/ - wait - I already
have that folder! Must have been created when I installed the Arduino IDE.
And it has some libraries in it that look familiar. Ok - I created the
LibCodeClass folder.

The next line

select the Arduino item in the menu->import an Arduino library.

was a bit difficult to parse but I decided you meant

Right click the project, select Import, select Arduino and select  "Import
Arduino Libraries into project"

of

Click Arduino in the menu bar, select "Add a library to the selected
project" and select the "library" (folder) you created above (LibCodeDemo)

See what I mean about hand-holding - As a complete idiot I find it more
comforting to say "yep - that looks like the tutorial" than "um - did
he/she mean ...?".

BTW - In general Sloeber is a very nice piece of work but departs from the
Eclipse tradition in some fairly significant ways. All the Arduino specific
behaviors can be a bit confusing because they create a dichotomy of methods
to do essentially the same thing. I realize you're trying to make Sloeber
feel like Arduino so the transition is easier but I think you're going to
find it increasingly difficult to maintain. Eclipse/CDT already knows how
to build C/C++ programs and can be configured to use special compilers and
linkers (cross compilers). Adding Arduino specifics is at best redundant.

Dave

sk8nfool, 2018-03-08:
oops - images were removed from email reply - 
This one is the first image in the tutorial
[image]

This is what I got by creating a new workspace and project
[image]

jantje, 2018-03-08:
The 2 images look pretty much the same to me.
Core has been renamed to arduino.
The variant folder is a subfolder of core/arduino so you need to open that one up. Note that not every board has a variant folder. For instance Teensy does not use variant.

> I tried to create the "Libraries" fold and ran into my next problem. 

 you did not use a library so libraries folder is not yet there.
I have the impression you are trying to do something advanced and miss the basics of using Sloeber.
http://eclipse.baeyens.it/how_to.shtml#/lib

> (right click the folder and select new class)

I'll change this to right click the project and select new class and change the last line to "So we create a class in our project with the name of the library we want to create."

>Um - what's a "private library folder"?

This is basic Sloeber configuration you can do in Arduino->preferences->arduino.
I also need to update the configure page :-(

>was to be done OUTSIDE of eclipse 

good point. Added this. I do not think it is wierd though ;-)

>Click Arduino in the menu bar, select "Add a library to the selected project" and select the "library" (folder) you created above (LibCodeDemo)    

Replaced my text with yours. By the way... Is Dutch your mother tongue?     

> Adding Arduino specifics is at best redundant.

I do not agree  (see below) By the way Dough of CDT started his own Arduino eclipse plugin so he doesn't agree as well :-) ( He I found something else dough and I agree on ;-) )

>departs from the Eclipse tradition in some fairly significant ways. 

I agree with that. But after 8 years I do not think it is " increasingly difficult to maintain". 
The ideology between eclipse/CDT and Arduino is completely different. Basically eclipse is bottom up and Arduino is top down.  
What I mean is: in arduino you select a board and that means you know the compile commands the compiler and so on. Everything is known because; "in arduino you selected the top of the pyramid".
Eclipse on the other hand provides you a way to make pyramids but you can (and have to) configure everything. CDT does not even come with compilers and does not have a way to "install compilers". CDT assumes you point to the location on disk where you have your compilers installed.
What Sloeber does is 
1) "take the arduino pyramid info" and translate it into eclipse configuration instructions. 
2) Maintain a multi compiler/tool environment and directs eclipse/CDT to the right tool.
3) Configure eclipse/CDT based on selecting the top of the pyramid.
4) Add support to CDT for using arduino libraries.

jantje, 2018-03-09:
The changes are available on the web site.
Any other improvements you can see?

sk8nfool, 2018-03-09:
What? Which version is the tutorial written to? I'm using V4.2 from
V4.2_linux64.2017-11-20_02-16-52.tar.gz which when I poke the Check For
Updates says it's up to date. Did you get this backwards? My version
doesn't have an Arduino folder.

I haven't used any libraries because I was following the tutorial and it
hadn't used any libraries.

My mouth speaks American English, My fingers generally do what they want
and only pretend to speak English. Why did you think I spoke Dutch?

I hadn't thought of this before but you're correct. Eclipse/CDT doesn't
come with compilers. My Linux (Ubuntu) did. But Sloeber supplied a copy of
the GNU tools which support a host of CPU architectures. Admittedly the
host CPU distribution may not be compiled for more that the host processor
but recompilation with the proper options fixes that. To be honest
rebuilding the GNU tools is not something I've ever enjoyed doing.

jantje, 2018-03-09:
Depending on your update settings I assume you are on 4.2 which is the latest stable.
>Did you get this backwards? 

yes
Core has been renamed to arduino.
Should have said:
Arduino has been renamed to core.

>I haven't used any libraries because I was following the tutorial and it hadn't used any libraries.

If you would have created a new project based on a example using a library you would have had a library folder. If the last project you created was with a library example and you clicked finish after providing a name you would have libraries folder.

> My mouth speaks American English, My fingers generally do what they want and only pretend to speak English. Why did you think I spoke Dutch?

Because you used the word of. But it will have been a typo for or. but a typo which means "or" in Dutch :-)

>To be honest rebuilding the GNU tools is not something I've ever enjoyed doing.

That you do not need to do this is to be credited to Arduino. The only package I'm aware of that uses your local gnu compiler (on linux and mac); is the local boards from the jantje package that allows for local debugging and unit testing.

But is the page ok now?

sk8nfool, 2018-03-09:
I have used Eclipse/CDT on another computer for work and it used the local
GNU C/C++ compilers by default. I had to re-build them for a different
processor (Freescale?) to do cross compiles for the project. I don't recall
if that was anything more than a painful experience and may have just
bought the dev kit and used their version. Not fun.

As for the current page - in no particular order

   1. the paragraph title "*Create a subfolder with the library name in you
   private library folder*" should read "... in you*r* private ..."
   2. "Create" in the following line shouldn't be capitalized
   3. "Outside" ... "End of outside..." - Very cute but ...
   4. "... good practise..." should read "... good practice ..."
   5. "*Import the library in your project*" should read "... into ..."
   instead of "... in ..."
   6. The layout of the "Select the Arduino libraries" is completely
   different in V4.2. I get a single linear list with no categories line
   "Hardware Libraries" or "Personal Libraries"
   7. Something in very strange about the page format - the "Select the
   Arduino Libraries" image bisects(?) the sentence "... and select the
   "library" (folder) you created above (LibCodeDemo).  now your project
   should look something like: " Not sure how to describe it.
   8. Missing an "e" at the end of "not" in "Also not the small arrow in
   the folder ..."

I asked before (possible in another issue) whether to use "New->Folder" or
"New->Source Folder". I don't understand the differences between these and
how Eclipse behaves with them. Are Library folders different too?

Dave

jantje, 2018-03-15:
Strange.. still wierd looking stuff:-(

sk8nfool, 2018-03-16:
Ignore the complaint about "practise" - I was confused by  the non-US
English spelling