Sloeber/arduino-eclipse-plugin #935: Sloeber fails to switch workspaces

sk8nfool, 2018-03-08:
When I select File->Switch Workspace and select a _different_ workspace the current workspace disappears (as expected), a box of some kind flashes and sloeber disappears.

jantje, 2018-03-08:
I tried it here. Works fine
Here = windows 10 nightly sloeber product
What is here with you?

jantje, 2018-03-08:
As to the warning...
If you know how to get rid of it ... please inform me :-)

sk8nfool, 2018-03-09:
I figured out how to configure log4j. First create a file with the following contents

```
<?xml version="1.0" encoding="UTF-8" ?>
<!DOCTYPE log4j:configuration SYSTEM "log4j.dtd">

<log4j:configuration xmlns:log4j="http://jakarta.apache.org/log4j/">
  <appender name="console" class="org.apache.log4j.ConsoleAppender"> 
    <param name="Target" value="System.out"/> 
    <layout class="org.apache.log4j.PatternLayout"> 
      <param name="ConversionPattern" value="%-5p %c{1} - %m%n"/> 
    </layout> 
  </appender> 

  <root> 
    <priority value ="debug" /> 
    <appender-ref ref="console" /> 
  </root>

</log4j:configuration>
```
This file can be named as you desire and placed in a folder of you choosing. I used /home/braun/log4j.xml.

Add the following to <wherever>/sloeber-ide.ini (I used /home/braun/.local/share/sloeber/V4.2_linux64.2017-11-20_02-16-52/sloeber-ide.ini)

```
-Dlog4j.debug
-Dlog4j.configuration=file:///home/braun/log4j.xml
```
I placed these with the other -D flags.

This did a few things

1. got rid of the warning
2. caused a bunch of debug/logging output
3. fixed the problem - switches workspaces just fine now

So - the reason you couldn't get it to fail is either because you're using Win 10 and/or have some kind of logging enabled for your development. I just run whatever was installed with V4.2_linux64.2017-11-20_02-16-52.tar.gz (with manual updates installed a few days ago).

sk8nfool, 2018-03-09:
hmm - I've been running sloeber-ide from a terminal. If I use a shortcut in the launcher it dies. I attached an strace to the process just before switching workspaces and got
```
strace: Process 5395 attached
futex(0x7fce2473c9d0, FUTEX_WAIT, 5397, NULL <unfinished ...>
+++ exited with 24 +++
```

sk8nfool, 2018-03-09:
sorry - Here for me is Ubuntu 16.04.4 LTS with latest updates

jantje, 2018-03-09:
I can reproduce the issue on ubuntu.
It seems like java crashes, a error log file is created but it didn't make me wiser.

wimjongman, 2018-03-10:
Do you think that the logging issues is preventing the workspace switch? I'm happy to add the log4j configuration to Sloeber.

jantje, 2018-03-10:
@wimjongman That would be cool :-). It annoyed me from time to time.
My gut-feel (no argumentation) is that the crash is caused by jdms which behaves "differently" in ubuntu

sk8nfool, 2018-03-10:
Just to throw a wrinkle into the mix. The switch works when I configured
logging.

jantje, 2018-04-04:
@wimjongman As to the pull request Iwould say this has been fixed but when I start a debug session I get
```
log4j:WARN No appenders could be found for logger (com.gradleware.tooling.toolingutils.distribution.PublishedGradleVersions).
log4j:WARN Please initialize the log4j system properly.
```
Is there something I need to configure or is it not yet fully fixed?

jantje, 2018-09-28:
Can this issue be closed?

closed 2020-01-18 by jantje