Sloeber/arduino-eclipse-plugin #938: Check for updates in help menu complains about being unable to complete operation

sk8nfool, 2018-03-09:
I'm using Sloeber V4.2 on Ubuntu V16.04 LTS - when I got to Help->Check for Updates i get a couple of popups that go too fast for me to read and end up with two windows. One (usually under the second) titled Problem Occurred  saying 

> 'Searching Alternate Solutions ...' has encountered a problem. No remedy found

and the second one titled "Available Updates' saying

>  The operation cannot be completed.  See the details.

followed by a list of what I presume are updates

> C/C++ Common GDB Support	9.4.0.201712051550
> C/C++ Development Platform	9.4.0.201712051550
> C/C++ Development Tools	9.4.0.201712051550
> C/C++ DSF GDB Debugger Integration	9.4.0.201712051550
> C/C++ GNU Toolchain Build Support	9.4.0.201712051550
> C/C++ GNU Toolchain Debug Support	9.4.0.201712051550

This is followed by a Details list saying

> Cannot complete the install because of a conflicting dependency.
>   Software being installed: C/C++ Development Platform 9.4.0.201712051550 (org.eclipse.cdt.platform.feature.group 9.4.0.201712051550)
>   Software currently installed: Sloeber, the Eclipse Arduino IDE 4.2.0.201711200109 (io.sloeber.product 4.2.0.201711200109)
>   Only one of the following can be installed at once: 
>     C/C++ Development Tooling Native Utilities 9.3.3.201710130142 (org.eclipse.cdt.native.feature.jar 9.3.3.201710130142)
>     C/C++ Development Tooling Native Utilities 9.3.2.201709131603 (org.eclipse.cdt.native.feature.jar 9.3.2.201709131603)
>     C/C++ Development Tooling Native Utilities 9.4.0.201712051550 (org.eclipse.cdt.native.feature.jar 9.4.0.201712051550)
>   Cannot satisfy dependency:
>     From: Sloeber, the Eclipse Arduino IDE 4.2.0.201711200109 (io.sloeber.product 4.2.0.201711200109)
>     To: org.eclipse.cdt.native.feature.group [9.3.3.201710130142]
>   Cannot satisfy dependency:
>     From: C/C++ Development Tooling Native Utilities 9.3.3.201710130142 (org.eclipse.cdt.native.feature.group 9.3.3.201710130142)
>     To: org.eclipse.cdt.native.feature.jar [9.3.3.201710130142]
>   Cannot satisfy dependency:
>     From: C/C++ Development Tooling Native Utilities 9.4.0.201712051550 (org.eclipse.cdt.native.feature.group 9.4.0.201712051550)
>     To: org.eclipse.cdt.native.feature.jar [9.4.0.201712051550]
>   Cannot satisfy dependency:
>     From: C/C++ Development Platform 9.4.0.201712051550 (org.eclipse.cdt.platform.feature.group 9.4.0.201712051550)
>     To: org.eclipse.cdt.native.feature.group [9.4.0.201712051550]

jantje, 2018-03-09:
This is probably a duplicate of #905

sk8nfool, 2018-03-10:
905 sound very similar. so what's the fix?

jantje, 2018-03-10:
Adding these update site
http://gnu-mcu-eclipse.netlify.com/v4-neon-updates
http://download.eclipse.org/tools/cdt/releases/9.4

sk8nfool, 2018-03-11:
Already installed. Tried to check for updates again and got the same thing:

> Cannot complete the install because of a conflicting dependency.
>   Software being installed: C/C++ Development Platform 9.4.2.201802122019 (org.eclipse.cdt.platform.feature.group 9.4.2.201802122019)
>   Software currently installed: Sloeber, the Eclipse Arduino IDE 4.2.0.201711200109 (io.sloeber.product 4.2.0.201711200109)
>   Only one of the following can be installed at once: 
>     C/C++ Development Tooling Native Utilities 9.4.0.201712020452 (org.eclipse.cdt.native.feature.jar 9.4.0.201712020452)
>     C/C++ Development Tooling Native Utilities 9.3.3.201710130142 (org.eclipse.cdt.native.feature.jar 9.3.3.201710130142)
>     C/C++ Development Tooling Native Utilities 9.4.0.201801130900 (org.eclipse.cdt.native.feature.jar 9.4.0.201801130900)
>     C/C++ Development Tooling Native Utilities 9.4.2.201802122019 (org.eclipse.cdt.native.feature.jar 9.4.2.201802122019)
>     C/C++ Development Tooling Native Utilities 9.3.2.201709131603 (org.eclipse.cdt.native.feature.jar 9.3.2.201709131603)
>     C/C++ Development Tooling Native Utilities 9.4.0.201712051550 (org.eclipse.cdt.native.feature.jar 9.4.0.201712051550)
>   Cannot satisfy dependency:
>     From: Sloeber, the Eclipse Arduino IDE 4.2.0.201711200109 (io.sloeber.product 4.2.0.201711200109)
>     To: org.eclipse.cdt.native.feature.group [9.3.3.201710130142]
>   Cannot satisfy dependency:
>     From: C/C++ Development Tooling Native Utilities 9.3.3.201710130142 (org.eclipse.cdt.native.feature.group 9.3.3.201710130142)
>     To: org.eclipse.cdt.native.feature.jar [9.3.3.201710130142]
>   Cannot satisfy dependency:
>     From: C/C++ Development Tooling Native Utilities 9.4.2.201802122019 (org.eclipse.cdt.native.feature.group 9.4.2.201802122019)
>     To: org.eclipse.cdt.native.feature.jar [9.4.2.201802122019]
>   Cannot satisfy dependency:
>     From: C/C++ Development Platform 9.4.2.201802122019 (org.eclipse.cdt.platform.feature.group 9.4.2.201802122019)
>     To: org.eclipse.cdt.native.feature.group [9.4.2.201802122019]

Here's a screenshot
[image]

and

[image]

jantje, 2018-03-11:
Seems like you do not have the sloeber nightly in your update sites.
You have 2 options
1) stay with the final release and cdt 9.3 by removing update site 
http://download.eclipse.org/tools/cdt/releases/9.4
2) use the nightly and cdt 9.4 by adding the nightly update site
http://eclipse.baeyens.it/update/V4/nightly

As you are  using esp8266 I advice option 1 as I did some serious changes in Sloeber and I think there is still a issue with esp8266 and Wi-Fi.

jantje, 2018-03-15:
Did this work for you?

closed 2018-03-30 by jantje