Sloeber/arduino-eclipse-plugin #941: Uploading to ESP8266 not working

salym8, 2018-03-20:
I cannot seem to upload to my ESP8266 board (an ESP-12F). I set up the project with the same details (board, upload protocol, etc) that worked in the Arduino IDE, but when I try to upload from Eclipse I get the following output:

> Starting upload
> using arduino loader
> Starting reset using DTR toggle process
> Toggling DTR
> Continuing to useCOM26
> Ending reset
> 
> 
> LaunchingC:\Program Files\eclipse\/arduinoPlugin/packages/esp8266/tools/esptool/0.4.9/esptool.exe -cd ck -cb -cp COM14 -ca 0x00000 -cf X:\Projects\esp-test\esp-test-project/Release/esp-test-project.bin 
> Output:
> /arduinoPlugin/packages/esp8266/tools/esptool/0.4.9/esptool.exe finished
> upload done

It does this instantly, and definitely has not uploaded. COM14 is definitely available and not in use anywhere else. Managed to get the sketch to compile without errors. Any ideas what to try next?

salym8, 2018-03-20:
Quick update... If I run the above esptool command from command line (windows), but include the upload baud rate after -cb then it works. The upload speed is set in the project settings, but is not being included in the command at upload. I guess I need to know where that command string is generated and add something in?

jantje, 2018-03-20:
Which version of sloeber are you running?

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

salym8, 2018-03-20:
I was using an older version, then updated to the latest version (4.2) and still having the same issue.

jantje, 2018-03-20:
in 4.2 it is a duplicate #907 
it used to be a problem, then it got fixed, then it reappeared. you must have jumped over the working version, or something else is wrong ;-)
as you are on windows you can safely go to the nightly which contains  the fix. 
You can opt to install a new version directly or upgrade your current install.
In the latter add following update sites (due to #938)
```
http://gnu-mcu-eclipse.netlify.com/v4-neon-updates
http://download.eclipse.org/tools/cdt/releases/9.4
http://eclipse.baeyens.it/update/V4/nightly
```

salym8, 2018-03-20:
Nightly fix worked for me :)

closed 2018-03-20 by salym8

salym8, 2018-03-20:
Just to clarify... I also had to remove the line `[myboard].upload.speed=XXXXX` as suggested in [#907](https://github.com/Sloeber/arduino-eclipse-plugin/issues/907), but without the nightly I wasn't getting a speed at all! The nightly defaulted to 19200 if that line was left in.

jantje, 2018-03-20:
In the nightly you should not have to modify the boards.txt.
Is that is necessary it is not properly fixed.
I've done lots of changes on the defaults (for testing), I can only know the default when I see it.
Also note the esp8266 community has released lots of new versions so the version used is important as well.

salym8, 2018-03-20:
Apologies! I thought it was not working before, but I just added the upload.speed line back in to boards.txt, then checked my baud rate was correct in project properties, and it compiles and uploads as expected. I must have got myself in a muddle before.