Sloeber/arduino-eclipse-plugin #942: debug esp8266: connection is not recognised

ghislainborremans, 2018-03-21:
starting Debug gives the message below. I did enter th com port the way it was described in the video but i tried a number of ways ( COM3, /COM3,.....)
Can this be explained so that it works on a W8 or W10 pc? 

thanks 

Error in final launch sequence
Failed to execute MI command:
-target-select remote \\.\\COM3
Error message from debugger back end:
\\.\\COM3: No such file or directory.
\\.\\COM3: No such file or directory.

jantje, 2018-03-21:
@ghislainborremans 
First of all thanks for being a patron :-)
the com port I used was \.\[windows name]
My last video esp8266 video was recorded on a windows 10 machine. https://www.youtube.com/watch?v=WFzVzgEA8Fo
The comport seems correct to me; are you sure you are pointing to the correct gdb debugger? https://youtu.be/WFzVzgEA8Fo?t=490

freeck, 2018-03-21:
Hi there, 

I get the same error in case "com3"  does not exist. So check whether the port is available. Could be that a background process is still occupying the debug port (could be an running gdb process of an earlier debug-session).  Both /./com3 or  \.\com3 should work on Win10.
Is this the first time you used the debugger, also check whether the debug message "$T05#b9" is output by the debug port when you reset the target.

ghislainborremans, 2018-03-21:
Getting frustrated:
reinstalled sloeber 4.2 
create new sketch Blinkwithoutdelay
compile OK ( D1 mini board 2.4.1)
create espDebug conf with jantje 1.0 
compile generates the below undefined references??
what can this be? 

D:\Electronica\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lib\libphy.a(phy_sleep.o): In function `pm_unmask_bt':
(.irom0.text+0xbec): undefined reference to `__wrap_register_chipv6_phy'
D:/Electronica/New_Workspace/blinkwithoutdelay/espDebug/arduino.ar(core_esp8266_main.cpp.o): In function `loop_task':
D:\Electronica\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/core_esp8266_main.cpp:111: undefined reference to `ets_post(unsigned char, unsigned int, unsigned int)'
D:/Electronica/New_Workspace/blinkwithoutdelay/espDebug/arduino.ar(core_esp8266_main.cpp.o): In function `esp_schedule':
D:\Electronica\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/core_esp8266_main.cpp:111: undefined reference to `ets_post(unsigned char, unsigned int, unsigned int)'
D:/Electronica/New_Workspace/blinkwithoutdelay/espDebug/arduino.ar(core_esp8266_main.cpp.o): In function `init_done()':
D:\Electronica\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/core_esp8266_main.cpp:111: undefined reference to `uart_div_modify(int, int)'
D:/Electronica/New_Workspace/blinkwithoutdelay/espDebug/arduino.ar(core_esp8266_main.cpp.o):(.text.user_init+0x1c): undefined reference to `ets_task(void (*)(ETSEventTag*), unsigned char, ETSEventTag*, unsigned char)'
D:/Electronica/New_Workspace/blinkwithoutdelay/espDebug/arduino.ar(core_esp8266_main.cpp.o): In function `user_init':
D:\Electronica\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/core_esp8266_main.cpp:111: undefined reference to `uart_div_modify(int, int)'
D:\Electronica\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266/core_esp8266_main.cpp:111: undefined reference to `ets_task(void (*)(ETSEventTag*), unsigned char, ETSEventTag*, unsigned char)'
D:/Electronica/New_Workspace/blinkwithoutdelay/espDebug/arduino.ar(umm_malloc.c.o):(.text.dump_mem+0x4): undefined reference to `printf'
D:/Electronica/New_Workspace/blinkwithoutdelay/espDebug/arduino.ar(umm_malloc.c.o): In function `dump_mem':
D:\Electronica\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\umm_malloc/umm_malloc.c:790: undefined reference to `printf'
D:/Electronica/New_Workspace/blinkwithoutdelay/espDebug/arduino.ar(umm_malloc.c.o): In function `check_poison':
D:\Electronica\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\umm_malloc/umm_malloc.c:821: undefined reference to `printf'
D:\Electronica\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\umm_malloc/umm_malloc.c:825: undefined reference to `printf'
D:/Electronica/New_Workspace/blinkwithoutdelay/espDebug/arduino.ar(umm_malloc.c.o): In function `check_poison_block':
D:\Electronica\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\umm_malloc/umm_malloc.c:839: undefined reference to `printf'
D:/Electronica/New_Workspace/blinkwithoutdelay/espDebug/arduino.ar(umm_malloc.c.o):D:\Electronica\sloeber\arduinoPlugin\packages\esp8266\hardware\esp8266\2.4.1\cores\esp8266\umm_malloc/umm_malloc.c:848: more undefined references to `printf' follow
D:\Electronica\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lib\liblwip_gcc.a(ip_addr.o): In function `ip4_addr_netmask_valid':
/Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/ipv4/ip_addr.c:110: undefined reference to `isxdigit'
D:\Electronica\sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.3.0/tools/sdk/lib\liblwip_gcc.a(ip_addr.o): In function `ipaddr_aton':
/Users/igrokhotkov/espressif/arduino/tools/sdk/lwip/src/core/ipv4/ip_addr.c:190: undefined reference to `isxdigit'
collect2.exe: error: ld returned 1 exit status
makefile:82: recipe for target 'blinkwithoutdelay.elf' failed
make: *** [blinkwithoutdelay.elf] Error 1

jantje, 2018-03-21:
Debugging currently only works with esp8266 V2.3
In your case 2.4.1 is used.

ghislainborremans, 2018-03-21:
I have compiled for the 2.3.0 version and that is OK . 
i now get the following:  Bogus trace status ...

Error in final launch sequence
Failed to execute MI command:
-target-select remote COM4
Error message from debugger back end:
Bogus trace status reply from target: timeout
Bogus trace status reply from target: timeout

jantje, 2018-03-22:
That is the exact same error as you can see in the video here https://youtu.be/WFzVzgEA8Fo?t=574
You need to add gdb.h and make loop and others ram funcs as I do in the video

jantje, 2018-03-30:
@ghislainborremans 
Does it work now?

ghislainborremans, 2018-03-30:
No, I have to retry. i could not spend more time and tried to debug via Serial.print.The AP mode on a ES8266 after STA mode has been disconnected is not working.
I will come back to this. I ordered a ESP32 to check if this is better. 
Is there support for the ESP32 in Sloeber?

jantje, 2018-03-30:
>Is there support for the ESP32 in Sloeber?

yes but not (yet?) for debugging

jantje, 2018-09-28:
Is there a need to keep this open?

ghislainborremans, 2018-09-29:
I am not active at this time, so it better be closed. If needed i will ask to reopen.

closed 2018-09-29 by ghislainborremans