Sloeber/arduino-eclipse-plugin #943: Serial Monitor  for YUN

GAM51, 2018-03-21:
I use YUN and  in SLOEBER V4.2 I can upload Sketches over the Network into the 'Atmel-Part' of YUN.
But in Sloeber I can't use the Serial Port.

In ARDUINO-IDE ist is possible to use the Serial-Port.

Is there any further configuration in SLOEBER?

jantje, 2018-03-21:
Short answer.
On windows I think Sloeber needs changes to get it to work
On linux you can got it to work with changes on the os level
on mac I don't know.

Long answer
I ssh into the yun and open the serial monitor there (I needed to install one, I don't recall which one I installed)
On linux I once rerouted stuff and could open the serial monitor that way. I started off this way but I went to the ssh solution because I liked it more. I connected the usb ports of the yun so I could use Serial instead of Serial1 and a different serial port than the upload port (more stuff to install on yun). This way the serial monitor running  in ssh can remain open during uploads and you can switch between directly connecting to your pc or ssh without having to change the sketch.
On linux I'm pretty sure you can reroute the serial output from the openwrt serial output to a serial port readable by Sloeber.
On windows I'm not sure (Actually I doubt that it will be possible)

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

GAM51, 2018-03-21:
Hi Jan,
thank you for the quick answer.
I will try to connect over ssh

I'm still a Patreon.

jantje, 2018-03-21:
>I'm still a Patreon.

thanks for that :-)

jantje, 2018-03-30:
Can we close this issue?

GAM51, 2018-03-30:
Hallo Jantje,

 

yes, you can close this issue.

 

I wish you a good Easter-Weekend

 

Georg Münch

 

Diese E-Mail wurde von Avast Antivirus-Software auf Viren geprüft.
https://www.avast.com/antivirus

jantje, 2018-03-30:
txs

closed 2018-03-30 by jantje