Sloeber/arduino-eclipse-plugin #944: String declaration for variables will freeze the code on the STM32 f103

ORGATHM, 2018-03-24:
Hello.
I got a strange problems. I startet a project in the original Arduino IDE. It works great. After hundrets of rows coding and a dozend of functions, i feel annoyed about the bad arduino IDE. 
So i searched for a better IDE. After a couples of hours i found the great Sloeber Eclipse mod.
So far so good, I inserted the board extension for STM32 (F103C8T8), STM32duino:
http://dan.drown.org/stm32duino/package_STM32duino_index.json
Some sample Scripts are running well. For example, the Analog Clock over SPI ILI9341 Display.
Only my own Project freezes. The Debuglog shows a couple of yellow and one red bugs, but with this bugs the Analog Clock demo skripts works. That was not the real Problem. 
The real problem is not marked as a probem... if the programm comes to this lines onto the microcontroller, it freezes.

For better understanding and replicating the problem, i modded the "blink" skript.
So everybody can check it by himself.

I used the String definition for Variables, a couple of times I used it to push static Text into my functions.
Both will run with ardiono original IDE, but not with Sloeber.

`/*
  Blink  by Scott Fitzgerald
  Modified by Roger Clark. www.rogerclark.net for Maple mini 25th April 2015 , where the LED is on PC13
  Modified by DOMin8or for "String" type variables problem demonstration.
  
  same freezing with function definition like
  void myFunction(String inputText)
                            {;}
*/

#include "Arduino.h"
// #include "header.h"
// #include <string.h>

void setup() {
                              // initialize digital pin PC13 (internal led from STM32 F103) as an output.
  pinMode(PC13, OUTPUT);
}

void loop() {                 // The 5 rounds loop will work
  for(byte xi=0;xi<5;xi++)
  {
  digitalWrite(PC13, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);                // wait for a second
  digitalWrite(PC13, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);                // wait for a second
  }

String text="test";           // this is the bad boy, that freezes the code

                              // FREEZE!
for(byte xi=0;xi<10;xi++)     // this code will NOT start. the LED is now ON
 {                            // this fast blink loop will not start.
 digitalWrite(PC13, HIGH);    // (turn the LED on (HIGH is the voltage level)
 delay(200);                  // (wait for 200ms)
 digitalWrite(PC13, LOW);     // (turn the LED off by making the voltage LOW)
 delay(200);                  // (wait for 200ms)
 }

}
`
[BlinkTest.ino.txt](https://github.com/Sloeber/arduino-eclipse-plugin/files/1844865/BlinkTest.ino.txt)

I make some log files, after finding the verbose options in both IDE´s .

ORGATHM, 2018-03-24:
Logfile with all verbose data on the original Arduino IDE 1.8,5

E:\Developement\Arduino\IDE\Arduino_Original_IDE\Arduino\arduino-builder -dump-prefs -logger=machine -hardware E:\Developement\Arduino\IDE\Arduino_Original_IDE\Arduino\hardware -hardware C:\Users\dom\AppData\Local\Arduino15\packages -tools E:\Developement\Arduino\IDE\Arduino_Original_IDE\Arduino\tools-builder -tools E:\Developement\Arduino\IDE\Arduino_Original_IDE\Arduino\hardware\tools\avr -tools C:\Users\dom\AppData\Local\Arduino15\packages -built-in-libraries E:\Developement\Arduino\IDE\Arduino_Original_IDE\Arduino\libraries -libraries C:\Users\dom\Documents\Arduino\libraries -fqbn=stm32duino:STM32F1:genericSTM32F103C:device_variant=STM32F103CB,upload_method=serialMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10805 -build-path C:\Users\dom\AppData\Local\Temp\arduino_build_994582 -warnings=all -build-cache C:\Users\dom\AppData\Local\Temp\arduino_cache_426374 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.stm32tools.path=C:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2018.2.26 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\dom\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\dom\AppData\Local\Temp\arduino_modified_sketch_443718\Blink.ino
E:\Developement\Arduino\IDE\Arduino_Original_IDE\Arduino\arduino-builder -compile -logger=machine -hardware E:\Developement\Arduino\IDE\Arduino_Original_IDE\Arduino\hardware -hardware C:\Users\dom\AppData\Local\Arduino15\packages -tools E:\Developement\Arduino\IDE\Arduino_Original_IDE\Arduino\tools-builder -tools E:\Developement\Arduino\IDE\Arduino_Original_IDE\Arduino\hardware\tools\avr -tools C:\Users\dom\AppData\Local\Arduino15\packages -built-in-libraries E:\Developement\Arduino\IDE\Arduino_Original_IDE\Arduino\libraries -libraries C:\Users\dom\Documents\Arduino\libraries -fqbn=stm32duino:STM32F1:genericSTM32F103C:device_variant=STM32F103CB,upload_method=serialMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10805 -build-path C:\Users\dom\AppData\Local\Temp\arduino_build_994582 -warnings=all -build-cache C:\Users\dom\AppData\Local\Temp\arduino_cache_426374 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.stm32tools.path=C:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2018.2.26 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\dom\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\dom\AppData\Local\Temp\arduino_modified_sketch_443718\Blink.ino
Using board 'genericSTM32F103C' from platform in folder: C:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26
Using core 'maple' from platform in folder: C:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26
Detecting libraries used...
"C:\Users\dom\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10805 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103CB  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB  -mthumb  -march=armv7-m -D__STM32F1__ "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple" "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple/include" "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple/stm32f1/include"                                                           "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple/usb/stm32f1" "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple/usb/usb_lib" "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\cores\maple" "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\variants\generic_stm32f103c" "C:\Users\dom\AppData\Local\Temp\arduino_build_994582\sketch\Blink.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\dom\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10805 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103CB  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB  -mthumb  -march=armv7-m -D__STM32F1__ "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple" "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple/include" "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple/stm32f1/include"                                                        [... cut ...]
"C:\Users\dom\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-objcopy" -O binary  "C:\Users\dom\AppData\Local\Temp\arduino_build_994582/Blink.ino.elf" "C:\Users\dom\AppData\Local\Temp\arduino_build_994582/Blink.ino.bin"
Der Sketch verwendet 12124 Bytes (9%) des Programmspeicherplatzes. Das Maximum sind 131072 Bytes.
Globale Variablen verwenden 3088 Bytes (15%) des dynamischen Speichers, 17392 Bytes für lokale Variablen verbleiben. Das Maximum sind 20480 Bytes.

ORGATHM, 2018-03-25:
i dont know where the verbose paramter is. i hope thats enough debug info. if not, please let me know
how to set it in Sloeber.

01:06:08 **** Build of configuration Release for project BlinkTest ****
"E:\\Developement\\sloeber2\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"E:\Developement\sloeber2\/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10802 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple" "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/include" "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/stm32f1/include"                              "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/usb/stm32f1" "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/usb/usb_lib"  -I"E:\Developement\sloeber2\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.3.18\cores\maple" -I"E:\Developement\sloeber2\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.3.18\variants\generic_stm32f103c" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
'Finished building: ..\sloeber.ino.cpp'
' '
'Building file: E:\Developement\sloeber2\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.3.18\variants\generic_stm32f103c\wirish\boards.cpp'
'Starting C++ compile'
"E:\Developement\sloeber2\/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10802 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple" "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/include" "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/stm32f1/include"                              "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/usb/stm32f1" "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/usb/usb_lib"  -I"E:\Developement\sloeber2\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.3.18\cores\maple" -I"E:\Developement\sloeber2\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.3.18\variants\generic_stm32f103c" -MMD -MP -MF"core\variant\wirish\boards.cpp.d" -MT"core\variant\wirish\boards.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "E:\Developement\sloeber2\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.3.18\variants\generic_stm32f103c\wirish\boards.cpp"  -o  "core\variant\wirish\boards.cpp.o"
'Finished building: E:\Developement\sloeber2\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.3.18\variants\generic_stm32f103c\wirish\boards.cpp'
' '
'Building file: E:\Developement\sloeber2\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.3.18\variants\generic_stm32f103c\wirish\boards_setup.cpp'
'Starting C++ compile'
"E:\Developement\sloeber2\/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10802 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple" "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/include" "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/stm32f1/include"                              "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/usb/stm32f1" "-IE:\Developement\sloeber2\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/usb/usb_lib"  -I"E:\Developement\sloeber2\arduinoPl [... cut ...]
.bss                 336   536873880
.debug_aranges      3744           0
.debug_info       113243           0
.debug_abbrev      20021           0
.debug_line        35670           0
.debug_frame       11764           0
.debug_str         26184           0
.debug_loc         34756           0
.ARM.attributes       41           0
.debug_ranges       6392           0
.comment             112           0
Total             269475

'Finished building target: BlinkTest'
' '

01:06:45 Build Finished (took 36s.474ms)

jantje, 2018-03-25:
Very strange behaviour indeed.
These are the build logs I requested.  This is important because Sloeber and Arduino IDE use  the exact same build commands. There are some differences because the files are at a different location and Arduino IDE, Sloeber and the tooling I use (CDT) have their own "specific behaviour". This is why the result you have now is strange ...
 
Unfortunately the Arduino IDE log only contains the compile of the .ino file but still there is a differences that may explain the different behaviour.
The stm32F1 platform used is different between Arduino IDE and Sloeber 2018.2.26 versus 2018.3.18
Please see whether you have the same issue when using version 2018.3.18 in Arduino IDE.
If not please provide  the complete arduino IDE build log.

Best regards
Jantje

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

ORGATHM, 2018-03-27:
Hello Jantje. I found an extra verbose option inside the Arduino Ide config. So I installed Arduino IDE new, into the standard Path. And i installed a fresh Sloeber to C:/Sloeber, because you wrote about different pathes. I added my private hardwarepath and the .json file. And i switched to the 2018.2.26 in Sloeber. Unfortunely the Sting command will freeze the code on the microcontroller, exactly when the command is executed.
Compiling is no problem at all. No warnings, no nothing. I send you the new Logfiles. I cant find any option in the Arduino IDE/preferences.ini to change the Platform Date.

Maybe you have 5 Minutes to check it with the STM32 F103 C8T8 board. (The C8T8 is introduced with 64kb, but in reallity they all have 128kb flash) If you are interested, i can send you 10$ and you order it on ebay. My test skript is fast uploaded, and the LED is onboard. The only thing you need is a FTDA Serial adapter connected to PA9+PA10+GND and a Microusb cable only for the 5V Power.

Thank you very much, and i am a new, small Jantje patreon supporter.  :-) 
Best Regards,
DOMin8or
Next 2 postings, the new Logfiles

ORGATHM, 2018-03-27:
Arduino LOG, extra verbose
Looking for library 'user32'
Adding paths from jna.library.path: null
Trying user32.dll
Found library 'user32' at user32.dll
Looking for library 'shell32'
Adding paths from jna.library.path: null
Trying shell32.dll
Found library 'shell32' at shell32.dll
Looking for library 'Ole32'
Adding paths from jna.library.path: null
Trying Ole32.dll
Found library 'Ole32' at Ole32.dll
Konfiguration wird geladen...
Pakete werden initialisiert...
Boards werden vorbereitet...
Startvorgang...
Using proxy DIRECT
Using proxy DIRECT
Using proxy DIRECT
Using proxy DIRECT
Using proxy DIRECT
C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\dom\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\dom\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\dom\Documents\Arduino\libraries -fqbn=stm32duino:STM32F1:genericSTM32F103C:device_variant=STM32F103CB,upload_method=serialMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10805 -build-path C:\Users\dom\AppData\Local\Temp\arduino_build_741763 -warnings=all -build-cache C:\Users\dom\AppData\Local\Temp\arduino_cache_281094 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.stm32tools.path=C:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2018.2.26 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\dom\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\dom\Documents\Arduino\sketch_mar27b\sketch_mar27b.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\dom\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\dom\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\dom\Documents\Arduino\libraries -fqbn=stm32duino:STM32F1:genericSTM32F103C:device_variant=STM32F103CB,upload_method=serialMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10805 -build-path C:\Users\dom\AppData\Local\Temp\arduino_build_741763 -warnings=all -build-cache C:\Users\dom\AppData\Local\Temp\arduino_cache_281094 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.stm32tools.path=C:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2018.2.26 -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\dom\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\dom\Documents\Arduino\sketch_mar27b\sketch_mar27b.ino
Using board 'genericSTM32F103C' from platform in folder: C:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26
Using core 'maple' from platform in folder: C:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26
Detecting libraries used...
"C:\Users\dom\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10805 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103CB  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB  -mthumb  -march=armv7-m -D__STM32F1__ "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple" "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple/include" "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple/stm32f1/include"                                                           "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple/usb/stm32f1" "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple/usb/usb_lib" "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\cores\maple" "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\variants\generic_stm32f103c" "C:\Users\dom\AppData\Local\Temp\arduino_build_741763\sketch\sketch_mar27b.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\dom\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11  -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10805 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103CB  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB  -mthumb  -march=armv7-m -D__STM32F1__ "-IC:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2018.2.26\system/libmaple" "-IC:\Users\dom\AppD [... cut ...]
C:\Users\dom\AppData\Local\Arduino15\packages\stm32duino\tools\stm32tools\2018.2.26/win/serial_upload.bat COM5 {upload.altID} {upload.usbID} C:\Users\dom\AppData\Local\Temp\arduino_build_741763/sketch_mar27b.ino.bin
stm32flash 0.4

http://stm32flash.googlecode.com/

Using Parser : Raw BINARY
Interface serial_w32: 115200 8E1
Version      : 0x22
Option 1     : 0x00
Option 2     : 0x00
Device ID    : 0x0410 (Medium-density)
- RAM        : 20KiB  (512b reserved by bootloader)
- Flash      : 128KiB (sector size: 4x1024)
- Option RAM : 16b
- System RAM : 2KiB
Write to memory
Erasing memory
Wrote address 0x0800333c (100.00%) Done.

Starting execution at address 0x08000000... done.

jantje, 2018-03-27:
Thanks for becoming a patron
Unfortunately there is only 1 post.. which is the arduino compile.
Waiting for the second post.

ORGATHM, 2018-03-27:
okay. one moment. i think the debug is same, like before. only the path may changed.
maybe you have an idea what i have to change, on simple thing i saw, but i dont know about the effect.
the compile parameter is a little bit different. 
arm-none-eabi-g++" -c -g -Og ................ (sloeber)
arm-none-eabi-g++" -c -g -Os ................. (arduino)

Sloeber log:
16:22:57 **** Build of configuration Release for project test1 ****
"C:\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:\sloeber\/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Og -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10802 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple" "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple/include" "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple/stm32f1/include"                              "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple/usb/stm32f1" "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple/usb/usb_lib"  -I"C:\sloeber\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.2.26\cores\maple" -I"C:\sloeber\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.2.26\variants\generic_stm32f103c" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
'Finished building: ..\sloeber.ino.cpp'
' '
'Building file: C:\sloeber\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.2.26\variants\generic_stm32f103c\wirish\boards.cpp'
'Starting C++ compile'
"C:\sloeber\/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Og -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10802 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple" "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple/include" "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple/stm32f1/include"                              "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple/usb/stm32f1" "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple/usb/usb_lib"  -I"C:\sloeber\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.2.26\cores\maple" -I"C:\sloeber\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.2.26\variants\generic_stm32f103c" -MMD -MP -MF"core\variant\wirish\boards.cpp.d" -MT"core\variant\wirish\boards.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "C:\sloeber\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.2.26\variants\generic_stm32f103c\wirish\boards.cpp"  -o  "core\variant\wirish\boards.cpp.o"
'Finished building: C:\sloeber\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.2.26\variants\generic_stm32f103c\wirish\boards.cpp'
' '
'Building file: C:\sloeber\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.2.26\variants\generic_stm32f103c\wirish\boards_setup.cpp'
'Starting C++ compile'
"C:\sloeber\/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Og -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10802 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple" "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple/include" "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple/stm32f1/include"                              "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple/usb/stm32f1" "-IC:\sloeber\/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.2.26/system/libmaple/usb/usb_lib"  -I"C:\sloeber\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.2.26\cores\maple" -I"C:\sloeber\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.2.26\variants\generic_stm3 [... cut ...]
.bss                 336   536873880
.debug_aranges      3880           0
.debug_info       114009           0
.debug_abbrev      19110           0
.debug_line        33337           0
.debug_frame       12228           0
.debug_str         25681           0
.debug_loc         36970           0
.ARM.attributes       41           0
.debug_ranges       3688           0
.comment             112           0
Total             268268

'Finished building target: test1'
' '

16:23:30 Build Finished (took 32s.357ms)

ORGATHM, 2018-03-27:
maybe i change the path in soeber to the arduino IDE compiler?  if we can not find any solution, did you think about my 10$ idea? i can send it to you by paypal or patreon. so you can order one of this microcontrollers on ebay, to check what wents wrong. now, i am on a point where i need the eclipse IDE. my source is now over 1000 lines and without the outline feature it is a torture... ;'-) i can copy the source fom eclipse to arduino ide, but this not as good as using your great plugin. :-)

jantje, 2018-03-27:
Og is no optimisation and is when you select debug.
OS is small optimisation and is used when selection one of the small options
As to the board.. I do not object but I do not see how this will get us any further.
I have no specific hardware to debug so I could only confirm (or disconfirm) your blinking led perception.

jantje, 2018-03-27:
I see there are lots of S files. This might be a duplicate of #660

ORGATHM, 2018-03-27:
I read threat #660, maybe it helps? is it possible to change the path from the sloeber compiler to the arduino compiler? make this sense?

ORGATHM, 2018-03-27:
I cant find any **.s** or **.asm** files, like threat #660 
No libraries included. i found a lot auf .c and .h files.
WString,h + Wstring.cpp are looking interesting, but they include some other files
#include "itoa.h"
#include "avr/dtostrf.h"

#include <stdlib.h>
#include <string.h>
#include <ctype.h>
#include <avr/pgmspace.h>

and the **string.h** its kind of chinese for me noob arduino coder... lol

--------------
/*
 * string.h
 *
 * Definitions for memory and string functions.
 */

#ifndef _STRING_H_
#define	_STRING_H_

#include "_ansi.h"
#include <sys/reent.h>
#include <sys/cdefs.h>
#include <sys/features.h>

#define __need_size_t
#define __need_NULL
#include <stddef.h>

_BEGIN_STD_C

_PTR 	 _EXFUN(memchr,(const _PTR, int, size_t));
int 	 _EXFUN(memcmp,(const _PTR, const _PTR, size_t));
_PTR 	 _EXFUN(memcpy,(_PTR __restrict, const _PTR __restrict, size_t));
_PTR	 _EXFUN(memmove,(_PTR, const _PTR, size_t));
_PTR	 _EXFUN(memset,(_PTR, int, size_t));
char 	*_EXFUN(strcat,(char *__restrict, const char *__restrict));
char 	*_EXFUN(strchr,(const char *, int));
int	 _EXFUN(strcmp,(const char *, const char *));
int	 _EXFUN(strcoll,(const char *, const char *));
char 	*_EXFUN(strcpy,(char *__restrict, const char *__restrict));
size_t	 _EXFUN(strcspn,(const char *, const char *));
char 	*_EXFUN(strerror,(int));
size_t	 _EXFUN(strlen,(const char *));
char 	*_EXFUN(strncat,(char *__restrict, const char *__restrict, size_t));
int	 _EXFUN(strncmp,(const char *, const char *, size_t));
char 	*_EXFUN(strncpy,(char *__restrict, const char *__restrict, size_t));
char 	*_EXFUN(strpbrk,(const char *, const char *));
char 	*_EXFUN(strrchr,(const char *, int));
size_t	 _EXFUN(strspn,(const char *, const char *));
char 	*_EXFUN(strstr,(const char *, const char *));

#ifndef _REENT_ONLY
char 	*_EXFUN(strtok,(char *__restrict, const char *__restrict));
#endif

size_t	 _EXFUN(strxfrm,(char *__restrict, const char *__restrict, size_t));

#ifndef __STRICT_ANSI__
char 	*_EXFUN(strtok_r,(char *__restrict, const char *__restrict, char **__restrict));

int	 _EXFUN(bcmp,(const void *, const void *, size_t));
void	 _EXFUN(bcopy,(const void *, void *, size_t));
void	 _EXFUN(bzero,(void *, size_t));
int	 _EXFUN(ffs,(int));
char 	*_EXFUN(index,(const char *, int));
_PTR	 _EXFUN(memccpy,(_PTR __restrict, const _PTR __restrict, int, size_t));
_PTR	 _EXFUN(mempcpy,(_PTR, const _PTR, size_t));
_PTR	 _EXFUN(memmem, (const _PTR, size_t, const _PTR, size_t));
_PTR 	 _EXFUN(memrchr,(const _PTR, int, size_t));
_PTR 	 _EXFUN(rawmemchr,(const _PTR, int));
char 	*_EXFUN(rindex,(const char *, int));
char 	*_EXFUN(stpcpy,(char *__restrict, const char *__restrict));
char 	*_EXFUN(stpncpy,(char *__restrict, const char *__restrict, size_t));
int	 _EXFUN(strcasecmp,(const char *, const char *));
char	*_EXFUN(strcasestr,(const char *, const char *));
char 	*_EXFUN(strchrnul,(const char *, int));
#endif
#if !defined(__STRICT_ANSI__) || (_XOPEN_SOURCE - 0) >= 500
char 	*_EXFUN(strdup,(const char *));
#endif
#ifndef __STRICT_ANSI__
char 	*_EXFUN(_strdup_r,(struct _reent *, const char *));
#endif
#if !defined(__STRICT_ANSI__) || (_XOPEN_SOURCE - 0) >= 700
char 	*_EXFUN(strndup,(const char *, size_t));
#endif
#ifndef __STRICT_ANSI__
char 	*_EXFUN(_strndup_r,(struct _reent *, const char *, size_t));
/* There are two common strerror_r variants.  If you request
   _GNU_SOURCE, you get the GNU version; otherwise you get the POSIX
   version.  POSIX requires that #undef strerror_r will still let you
   invoke the underlying function, but that requires gcc support.  */
#ifdef _GNU_SOURCE
char    *_EXFUN(strerror_r,(int, char *, size_t));
#else
# ifdef __GNUC__
int      _EXFUN(strerror_r,(int, char *, size_t)) __asm__ (__ASMNAME ("__xpg_strerror_r"));
# else
int      _EXFUN(__xpg_strerror_r,(int, char *, size_t));
#  define strerror_r __xpg_strerror_r
# endif
#endif
size_t	 _EXFUN(strlcat,(char *, const char *, size_t));
size_t	 _EXFUN(strlcpy,(char *, const char *, size_t));
int	 _EXFUN(strncasecmp,(const char *, const char *, size_t));
size_t	 _EXFUN(strnlen,(const char *, size_t));
char 	*_EXFUN(strsep,(char **, const char *));
char	*_EXFUN(strlwr,(char *));
char	*_EXFUN(strupr,(char *));
#ifndef DEFS_H	/* Kludge to work around problem compiling in gdb */
char  *_EXFUN(strsignal, (int __signo));
#endif
#ifdef __CYGWIN__
int     _EXFUN(strtosigno, (const char *__name));
#endif

/* Recursive version of strerror.  */
char *	_EXFUN(_strerror_r, (struct _reent *, int, int, int *));

#if defined _GNU_SOURCE && defined __GNUC__
#define strdupa(__s) \
	(__extension__ ({const char *__in = (__s); \
			 size_t __len = strlen (__in) + 1; \
			 char * __out = (char *) __builtin_alloca (__len); \
			 (char *) memcpy (__out, __in, __len);}))
#define strndupa(__s, __n) \
	(__extension__ ({const char *__in = (__s); \
			 size_t __len = strnlen (__in, (__n)) + 1; \
			 char *__out = (char *) __builtin_alloca (__len); \
			 __out[__len-1] = '\0'; \
			 (char *) memcpy (__out, __in, __len-1);}))
#endif /* _GNU_SOURCE && __GNUC__ */

/* These function names are used on Windows and perhaps other systems.  */
#ifndef strcmpi
#define strcmpi strcasecmp
#endif
#ifndef stricmp
#define stricmp strcasecmp
#endif
#ifndef strncmpi
#define strncmpi strncasecmp
#endif
#ifndef strnicmp
#define strnicmp strncasecmp
#endif

#endif /* ! __STRICT_ANSI__ */

#include <sys/string.h>

_END_STD_C

#endif /* _STRING_H_ */

jantje, 2018-03-28:
Maybe you have to many libraries. Can you delete all the libraries in the libraries folder? If they are needed they will reappear (some builds may be required)

ORGATHM, 2018-03-28:
i see no libraries in the left tree.
but a attention message, i cant understand. it seems to me as a wrong interpreted command?

**Type 'String' could not be resolved**
 What does it mean? I

[image]

ORGATHM, 2018-03-28:
after pressing cleaning and rebuild index, its gone, now it show the class.
but it freezes if it runs to the line.

[image]

ORGATHM, 2018-03-28:
i am totally down. i only need a better gui/ide then arduino IDE, i`ve testet platformIO, eclipse and so on.
this costs me 1 week of frustration. i am 7 days behind my schedule. i think i will code in eclipse
and copy paste it to arduino IDE. there is no finish line in sight. 
thank you for your assistance, i think we can close this threat.
maybe, in future comes an update and everything works fine, but i have no more time to waste, for
hunting a IDE feature like outline.
i like your plugin so much, but i cant use it. its frustrating.
best regards,
DOM 

[image]

ORGATHM, 2018-03-29:
i found an interesting thing...
if i set a text string like this: **const char* text = "this is my text";**
it does workwithout freezing and does the same like: **String text="this is my text";**

Maybe this helps to find the failure. Maybe its inside the string class?

jantje, 2018-03-29:
The string is green here with me as well.
But I do not get the warning. I would assume this warning to be an error and a explanation of the behaviour you see. I always go for 0 errors and 0 warnings :-)
Are you sure you didn't change any of the core files? You should be able to tell based on the date/times on your file system.

ORGATHM, 2018-03-29:
Hello Jantje, thanks for fast reply.
Yesterday I was so frustrated, i deleted all eclipse stuff. But without eclipse it isn`t an option.
So i thought i code inside eclipse and copy paste it to Arduino IDE... Just test the 32 Bit Version...
Then i downloaded the 32 Bit version 4.2 from yor server. (Before I used the 64bit Version)
 (Nightly updates are not working for me. There is a .ini missing in the JNE folder, if i copy the nightly build over the existing files)
Okay, On C:\ Sloeber will not start anymore. A black window appears and the Log said, all .JSON downloads not possible, "TIME OUT". Grande Catastrophe...
Starting an starting over, fresh unzip copied to C:/, allways no Plugins download, "TIME OUT".
I cant belive what is going on, so I startet an old installed version on Drive E:. It works perfekt.
okay, everything back to start. New folder e:/sloeber createt, and inserted a clean 32 bit sloeber.
inserted the stm boards .json link. waiting for download the addon stuff.
my arduino ide (1.8.5) is now, fresh installed inside c:/Program Files (x86)\Arduino
then i added the Private Library Path: 
C:\Users\dom_is_my_profile_name\Arduino\libraries and adding 
THE PRECONFIGURED PATH DOES NOT CONTAINS ANY ARDIUNO STUFF.....
C:/Users/dom/Arduino/libraries
AND IT WAS CONFIGURED WITH SLASHES , but if i use [NEW] button, and select a new 
path with the file explorer dialog, it creates a path with BACKSLASHES... 
(Seems confusing for me...)
Then i deleted the preconfigured entry inside Private hardware path...
to existing C:\Program Files(x86)\Arduino\hardware\avr
inside this folder is: boards.txt, platform.txt, programmers.txt

[image]

[image]

ORGATHM, 2018-03-29:
maybe i delete the old log files, they are outdated and very long and my samsung tablet freezes, if i go to this threat... :-)
if you like, i can make new ones. and if you want i can upload it to pastebin.

jantje, 2018-03-29:
>Then i deleted the preconfigured entry inside Private hardware path...
to existing C:\Program Files(x86)\Arduino\hardware\avr

Don't do this.   ever!!

ORGATHM, 2018-03-30:
okay, i got this from stm32duino site. 
mostly it works, only String command does not work...
what path i have to locate?
thanks for assistance :-)

ORGATHM, 2018-03-30:
the path variable is red... what does it mean?

[image]

jantje, 2018-03-30:
If you refer to these http://wiki.stm32duino.com/index.php?title=Arduino_with_Eclipse installation instructions... the last update is from 2016!!!! And even then they were outdated.
The bottom part states these  are instructions fro V2 (not even V3)
Talking about crappy documentation :-(

jantje, 2018-03-30:
I'm not sure what you are doing but you miss Sloeber basics.
It is all lots simpler than what you are doing.
You do not need to set private hardware nor private libraries
Just install the boards as you did in arduino IDE
Maybe you should watch some of  my video's to get the basics  https://www.youtube.com/watch?v=zc2JLbtbI7Q&list=PLCSjf7QmP3TDPSjT_5UIp0DHXzKdhIQsF
And please update the page to reflect V2 V3 at the top http://wiki.stm32duino.com/index.php?title=Arduino_with_Eclipse

ORGATHM, 2018-03-30:
now i am more then 5 days installing sloeber, over and over and over. more then 30 times...
i am so desperated, so i took all little hints & options and outdated infos from the internet.

now i installed a fresh windows into VM-Ware, 
installed arduino 1.8.5 to standart path, installed the board repo (.json)
installed java (JavaSetup8u161.exe)
unziped io.sloeber.product-4.2.0-20180116.154118-36-win32.win32.x86_64.tar.gz to c:/sloeber
installed .json repo as 3d party
created new arduino project, selected STM32 F103C board with date 28.2..
selected board and flash methode, nothing else configured.
i think as simple is that, and thats the right way. it compiles and flashes, so i am not wrong.

and it does work....., like before.
most skripts are running, only if the command: String myRandomVar ="text"; comes in the source it does freeze.
it makes no difference if i use the 4.2 stable release or the nightly.
and the nightly will not start on my master machine. there comes a not nice java error.
(i have java deinstalled and fresh installed, but the error comes again and again.)
in vm ware the nightly  works well, but freezes at String, like all the time with the stable version.
i think sloeber is great for noobs+pros, they want a better arduino ide. and for the most guys it is.
but with my STM F103 C8T8 it freezes. i wont believe it and i give one try on another. 5 days & nights. but it does not work. i hoped it so much, but i can`t do even more, then use a fresh windows
and install & testing it, over and over again.  

now i have seen a lot of your movies and it teaches me a lot of deeply hidden features.
its a master piece of software, you have done. now i am more interested then before to use Sloeber... :-)
but i learned, that Sloeber is greatfully preconfigured and it does everything automaticly.
so the failure can not be on my side :-) good to know, but i cant see the finishline and it makes me
sad. i did know that eclipse is good and sloeber makes it much better, but what is it usefull for, if 
i cant use it with my microcontroller, spezially with my own code that contains the String command? i am one little step before, but without the String command it is absolutely not usable for me. i become gray hairs about it. :-) so near, and not close enough...
i think i write into stm32 forums, where are active stm32 f103 users, they can proof the scenario. 
unfortunely i see no other way.
thanks for the time you spent me and my problem.
Sloeber rulez! :-))

jantje, 2018-03-30:
>now i installed a fresh windows into VM-Ware,

Not sure why you want to use a vm as sloeber runs on windows mac and linux
>installed arduino 1.8.5 to standart path, installed the board repo (.json)

This is not needed
>installed java (JavaSetup8u161.exe)
:-) good
Note that this is not necessary when you install a release. Sloeber releases come with a jre

> unziped io.sloeber.product-4.2.0-20180116.154118-36-win32.win32.x86_64.tar.gz to c:/sloeber
> installed .json repo as 3d party
> created new arduino project, selected STM32 F103C board with date 28.2.. [...]

This sounds like it should be.

>i think i write into stm32 forums, where are active stm32 f103 users, they can proof the scenario.
unfortunely i see no other way.

I think that is a good idea. As we have seen, Sloeber uses the same build commands as Arduino IDE so there must be a problem with "files that should not be part of the build" or linking problems. There should be more knowledge about that on the stm32 forums.

>so the failure can not be on my side 

I have been wrong before but at this point in time I agree with this. This is also why I think me having the same hardware will not help and why I think the stm32 forums is the place to ask questions.
Please post the link so I can read/discuss along.

ORGATHM, 2018-03-30:
thanks for reply.
i tested with a fresh new windows inside the vm ware, to eleminate every possible missconfiguration on my machine.
after 30 rounds of installing Sloeber and some other IDE`s , it seems my Java does not exactly what it has to to. I wrote you about the error.

if i unzip the nightly to C:/ on my master machine it does not start. SAME zip file in VM works properly.

[image]

After deleting c:/sloeber and unzip the stable 32bit bundle, no plugin download possible...
new error....
[image]

and the best thing is,..... Sloeber runs on drive E:/developement/sloeber without any problems.
this was too chaotic for me, so i used a fresh, clean windows in VM-WARE.
after 1 week testint Sloeber, atom + platformIO, Arduino Ide 1.8.5 + 1.9.0, System Workbench, and a lot more programs, something with my $path went wrong, or what ever is wrong.
its catastrophic. the only thing i nees, is a better ardiuno IDE. nothing more. just an IDE... :-)

jantje, 2018-03-31:
According to this stackoverflow post running natively doesn't work due to incompatible java runtime https://stackoverflow.com/questions/7352493/failed-to-load-the-jni-shared-library-jdk

>After deleting c:/sloeber and unzip the stable 32bit bundle, no plugin download possible...

As all json downloads fail I assume you have told windows that the java runtime is not allowed to access the internet/ You could disable your firewall and start sloeber to see if that assumption is true.

Note that the stable contains a jre folder. This is the java runtime included with sloeber and when you have a jre subfolder that one gets used.

>after 1 week testint Sloeber, atom + platformIO, Arduino Ide 1.8.5 + 1.9.0, System Workbench, and a lot more programs,

IMHO you have been trying to many things and missed some basic things. Add to that the "security features" of windows 10 and 32bit versus 64 bit; and the tool providers are off for "sitting with a blind man trying to describe yellow". Not a nice place for any of us ;-)
Really no offence because I know that when I am in this situation; I start mixing stuff from different versions and tools. 
My advice: stick to one version (like sloeber V4.2 stable 64 bit on windows native) to get it running. Do standard things first (like compile and upload to blink to the uno) and gradually add to get to your target environment.
Also stick to one source of information. There are many ways to skin a cat. Mixing ways is "advanced". IMHO you are not up  to that yet.

ORGATHM, 2018-03-31:
the java works, and here are no firewall restrictions. and i am not a noob at all. :-)

i do not have experience with gcc and c, but i have more then 30 years expirence with 6510 assembler, basic, html, php, mysql, apache, ms dos+batch, win 3.1, win95, win 2000, win7, win8, win10, small linux, photoshop, and now i am learning c and microcontroller programming. its not my first rodeo. 
it is only a new language and enviroment. and my project works very well. it is coded from the ground without any copy pasted code.
the only thing what brakes me out, is the search for a better UI/IDE.

"sitting with a blind man trying to describe yellow" hurts and insults me.
i have no bookware about microcontrollers and gcc... that could be a problem. the intenet is
full of information. but it is not as easy to find the right informations... :-)

"My advice: stick to one version (like sloeber V4.2 stable 64 bit on windows native) to get it running. "
it is running, from the beginning. the problem is the String declaration. its not my fault.

[image]

ORGATHM, 2018-03-31:
i found an other board definition package, copyright seems directly from stm for arduino. 
https://raw.githubusercontent.com/stm32duino/BoardManagerFiles/master/STM32/package_stm_index.json
the gcc call looks years newer... arm-none-eabi-gcc/6-2017-q2-update/bin/arm-none-eabi-g++
in my old logs i see: arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g+
maybe this could fix the problem, a newer gcc...
i`ve tested it in the arduino ide and it works. now the device has the codename bluepill, thats great.
buuuut.... after installing it into Sloeber, Sloeber cant cook any source code.
i start new project, select the new device named bluepill, and selected everything right, an error occurs.
maybe its a path problem?  "C:\sloeber\/arduinoPlugin/..........................................."
backslash follewd by slash? thats different to my previous log files... 

19:33:24 **** Build of configuration Release for project clock ****
"E:\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"E:\sloeber\/arduinoPlugin/packages/STM32/tools/arm-none-eabi-gcc/6-2017-q2-update/bin/arm-none-eabi-g++" -mcpu=cortex-m3 -mthumb @C:\Users\dom\Documents\sloeber-workspace\clock/Release/sketch/build_opt.h -c -Os -Wall -Wextra -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-I" "-IE:\sloeber\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/cores/arduino/avr" "-IE:\sloeber\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/cores/arduino/stm32" "-IE:\sloeber\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/Drivers/STM32F1xx_HAL_Driver/Inc/" "-IE:\sloeber\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/Drivers/STM32F1xx_HAL_Driver/Src/" "-IE:\sloeber\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/STM32F1xx/" "-IE:\sloeber\arduinoPlugin\packages\STM32\hardware\stm32\1.2.0\variants\BLUEPILL_F103C8/usb" "-IE:\sloeber\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IE:\sloeber\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -DSTM32F1xx -DARDUINO=10802 -DARDUINO_BLUEPILL_F103C8 -DARDUINO_ARCH_STM32 -DSTM32F103xB -DNO_HWSERIAL "-IE:\sloeber\/arduinoPlugin/packages/STM32/tools/CMSIS/5.3.0/CMSIS/Core/Include/" "-IE:\sloeber\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/" "-IE:\sloeber\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/"  -I"E:\sloeber\arduinoPlugin\packages\STM32\hardware\stm32\1.2.0\cores\arduino" -I"E:\sloeber\arduinoPlugin\packages\STM32\hardware\stm32\1.2.0\variants\BLUEPILL_F103C8" -I"E:\sloeber\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.2.26\libraries\Adafruit_GFX_AS" -I"E:\sloeber\arduinoPlugin\packages\stm32duino\hardware\STM32F1\2018.2.26\libraries\Adafruit_ILI9341_STM" -I"E:\sloeber\arduinoPlugin\packages\STM32\hardware\stm32\1.2.0\libraries\SPI" -I"E:\sloeber\arduinoPlugin\packages\STM32\hardware\stm32\1.2.0\libraries\SPI\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
arm-none-eabi-g++: error: @C:\Users\dom\Documents\sloeber-workspace\clock/Release/sketch/build_opt.h: Invalid argument
subdir.mk:31: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1

19:33:25 Build Finished (took 948ms)

its very interesting, how much errors i can produce, with a full automatic tool. :-))
(btw. all plugins downloaded without problems...no firewall or problems like this)
**** HAPPY EASTER***
:-)

jantje, 2018-04-01:
the bluepill thing is issue #927 
Creating the file c:/Users/dom/Documents/sloeber-workspace/stmduinotest1/Release/sketch/build_opt.h may help but I fear on windows the @ causes issues as well

I'm not sure which repository you used for the esp32 (which has a json by now as well) but in the one I use in the video platform.txt contains following line
`runtime.tools.xtensa-esp32-elf-gcc.path={runtime.platform.path}/tools/xtensa-esp32-elf`
yours seems to be missing this one (or something like it)

ORGATHM, 2018-04-01:
hello Jantje,
after having no success with all the bluepill stuff aka STM32 f103 (the mikrocontroller i am using), i need a little success. and to verify the String problem on an other platform, i connected my ESP32 and installed it the .git way. seen in your esp32 video.   https://youtu.be/szRrvJPvQz4
( https://github.com/espressif/arduino-esp32.git )
**with this success**, after your great but tricky manual, i feel better and hopeful, that the git methode will work for arduino-STM also.
thats the reason why i deleted this two arduino stm .json imports. and imported/cloned this two arduino stm git repos.
same same, but different, i hoped... lol 
https://github.com/stm32duino/Arduino_Core_STM32.git
https://github.com/rogerclarkmelbourne/Arduino_STM32.git
inside the Arduino_Core_STM32.git, is a 2017 gcc compiler, not the 2014. my hope is, maybe, this helps for my String declaration problem.

**your hint with sketch/build_opt.h works.** now it compiles the source. *great*. unbeleavable. for real, how could i imagine to insert a blank textfile into a not existing folder?

FINALY, IT WORKS! String declaration WORKS. *YOU ARE THE BEST* no more freezing.
one little problem, i got. but i could fix it by my self.
after compiling, it was not possible to flash the recipe. could not found path to stm tools.
so i deflated the windows tools from E:\sloeber\arduinoPlugin\downloads, by hand 
into: E:\sloeber\git-import\arduino_core_stm32\tools

in short:
import https://github.com/stm32duino/Arduino_Core_STM32.git
mkdir Project/Release/sketch/build_opt.h
unzip tools to the .github package. 

but after compiling, the build_opt.h is gone... 
is it possible to fix that an other way? who calls build_opt.h? maybe i can kick this line?
or insert a little skript, that creates the build_opt.h automaticly?

ORGATHM, 2018-04-01:
now i have little problems to auto integrate Libraries...
if i delete folder Libraries, only the SPI lib re-integrates, but i included some more in my source.
#include <SPI.h>                         // this will auto add
#include <OneWireSTM.h>        // no import
#include <DHT.h>                      // no import
#include <Adafruit_GFX_AS.h>    // no import
#include <Adafruit_ILI9341_STM.h>  // no import

arduino IDE downloaded it to: C:\Users\dom\Documents\Arduino\libraries

jantje, 2018-04-01:
>your hint with sketch/build_opt.h works. now it compiles the source. great. unbeleavable. for real, how could i imagine to insert a blank textfile into a not existing folder?

the command line states `@c:/Users/dom/Documents/sloeber-workspace/stmduinotest1/Release/sketch/build_opt.h `
And the error states `Das System kann den angegebenen Pfad nicht finden.`

>but after compiling, the build_opt.h is gone...
is it possible to fix that an other way? who calls build_opt.h? maybe i can kick this line?
or insert a little skript, that creates the build_opt.h automaticly?

Sloeber assumes everything in the release folder to be "created resources" as such these get deleted when Sloeber or CDT thinks it is necessary.
I really do not know why the file build_opt.h is needed nor why it is in the gcc command with a @ and for sure not what the reasoning is to put this file in the build location.
I do see in the platform.txt
```
compiler.extra_flags=-mcpu={build.mcu} -mthumb @{build.opt.path}
build.opt.path={build.path}/sketch/{build.opt.name}
```
Which explain the existence of this file in the gcc command. There is also
`recipe.hooks.prebuild.1.pattern.windows=cmd /c "if not exist {build.opt.sourcepath} mkdir {build.path}\sketch & type NUL > {build.opt.path}"`
In the platform.txt.
I knew all this because I try to compile all boards in the third party boards.
This board failed so I made the issue.

to work around this: add following command to project properties->C/C++build->settings->build steps->pre build steps->command
`cmd /c "if not exist  sketch mkdir sketch & type NUL > sketch/build_opt.h`

jantje, 2018-04-01:
>#include <Adafruit_GFX_AS.h> // no import
#include <Adafruit_ILI9341_STM.h> // no import

Automatic install and import in the project is only supported by for libraries that have a foldername = includename-.h
Adafruit does not follow this convention so needs to be installed and imported manually
install arduino->preferences
import see http://eclipse.baeyens.it/how_to.shtml#/lib the V2 and V3 way

ORGATHM, 2018-04-01:
thank you so much for your assitance. great. actually the adafruits importing, only the DHT makes a bit trouble, but i follow your manual.

ORGATHM, 2018-04-03:
hello jantje.
the,  cmd /c "if not exist sketch mkdir sketch & type NUL > sketch/build_opt.h"
does not work as precompile command.
in the windows CMD, copy pasted, it works.
its no big deal. i created a batch file. one click, and its done. but why does it not work as precompile command?  tested wit 4.2 release and the nightly (not updated) from your download server.

my batch started from inside the project directory and works. its not the coolest, but a working way.
cmd /c "if not exist .\Release\sketch md .\Release\sketch & type NUL > .\Release\sketch\build_opt.h"

actually i am fighting with the libs. the optimized STM32 libs, for using DMA GFX lib, calls something that is not inside the <SPI.h>. tested with 3 different SPI.h libs ... later i search deeper inside the libs for the dependencies and calls.
with arduino IDE it works well, without any configuring....... :-) its allways fun :-)
the easiest way, would be to use the arduino IDE managed libraries :-)

best regards,
DOM

jantje, 2018-04-03:
>the, cmd /c "if not exist sketch mkdir sketch & type NUL > sketch/build_opt.h"
does not work as precompile command.

I tested it here and it did work for me. It is pure CDT functionality (nothing to do with Sloeber) heavily used by lots of users.

>the easiest way, would be to use the arduino IDE managed libraries :-)
you can not point to Arduino managed libraries because arduino IDE only extracts the zips when building.
You can however turn on verbose in arduino IDE and see what libraries are being used. This story and the String issue look as if something is(not) linked in correctly.

ORGATHM, 2018-04-03:
its unbeleavable, here it worked one time, too. i have nothing changed. 
after deleting the sketch file, and test again....

22:50:19 **** Build of configuration Release for project STM32coreTest1 ****
"E:\\SloeberX64n\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"E:\SloeberX64n\/arduinoPlugin/packages/STM32/tools/arm-none-eabi-gcc/6-2017-q2-update/bin/arm-none-eabi-g++" -mcpu=cortex-m3 -mthumb @E:\SloeberX64n\sloeber-workspace\STM32coreTest1/Release/sketch/build_opt.h -c -Os -Wall -Wextra -std=gnu++14 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-I" "-IE:\SloeberX64n\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/cores/arduino/avr" "-IE:\SloeberX64n\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/cores/arduino/stm32" "-IE:\SloeberX64n\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/Drivers/STM32F1xx_HAL_Driver/Inc/" "-IE:\SloeberX64n\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/Drivers/STM32F1xx_HAL_Driver/Src/" "-IE:\SloeberX64n\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/STM32F1xx/" "-IE:\SloeberX64n\arduinoPlugin\packages\STM32\hardware\stm32\1.2.0\variants\BLUEPILL_F103C8/usb" "-IE:\SloeberX64n\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Inc" "-IE:\SloeberX64n\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/Middlewares/ST/STM32_USB_Device_Library/Core/Src" -DSTM32F1xx -DARDUINO=10802 -DARDUINO_BLUEPILL_F103C8 -DARDUINO_ARCH_STM32 -DSTM32F103xB  "-IE:\SloeberX64n\/arduinoPlugin/packages/STM32/tools/CMSIS/5.3.0/CMSIS/Core/Include/" "-IE:\SloeberX64n\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Include/" "-IE:\SloeberX64n\/arduinoPlugin/packages/STM32/hardware/stm32/1.2.0/system/Drivers/CMSIS/Device/ST/STM32F1xx/Source/Templates/gcc/"  -I"E:\SloeberX64n\arduinoPlugin\packages\STM32\hardware\stm32\1.2.0\cores\arduino" -I"E:\SloeberX64n\arduinoPlugin\packages\STM32\hardware\stm32\1.2.0\variants\BLUEPILL_F103C8" -I"E:\SloeberX64n\arduinoPlugin\packages\STM32\hardware\stm32\1.2.0\libraries\SPI\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
arm-none-eabi-g++: error: @E:\SloeberX64n\sloeber-workspace\STM32coreTest1/Release/sketch/build_opt.h: Invalid argument
subdir.mk:31: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1

22:50:20 Build Finished (took 978ms)

its funny. i dont have to understand this. :-)

ORGATHM, 2018-04-03:
apropos not linked in... at the end of installing the .json repo, a error occours....
( i know, i am annoying... lol.)

[image]

jantje, 2018-04-04:
It would be cool to see if the nightly fixes you String issue now 951 is fixed.
you do need to delete the folder [eclipseinstall]/arduinoplugin/packages/STM32 and reinstall the package in Sloeber after the update for the fix to work.

ORGATHM, 2018-04-04:
the problem is, i tried 3 different STM32duino projects... all has problems.

STM32 Arduino PROJECT 1

https://github.com/stm32duino/BoardManagerFiles/raw/master/STM32/package_stm_index.json
located in:
 **this STM32 package does work with "String" declaration... all the time...**
it contains a different core (arduino) and a gcc from 2017, this makes trouble with hSPI, and DMA activated GFX libraries... 
it was only the error message at the end while installing, **but it works. String declaration works**
here it seems, this package has a different core, and the gfx libraries will not work with the integrated SPI lib. selecting an other SPI is not possible, because it needs an other core, with MAPLE libs.

after reading your last posting, i reinstalled the .json, the error occours like before.
OK
OK
OK
Failed to install STM32 Cores
Failed to extract tar.bz2.E:/SloeberX64n/arduinoPlugin/downloads/STM32Tools-1.0.2.tar.bz2
Cannot run program "ln": CreateProcess error=2, Das System kann die angegebene Datei nicht finden
-----------------------------------------------------------------------------------------------

STM32 Arduino PROJECT 2

there is one more arduino stm git...
https://github.com/rogerclarkmelbourne/Arduino_STM32.git
located in: E:\SloeberX64n\arduinoPlugin\packages\Arduino_STM32
importing is no problem, but whe i compile,....
23:33:43 **** Build of configuration Release for project arduinoSTM32test3 ****
"E:\\SloeberX64n\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"/bin/arm-none-eabi-g++" -c -g -Os -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10802 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32F1__ "-IE:\SloeberX64n\/arduinoPlugin/packages/Arduino_STM32/STM32F1/system/libmaple" "-IE:\SloeberX64n\/arduinoPlugin/packages/Arduino_STM32/STM32F1/system/libmaple/include" "-IE:\SloeberX64n\/arduinoPlugin/packages/Arduino_STM32/STM32F1/system/libmaple/stm32f1/include" "-IE:\SloeberX64n\/arduinoPlugin/packages/Arduino_STM32/STM32F1/system/libmaple/usb/stm32f1" "-IE:\SloeberX64n\/arduinoPlugin/packages/Arduino_STM32/STM32F1/system/libmaple/usb/usb_lib"  -I"E:\SloeberX64n\arduinoPlugin\packages\Arduino_STM32\STM32F1\cores\maple" -I"E:\SloeberX64n\arduinoPlugin\packages\Arduino_STM32\STM32F1\variants\generic_stm32f103c" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
Das System kann den angegebenen Pfad nicht finden.
subdir.mk:31: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1

23:33:43 Build Finished (took 559ms)

here it could not find the path of ......... ???? may this works, but i cant compile...

---------------------------------------------------------------------------------------------

STM32 Arduino PROJECT 3 (THIS IS THE .JSON REPO I WOULD PREFER)

http://dan.drown.org/stm32duino/package_STM32duino_index.json
located in: E:\SloeberX64n\arduinoPlugin\packages\stm32duino

with this stm32duino .json repo i startet on arduno ide, and it worked perfectly.

works with hSPI, DMA optimzed GFX llibs. 
**but this project has the String declaration & freezing problem in sloeber**. 
with arduino ide this repo works perfectly.
btw. this project uses libmaple as core.
maybe i make an issue threat at dan.drown, but i cant find dan drown on github.

ORGATHM, 2018-04-04:
i think a make a couple of days a break with the bluepill... i need blue pills for my headache.
after nearly 2 weaks of frustrating, i need success. :-)  i think i will check the next days
the esp8266 and the esp32. they has got more flash disk, wifi, and so on.
for cheap projects, the bluepill is a great and extremly cheap device, the hSPI works so fast with ILI9341.
but i give up. its not in my nature, but i am completely done... :-)

jantje, 2018-04-05:
Bleading edge feelings ;-)

ORGATHM, 2018-04-06:
its a bit offtopic, sorry.
i ported my project to esp32. works now with hardware spi and the ili9341 runs like a ferrari.
i better changed two weeks ago to esp32... lol. the wifi and dualcore opens a lot of future features.
but one thing looks strange to me, i cant find the option to deactivate this debug garbage.
the bin file is extremly big, because of a lot debug infos? megabytes of debug infos, i dont need.
on the stm the bin was ~50 kb, inclusive 7 fonts for the display. the fonts took the most of the 50kb...

'Building target: ESP32_myHumi'
'Printing size:'
"E:\SloeberX64n\/arduinoPlugin/packages/arduino-esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-size" -A "E:/SloeberX64n/sloeber-workspace/ESP32_myHumi/Release/ESP32_myHumi.elf"
E:/SloeberX64n/sloeber-workspace/ESP32_myHumi/Release/ESP32_myHumi.elf  :
section                                                         size         addr
.rtc.text                                                          0   1074528256
.iram0.vectors                                                  1024   1074266112
.iram0.text                                                    37096   1074267136
.dram0.data                                                     8768   1073479680
.dram0.bss                                                      2680   1073488448
.flash.rodata                                                  34180   1061158944
.flash.text                                                   114637   1074593816
.debug_frame                                                   63008            0
.debug_info                                                  1059781            0
.debug_abbrev                                                 147175            0
.debug_loc                                                    396858            0
.debug_aranges                                                 25160            0
.debug_ranges                                                  26512            0
.debug_macro                                                  161095            0
.debug_line                                                   540675            0
.debug_str                                                    839290            0
.comment                                                         419            0
.xtensa.info                                                      56            0
.xt.lit._ZN16Adafruit_ILI9341D5Ev                                  0            0
.xt.prop._ZN16Adafruit_ILI9341D5Ev                                 0            0
.xt.prop._ZN16Adafruit_ILI9341D2Ev                                36            0
.xt.prop._ZN16Adafruit_ILI9341D0Ev                                36            0
.xt.prop._ZTV16Adafruit_ILI9341                                   12            0
.xt.prop._ZTV12Adafruit_GFX                                       12            0
.xt.lit._ZN14HardwareSerialD5Ev                                    0            0
.xt.prop._ZN14HardwareSerialD5Ev                                   0            0
.xt.prop._ZN14HardwareSerialD2Ev                                  36            0
.xt.prop._ZN14HardwareSerialD0Ev                                  36            0
.xt.prop._ZTV14HardwareSerial                                     12            0
.xt.lit._ZN9IPAddressD5Ev                                          0            0
.xt.prop._ZN9IPAddressD5Ev                                         0            0
.xt.prop._ZN9IPAddressD2Ev                                        36            0
.xt.prop._ZN9IPAddressD0Ev                                        36            0
.xt.prop._ZTV9IPAddress                                           12            0
.xt.lit._ZN5Print5writeEPKc                                        8            0
.xt.prop._ZN5Print5writeEPKc                                      60            0
.xt.prop._ZTISt9exception                                         12            0
.xt.prop._ZTISt9bad_alloc                                         12            0
.xt.prop._ZTVN10__cxxabiv117__class_type_infoE                    12            0
.xt.lit._ZNK9__gnu_cxx24__concurrence_lock_error4whatEv            8            0
.xt.lit._ZNK9__gnu_cxx26__concurrence_unlock_error4whatEv          8            0
.xt.lit._ZN9__gnu_cxx24__concurrence_lock_errorD5Ev                8            0
.xt.lit._ZN9__gnu_cxx26__concurrence_unlock_errorD5Ev              8            0
.xt.lit._ZN9__gnu_cxx7__mutex4lockEv                               8            0
.xt.lit._ZN9__gnu_cxx13__scoped_lockD5Ev                           8            0
.xt.prop._ZNK9__gnu_cxx24__concurrence_lock_error4whatEv          48            0
.xt.prop._ZNK9__gnu_cxx26__concurrence_unlock_error4whatEv        48            0
.xt.prop._ZN9__gnu_cxx24__concurrence_lock_errorD5Ev              12            0
.xt.prop._ZN9__gnu_cxx26__concurrence_unlock_errorD5Ev            12            0
.xt.prop._ZN9__gnu_cxx7__mutex4lockEv                             60            0
.xt.prop._ZN9__gnu_cxx13__scoped_lockD5Ev                         12            0
.xt.prop._ZN9__gnu_cxx24__concurrence_lock_errorD2Ev              36            0
.xt.prop._ZN9__gnu_cxx26__concurrence_unlock_errorD2Ev            36            0
.xt.prop._ZN9__gnu_cxx24__concurrence_lock_errorD0Ev              36            0
.xt.prop._ZN9__gnu_cxx26__concurrence_unlock_errorD0Ev            36            0
.xt.prop._ZN9__gnu_cxx13__scoped_lockD2Ev                         72            0
[... cut ...]
.xt.prop._ZTIN9__gnu_cxx26__concurrence_unlock_errorE             12            0
.xt.prop._ZTVN9__gnu_cxx24__concurrence_lock_errorE               12            0
.xt.prop._ZTVN9__gnu_cxx26__concurrence_unlock_errorE             12            0
.xt.prop._ZTVSt9bad_alloc                                         12            0
.xt.prop._ZTVN10__cxxabiv120__si_class_type_infoE                 12            0
Total                                                        3459310

its compiled as a release build.

jantje, 2018-04-06:
@ORGATHM 
As a user it is important to understand that Sloeber is responsible to translate the information provided by the json file in build commands. There are about 500 boards that work with sloeber out of the box. 
There is no way anyone can know all of these boards.
If there is a problem you should try to find out whether it is a "translation" issue or a board issue.
Board issue->go to the boards support
translation issue->Sloeber
Gui user issue->Sloeber (I'll see wether it is a CDT or sloeber issue)

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

closed 2018-04-06 by jantje

RickKimball, 2018-08-03:
This issue has been fixed by #1005. * at least I tried the given Blink.ino.txt on a bluepill and it now works fine. I'm sure that the String object was using a malloc/_brk from the newlib instead of the variant wirish/syscalls.c

ORGATHM, 2018-08-05:
Thanks for the info.
I was forced to switch to ESP boards. And I am very happy, now. They has got more CPU Power, more Ram, more Rom, Wifi and a lot of I/O ports, and they are programmable with C++ (Arduino compatible).
I will go to test and confirm your info, in near future.
What Arduino Lib Repo/Json you are using?
Best regards,
DOM