Sloeber/arduino-eclipse-plugin #945: subdir.mk:31: recipe for target 'sloeber.ino.cpp.o' failed // 3rd Party Board Support Radino32

88aurora88, 2018-03-25:
hello,

i have an issue with sloeber by trying to use a specific 3rd party board library. I receive the following error message:

"C:\\ECLIPSE\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:\ECLIPSE\/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_radino32 -DVECT_TAB_BASE -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -mcpu=cortex-m3 -DF_CPU=24000000L -DARDUINO=10802 -DARDUINO_RADINO32 -DARDUINO_ARCH_STM32L1 -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32L1__  "-IC:\ECLIPSE\/arduinoPlugin/packages/In-Circuit/hardware/STM32L1/1.0.3/system/libopenstm32l1" "-IC:\ECLIPSE\/arduinoPlugin/packages/In-Circuit/hardware/STM32L1/1.0.3/system/libmaple" "-IC:\ECLIPSE\/arduinoPlugin/packages/In-Circuit/hardware/STM32L1/1.0.3/system/libmaple/include" "-IC:\ECLIPSE\/arduinoPlugin/packages/In-Circuit/hardware/STM32L1/1.0.3/system/libmaple/stm32f1/include"                              "-IC:\ECLIPSE\/arduinoPlugin/packages/In-Circuit/hardware/STM32L1/1.0.3/system/libmaple/usb/stm32f1" "-IC:\ECLIPSE\/arduinoPlugin/packages/In-Circuit/hardware/STM32L1/1.0.3/system/libmaple/usb/usb_lib"  -I"C:\ECLIPSE\arduinoPlugin\packages\In-Circuit\hardware\STM32L1\1.0.3\cores\radino32" -I"C:\ECLIPSE\arduinoPlugin\packages\In-Circuit\hardware\STM32L1\1.0.3\variants\radino32" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
The system cannot find the path specified.
subdir.mk:31: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1

Here are the simple steps to recreate this error.

STEP 01

I copied the link of the 3rd party library to sloeber to the third party index list.

[https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls](https://github.com/arduino/Arduino/wiki/Unofficial-list-of-3rd-party-boards-support-urls)

[image]

[image]

STEP02

after everything has been installed properly I start a new arduino sketch called "TEST" based on the freshly installed package:

[image]

I set the following project parameters and press finish:

[image]

Out of the box the new project shows the following error message mentioned earlier:

[image]

jantje, 2018-03-25:
This platform is dependent on Arduino provided tools but it does not provide the tool info.
Sloeber does not handle this so these tools are not installed.
You will need to install the correct arduino arm platform version for this to build

jantje, 2018-03-25:
Technically what happens is the following:
Looking in the content of the json file you can find a section like
http://library.radino.cc/Arduino_1_8/package_radino_radino32_index.json
```
          "toolsDependencies": [
            { "packager": "arduino",
              "name": "arm-none-eabi-gcc",
              "version": "4.8.3-2014q1"
            },
            {
              "packager": "arduino",
              "name": "dfu-util",
              "version": "0.9.0-arduino1"
            }
```

however the tools section looks like 

```
      "tools": [
        
      ]
```
The arduino IDE goes and looks for the correct json file to find the installation information. Sloeber does not do this.
In this case the installation information is located at `package_index.json`
and used by the "Arduino STM32F4 Boards"

As a workaround install the the platform version using these tools (same or newer version depending on your Sloeber preference settings "Select the toolchain like the arduino IDE" the version needs to be the "same or newer" if on and the same if off)
If you do not want to see the boards in the gui you can remove the platform as this will not remove the tools

88aurora88, 2018-03-25:
hello,

I think I do not unterstand what exactly I have to do now to make this compile.

I installed the STM32F4 Boards but nothing has changed the error ?

[image]

jantje, 2018-03-25:
Do you have the executable C:\ECLIPSE/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ on your system?

uzi18, 2018-03-25:
go into project properties -> Arduino -> apply ->apply&save this will rebuild project configuration, and You will be able to build project.

88aurora88, 2018-03-28:
@jantje i think i now now what the issue is.

the path to the executeable is:

"C:\ECLIPSE\arduinoPlugin\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\bin"

but sloeber searches in:

"C:\ECLIPSE\/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++"

seems like something is messed up.

how do i fix this ?

jantje, 2018-03-28:
I know it is confusing but ...
There is the gcc command
[image]
and the arm-none-eabi-g++ command
[image]
The platform.txt refers to arm-none-eabi-g++.  According to my install that is the correct installation location.
If you do not have the arm-none-eabi-g++ installed something may have gone wrong in the Sloeber download or install. I have seen unreported failures.
Easiest to fix is: remove the `C:\ECLIPSE\arduinoPlugin\downloads` folder remove the `"C:\ECLIPSE\arduinoPlugin\packages\arduino\tools\arm-none-eabi-gcc"` folder. in sloeber deinstall "Arduino SAM Boards (32-bits ARM Cortex-M3)" platform and reinstall it.

88aurora88, 2018-04-02:
@jantje thank you removing the folder and reinstalling the package worked.

88aurora88, 2018-04-11:
@jantje hello thank you again for your help regarding the previous issue. Unfortunately, I encoutered a follow up issue. as described before I installed the package with the following steps:

[image]

[image]

[image]

so far so good. the new issue occurs if I select the following predefined example:

[image]

in the arduino IDE this example sketch compiles perfectly out of the box. however, in sloeber i run in to the following issues: I would be forever grateful if you can help me to make this example sketch compile properly in sloeber. thank you in advance.

[image]

"C:\\ECLIPSE\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ..\sloeber.ino.cpp'
'Starting C++ compile'
"C:\ECLIPSE\/arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_radino32 -DVECT_TAB_BASE -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -mcpu=cortex-m3 -DF_CPU=24000000L -DARDUINO=10802 -DARDUINO_RADINO32 -DARDUINO_ARCH_STM32L1 -DMCU_STM32F103CB -mthumb -march=armv7-m -D__STM32L1__  "-IC:\ECLIPSE\/arduinoPlugin/packages/In-Circuit/hardware/STM32L1/1.0.3/system/libopenstm32l1" "-IC:\ECLIPSE\/arduinoPlugin/packages/In-Circuit/hardware/STM32L1/1.0.3/system/libmaple" "-IC:\ECLIPSE\/arduinoPlugin/packages/In-Circuit/hardware/STM32L1/1.0.3/system/libmaple/include" "-IC:\ECLIPSE\/arduinoPlugin/packages/In-Circuit/hardware/STM32L1/1.0.3/system/libmaple/stm32f1/include"                              "-IC:\ECLIPSE\/arduinoPlugin/packages/In-Circuit/hardware/STM32L1/1.0.3/system/libmaple/usb/stm32f1" "-IC:\ECLIPSE\/arduinoPlugin/packages/In-Circuit/hardware/STM32L1/1.0.3/system/libmaple/usb/usb_lib"  -I"C:\ECLIPSE\arduinoPlugin\packages\In-Circuit\hardware\STM32L1\1.0.3\cores\radino32" -I"C:\ECLIPSE\arduinoPlugin\packages\In-Circuit\hardware\STM32L1\1.0.3\variants\radino32" -I"C:\ECLIPSE\arduinoPlugin\packages\In-Circuit\hardware\STM32L1\1.0.3\libraries\In-Circuit" -I"C:\ECLIPSE\arduinoPlugin\packages\In-Circuit\hardware\STM32L1\1.0.3\libraries\In-Circuit\src" -I"C:\ECLIPSE\arduinoPlugin\packages\In-Circuit\hardware\STM32L1\1.0.3\libraries\SPI" -I"C:\ECLIPSE\arduinoPlugin\packages\In-Circuit\hardware\STM32L1\1.0.3\libraries\SPI\src" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "..\sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
..\sloeber.ino.cpp:10:27: fatal error: DW1000Ranging.h: No such file or directory
 #include <DW1000Ranging.h>
                           ^
compilation terminated.
subdir.mk:31: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 1

jantje, 2018-04-11:
>#include <DW1000Ranging.h>

This library is not in the libraries json  file of arduino. Did you install this somewhere in Arduino IDE? If you you have to add that location as private library path.

>I would be forever grateful if you can help me to make this example sketch compile properly in sloeber. 

Why don't you become a patreon?
As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
For 5$ a month you can get rid of the nag screen telling you that some people work hard for this tool.
As a supporter you can also drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

88aurora88, 2018-04-12:
> Why don't you become a patreon?

I just signed up and became a patreon in the hope that you will help me to get this sketch compiled in sloeber.

> This library is not in the libraries json file of arduino. Did you install this somewhere in Arduino IDE? If you you have to add that location as private library path.

The library is automatically installed with the package named "in-Circuit" see the following picture. The question is how do I make sloeber find the library. As I mentioned earlier the Arduino IDE compiles this sketch without errors out of the box.

[image]

jantje, 2018-04-12:
this is a duplicate of #461 
You can add include "dw1000.h" to your sketch
As the library is already installed, you can add the library mmanually via arduino->add library