Sloeber/arduino-eclipse-plugin #946: wrong path in Elf file creation

devops77, 2018-03-26:
**Seems that this should be a simple fix but, I am new to eclipse and arduino. Have done lots of  CPP code in past so I am a new old guy.
When I build all files compile but when I get to the make *.Elf file is fails:**

"C:\E\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o "C:/E/sloeber-WS/DeathStar2018/Release/DeathStar2018.elf"    .\Actor.cpp.o .\DeathRay.cpp.o .\NeoPixelColor.cpp.o .\SceenBase.cpp.o .\TieFighter.cpp.o .\TieFighterExplode.cpp.o .\VentPort.cpp.o .\global.cpp.o .\sloeber.ino.cpp.o  .\AdaFruitNeoPixels\Adafruit_NeoPixel.cpp.o .\AdaFruitNeoPixels\esp8266.c.o    C:/Users/BigDaddy/git/GitRepo/DeathStar2018/Release/arduino.ar  "-LC:/E/sloeber-WS/DeathStar2018/Release" -lm
c:/e/sloeber/arduinoplugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld.exe: cannot open output file C:/E/sloeber-WS/DeathStar2018/Release/DeathStar2018.elf: No such file or directory
collect2.exe: error: ld returned 1 exit status
makefile:80: recipe for target 'DeathStar2018.elf' failed
make: *** [DeathStar2018.elf] Error 1

**If I create the directory that make is looking for I get a different, but related issue:**

"C:\E\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-size" -A "C:/Users/BigDaddy/git/GitRepo/DeathStar2018/Release/DeathStar2018.elf"
C:\E\sloeber\/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-size: 'C:/Users/BigDaddy/git/GitRepo/DeathStar2018/Release/DeathStar2018.elf': No such file
makefile:67: recipe for target 'DeathStar2018' failed
make: *** [DeathStar2018] Error 1

**seems like an env option is set wrong.**

created my project by 
1) import repro from github
2) create project from local repo

jantje, 2018-03-26:
Have you seen this? https://youtu.be/FrtiVYjF7vU

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

devops77, 2018-03-26:
Happy to support you, I am now a patron.
What do I do next to "drag your attention"?

devops77, 2018-03-26:
I am now a patron

jantje, 2018-03-26:
Thanks
Have you looked at the video I provided above? As there is no project.ini in your repository ... that is what should be done. All other ways I tried to import failed.

devops77, 2018-03-27:
The video helped. I still got errors following the steps but found that a few adjustments to settings would allow it to work.  Thanks for your help!

Error msg  during second import was something like "<Project Name>/core already exists."
fix was to just exit import, most of project was imported, and adjust project settings that did not get set.
Can close this now.

closed 2018-03-27 by devops77

jantje, 2018-03-27:
txs for the feedback