Sloeber/arduino-eclipse-plugin #947: avr-gcc / ld segmentation fault

winginitau, 2018-03-26:
G'day,

Out of my depth. Any ideas on this? It seems to be optimisation related. Same project as a linux simulation compiles and links fine. If I remove all -O optimisation it compiles fine but is huge. Just thought it might be worth bouncing it past you guys before I head off down other paths (have been fighting with it all day). 

Many thanks
Brendan

16:38:35 **** Incremental Build of configuration Release for project ff_arduino ****
make all 
Starting combiner
"/home/brendan/eclipse/cpp-oxygen/eclipse//arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc" -Wall -Wextra -Os -Wl,--gc-sections,--relax -mmcu=atmega2560  -o "/home/brendan/git/FodderFactory/ff_arduino/Release/ff_arduino.elf"    ./sloeber.ino.cpp.o  ./src/SerialPort.cpp.o ./src/ff_HAL.cpp.o ./src/ff_IniFile.cpp.o ./src/ff_bucket.cpp.o ./src/ff_controllers.cpp.o ./src/ff_datetime.cpp.o ./src/ff_debug.cpp.o ./src/ff_display.cpp.o ./src/ff_events.cpp.o ./src/ff_filesystem.cpp.o ./src/ff_ini_config.cpp.o ./src/ff_inputs.cpp.o ./src/ff_main.cpp.o ./src/ff_monitors.cpp.o ./src/ff_outputs.cpp.o ./src/ff_registry.cpp.o ./src/ff_rules.cpp.o ./src/ff_schedules.cpp.o ./src/ff_string_consts.cpp.o ./src/ff_utils.cpp.o ./src/ff_validate.cpp.o  ./libraries/Wire/src/utility/twi.c.o  ./libraries/Wire/src/Wire.cpp.o  ./libraries/SPI/src/SPI.cpp.o  ./libraries/SD/utility/Sd2Card.cpp.o ./libraries/SD/utility/SdFile.cpp.o ./libraries/SD/utility/SdVolume.cpp.o  ./libraries/SD/File.cpp.o ./libraries/SD/SD.cpp.o  ./libraries/RTClib/RTClib.cpp.o  ./libraries/OneWire/OneWire.cpp.o  ./libraries/Ethernet/src/utility/socket.cpp.o ./libraries/Ethernet/src/utility/w5100.cpp.o  ./libraries/Ethernet/src/Dhcp.cpp.o ./libraries/Ethernet/src/Dns.cpp.o ./libraries/Ethernet/src/Ethernet.cpp.o ./libraries/Ethernet/src/EthernetClient.cpp.o ./libraries/Ethernet/src/EthernetServer.cpp.o ./libraries/Ethernet/src/EthernetUdp.cpp.o  ./libraries/DallasTemperature/DallasTemperature.cpp.o  ./itch/NodeMap.cpp.o ./itch/Parser.cpp.o ./itch/TokenList.cpp.o ./itch/itch.cpp.o  ./glitch/OutputBuffer.cpp.o ./glitch/StringList.cpp.o ./glitch/out.cpp.o ./glitch/out_user_code.cpp.o    /home/brendan/git/FodderFactory/ff_arduino/Release/arduino.ar  "-L/home/brendan/git/FodderFactory/ff_arduino/Release" -lm
/home/brendan/git/FodderFactory/src/ff_registry.cpp: In function 'Operate':
/home/brendan/git/FodderFactory/src/ff_registry.cpp:137:1: internal compiler error: Segmentation fault
 }
 ^
Please submit a full bug report,
makefile:92: recipe for target 'ff_arduino.elf' failed
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
lto-wrapper: /home/brendan/eclipse/cpp-oxygen/eclipse//arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/avr-gcc returned 1 exit status
/home/brendan/eclipse/cpp-oxygen/eclipse/arduinoPlugin/packages/arduino/tools/avr-gcc/4.9.2-atmel3.5.4-arduino2/bin/../lib/gcc/avr/4.9.2/../../../../avr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make: *** [ff_arduino.elf] Error 1

16:38:37 Build Finished (took 1s.459ms)

jantje, 2018-03-26:
I assume you are modifying platform.txt and boards.txtx file.
Seems like the command you generate causes a avr-gcc issue. gnu.org even asks to "submit a full bug report"
I don't think this is a Sloeber issue.

winginitau, 2018-03-26:
No I haven’t modified any settings - I just run with the standard sloeber setup and have tried it as a new project too (selecting the board etc as part of the wizard).

It’s occurred when I added in a new source folder and some modules. From those, I include a new object (all ok). It’s only when I then call a method from that object that the problems start. If I comment out the call then it’s fine (although the object is still initialized). Am wondering if I’ve exceeded the memory model or some such.

Perhaps some mapping of compiler options through to LDFLAGS? Interestingly, if I remove -Os from the CPP Flags then it works fine (but the .data and .bss end up over 30k which ain’t going to work on an 8k Arduino).

Agree most likely not a Sloeber issue but thought worth asking in case you had come across it in a Sloeber context.

Brendan

winginitau, 2018-03-27:
Hold for now - I'll do some more triage and see if it can be isolated to something easily repeatable. 

FYI re your questions:
. Sloeber nightly / Eclipse Oxygen 3 on ubuntu mate
. 2560 mega
. I modify Project/Pref/C-C++Build/Environment/A.COMPILER.CPP.FLAGS to remove -Os
. Linux sim is the same source / hierarchy using pre-proc directives to remove anything AVR specific and substituting linux IO and random / simulated data (very useful for dev/debug). So not equivalent at all re linking, but it does prove the overall structure of my code / modules / source files etc
. Yes I am a patreon.

I'll report back once I've done more more digging...

jantje, 2018-03-27:
First of all... thanks for being a patron :-)
Googling " internal compiler error: Segmentation fault" confirms me thinking it is a gcc problem. One of these Weird once were adding a space can fix it. (but hey.. I'm biased ;-) )
Modifying Project/Pref/C-C++Build/Environment/ is a bad idea. Soeber can decide to set the values back at any  time. I do not advice it but "Project/Pref/C-C++Build/build variables" is a better place (at least it won't be overwritten by Sloeber).

> . Linux sim is the same source / hierarchy using pre-proc directives to remove anything AVR specific and substituting linux IO and random / simulated data (very useful for dev/debug). So not [...]

Have you seen jantje local? Supports local debugging and unit testing. If you have... any feedback?

winginitau, 2018-04-09:
Update. Definitely not a sloeber problem though I'm yet to fully work it out. 
FYI points of relevance in my research so far:
. Unoptimised code > 64k plus about 50k in static strings
. I make a lot of use of PROGMEM for static strings. 
. basic PROGMEM macros return 16 bit "NEAR" pointers. 
. Usage of the PROGMEM "FAR" macros would return 32 bit pointers 
. Looking into ways to force the linker to allocate specific PROGMEM strings to predefined memory segments.
. Which would allow mixed use of far and near versions - eg put all debug strings into a static >64k location and use the far pointers / macros only for the debug strings 

So I'm guessing that its a bug in ar/ld where an exception is not caught that results from a bug in the code that tries to access a >64k flash memory location using only 16 bit pointers.... still digging.

Thanks again for your help and pointers on the environment vars. What is jantje local and where can I find it?

Cheers
BM

closed 2018-04-09 by winginitau

uzi18, 2018-04-09:
GCC locates all progmem strings default in first 64k, so You not need to do any magic there with far pointers etc.

jantje, 2018-04-09:
> What is jantje local and where can I find it?

Jantje local allows you to compile your arduino code as a local executable. If you seperate your business logic (read stuff that is not hardware specific) from your Arduino code (read hardware specific) you can compile and run your sketch and debug/unit test it on your development machine.
I packaged it as a board so you can install it like any other board and use multiple configurations to have a release, local debug and unit test configuration. Allowing to swap between them without having to modify code. 
The json file is: https://raw.githubusercontent.com/jantje/hardware/master/package_jantje_index.json