Sloeber/arduino-eclipse-plugin #948: Issue when I upload the STM32F103C series blink sketch on sloeber V4.2

lewiss66, 2018-03-28:
Hi there, I wanted to report an issue when I upload the STM32F103C series blink sketch on sloeber V4.2
`Starting C++ compile
"/home/lewis/sloeber//arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10802 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ "-I/home/lewis/sloeber//arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple" "-I/home/lewis/sloeber//arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/include" "-I/home/lewis/sloeber//arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/stm32f1/include"                              "-I/home/lewis/sloeber//arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/usb/stm32f1" "-I/home/lewis/sloeber//arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/system/libmaple/usb/usb_lib"  -I"/home/lewis/sloeber/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/cores/maple" -I"/home/lewis/sloeber/arduinoPlugin/packages/stm32duino/hardware/STM32F1/2018.3.18/variants/generic_stm32f103c" -MMD -MP -MF"sloeber.ino.cpp.d" -MT"sloeber.ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../sloeber.ino.cpp"  -o  "sloeber.ino.cpp.o"
/bin/sh: 1: /home/lewis/sloeber//arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++: not found
subdir.mk:31
: recipe for target 'sloeber.ino.cpp.o' failed
make: *** [sloeber.ino.cpp.o] Error 127`

jantje, 2018-03-28:
Have you installed the arduino due platform?
I assume this is a duplicate of #945 

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

lewiss66, 2018-03-28:
Yes, I installed the platform Arduino SAM boards V 1.6.11. Is it the right one? By the way I'm on linux...
(I installed the Arduino STM32F4 as well.)
There is something weird when I install a new platform from Arduino/preference/Platforms and Boards menu. An error windows shows up saying:
 " Multiple problems have occured" Installing arduino board plarforms - adopting arduino Boards platforms. I click Yes and it seems that the boards platform is now installed ?!?

lewiss66, 2018-03-28:
Under the directory /arduinoPlugin/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1 I got only one folder called "lib".

lewiss66, 2018-03-28:
Ok, So I did the trick you explained in the #945 and I did work but not straight forward.
I removed the folders "Download" and "arm-none-eabi-gcc". in sloeber deinstall "Arduino SAM Boards (32-bits ARM Cortex-M3)" platform and reinstall it. But then again I got the error "Problem Occured" windows saying:
 'Adopting Arduino Board Platforms' has encountered a problem. Installing Aduino Board PLatfroms.
Then I reopened Sloeber but still problem for the project build. Checked the arm-none-eabi-gcc directory which was not present! 
So then I got back to the Arduino/preferences/ platforms and press again apply & close. and now I could see the folder arm-none-eabi-gcc in the ardhuino-plugin directory!
Something seems to be not consistent...
Thanks anyway.

lewiss66, 2018-03-28:
On the details It says: "Installing Arduino Board Platforms   BoardsManager is still busy"

jantje, 2018-03-28:
That is probably a bug due to the previously failed install.
Normally the error log contains the previous install failure with more details on what went wrong then.

jantje, 2018-03-29:
You will need to restart Sloeber to get rid of the "Installing Arduino Board Platforms BoardsManager is still busy"

closed 2018-03-30 by lewiss66