Sloeber/arduino-eclipse-plugin #950: Arduino Language Settings Provider trims some defines at the first space

EmbeddedG, 2018-04-04:
Comparing discovered entries provided by Arduino Compiler Settings Provider and the specs generated directly by GCC, I see that some macros with space in the value are trimmed at the first space, eg.:
`__UINT32_TYPE__` becomes "`long `" instead of "`long unsigned int`"
but some are ok, eg.:
`__UINT32_C(c)` is correctly "`c ## UL`"

jantje, 2018-04-04:
I can confirm the issue.
the issue does not happen with the CDT compiler settings provider. So it is a Sloeber issue.

As an independent open source project we value your submission, but we ask for support, either by helping us out coding (yeah, we do understand it does require time) or a Patreon contribution (starting from as low as 1$ a month): this allows us to support people who support us back!
As a supporter you can drag our attention and prioritize your submission...
please become a supporter!
https://www.patreon.com/jantje

jantje, 2018-04-11:
Fixed in nightly so closing

closed 2018-04-11 by jantje